zephyrproject-rtos/zephyr #25820: kernel: k_timer_start(timer, K_FOREVER, K_NO_WAIT) expires immediately

nordic-krch, 2020-05-29:
**Describe the bug**
When timer is started with `K_FORVER` it shall never expire but it expires immediately.

**To Reproduce**
Use `hello_world` sample and modify `main.c`:
```
#include <zephyr.h>
#include <sys/printk.h>

void timeout(struct k_timer *t)
{
	printk("timeout %d\n", k_cycle_get_32());
}
K_TIMER_DEFINE(tmr, timeout, NULL);

void main(void)
{
	k_timer_start(&tmr, K_FOREVER, K_NO_WAIT);

	k_sleep(K_FOREVER);
}
```
Observe something like:
```
*** Booting Zephyr OS build v2.3.0-rc1-366-g3a0453648779  ***
timeout 120104
```

**Expected behavior**
`timeout 120104` shall never appear on the console.

**Impact**
Expected timer misbehaves and might have critical impact.

**Environment (please complete the following information):**
commit: 3a04536487

nordic-krch, 2020-05-29:
@andyross the problem is caused by https://github.com/zephyrproject-rtos/zephyr/blob/ba487feb0f2c6c1edeb4ddb52db3405f94c7414e/kernel/timer.c#L132

if ticks is `K_FORVER_TICKS` then `Z_TICKS_ABS` returns `-1` and `duration.ticks` is set to 0.
```
if (Z_TICK_ABS(duration.ticks) < 0) {
	duration.ticks = MAX(duration.ticks - 1, 0);
}
```

I would change it to:
```
if (Z_TICK_ABS(duration.ticks) < K_FORVER_TICKS) {
```
but leaving it to you as there might be better/cleaner solution.

andyross, 2020-05-29:
Good point.  FWIW: your fix would have saved a comparison, but didn't address the same bug in LEGACY mode.  In this situation it was easier to just add the single test/branch to the top of the function.

PR #25830 (2020-05-29): kernel/timer: Handle K_FOREVER in k_timer_start()
The possibility of passing K_FOREVER as the initial duration argument
to k_timer_start() wasn't being handled, with the result that the
computed value became an zero timeout (effecitvely treating it as
K_NO_WAIT and firing at the next tick).

This is obviously pathlogical, but it should still do what the code
says it should and wait forever.

Make k_timer_start(..., K_FOREVER, ...) a noop.

Fixes #25820
commit a343cf94 (2020-05-29): kernel/timer: Handle K_FOREVER in k_timer_start()

The possibility of passing K_FOREVER as the initial duration argument
to k_timer_start() wasn't being handled, with the result that the
computed value became an zero timeout (effecitvely treating it as
K_NO_WAIT and firing at the next tick).

This is obviously pathlogical, but it should still do what the code
says it should and wait forever.

Make k_timer_start(..., K_FOREVER, ...) a noop.

Fixes #25820
closed 2020-05-29 by carlescufi