zephyrproject-rtos/zephyr #25832: [test][kernel][lpcxpresso55s69_ns] kernel cases meet ESF could not be retrieved successfully

hakehuang, 2020-05-29:
**Describe the bug**
kernel test case: kernel.memory_protection.gap_filling reports
ESF could not be retrieved successfully

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=lpxxpresso55s69_ns -D
3. make flash

**Expected behavior**
ALL test pass

**Impact**
non-security core may have problem

**Screenshots or console output**
```
*** Booting Zephyr OS version 2.3.0-rc1  ***
Running test suite memory_protection_test_suite
===================================================================
starting test - test_permission_inheritance
PASS - test_permission_inheritance
===================================================================
starting test - test_mem_domain_valid_access
PASS - test_mem_domain_valid_access
===================================================================
starting test - test_mem_domain_invalid_access
ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:945
          ESF could not be retrieved successfully. Shall never occur.
ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core

```

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain ( Zephyr SDK)
 - v2.3.0-rc1

**Additional context**
many other kernel test report same error:
kernel.memory_protection.syscalls
kernel.memory_protection.userspace
kernel.memory_protection.userspace.gap_filling
kernel.common.stack_protection_arm_fpu_sharing
kernel.common.stack_protection_armv8m_mpu_stack_guard
kernel.common.stack_protection_no_userspace

hakehuang, 2020-05-29:
@MaureenHelm

agansari, 2020-08-03:
Issue is still under investigation, please remove Stale label.

hakehuang, 2021-06-23:
this issue still occures in zephyr-v2.6.0-286-g46029914a7ac
```
*** Booting Zephyr OS build zephyr-v2.6.0-286-g46029914a7ac  ***
Running test suite fatal
===================================================================
START - test_fatal
test alt thread 1: generic CPU exception
ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
ESF could not be retrieved successfully. Shall never occur.
ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
ESF could not be retrieved successfully. Shall never occur.
```

dleach02, 2021-08-17:
@hakehuang , can you retest this? I don't have a 55s69 board

hakehuang, 2021-08-18:
@dleach02  as lpcxpresso55s69_ns need security counter part install, so this case need run with lpcxpresso55s69_cpu1, Let me close this issue. which is OK now.

closed 2021-08-18 by hakehuang