zephyrproject-rtos/zephyr #25847: Problems using math functions and double.

jaslob, 2020-05-30:
I'm new to Zephyr and did another small test application.
I'm able to build the application but when I try to run the application I receive an error.
I can't use the math functions and have also trouble just adding two doubles. Why is that? I guess I have problems with the configuration. Thanks for your help.

main.cpp
#include <zephyr.h>
#include <sys/printk.h>
#include <math.h>
#include <stdio.h>
#include <iostream>

int main()
{
    double angle = 90;
    double rad;
    double sine;
    double cosine;
    
    const double Pi = 3.141592653;
    
    rad = angle * Pi / 180;
    sine = sin(rad);
    cosine = cos(rad);
    
    std::cout << "Angle: " << angle << "\nSine: " << sine << "\nCosine: " << cosine << std::endl;
    
    return 0;
}

prj.conf
CONFIG_NEWLIB_LIBC=y
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_STD_CPP17=y

Error:
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200214_052440-f7294c49af13-zephyr
)
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 7
*** Booting Zephyr OS build v2.3.0-rc1-82-gc8a30ecacc50  ***
FAILED: zephyr/CMakeFiles/run 
cd /home/user/zephyrproject/zephyr/project/math/sincos/build && /home/user/Zephyr/zephyr-sdk-0.11.2/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-i386 -m 9 -cpu qemu32,+nx,+pae -device isa-debug-exit,iobase=0xf4,iosize=0x04 -no-reboot -nographic -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -kernel /home/user/zephyrproject/zephyr/project/math/sincos/build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/user/zephyrproject/zephyr/project/math/sincos/build --target run

stephanosio, 2020-05-30:
Try setting `CONFIG_MAIN_STACK_SIZE=2048`

jaslob, 2020-05-30:
Thanks but that didn't help.

stephanosio, 2020-05-30:
Try setting:
```
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_FPU=y
CONFIG_SSE=y
```

jaslob, 2020-05-30:
Thank you so much. That helped.

carlescufi, 2020-05-31:
Closing this as it seems solved. Thanks @stephanosio

closed 2020-05-31 by carlescufi