zephyrproject-rtos/zephyr #25853: updatehub: Cannot connect to UDP remote

mtahirbutt, 2020-05-31:
**Describe the bug**
A clear and concise description of what the bug is.

I am using modem with updatehub, #25558. But I see the similar error pattern as that with #25600. 
Once I switch off (remove) antenna. The signal levels go down and socket connection gets lost but cannot connect again unless I reset the board.  

What have you tried to diagnose or workaround this issue?

I also tried to reconnect via various ways; by closing the socket connection explicitly before starting another one. but failed unless I reset the board.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=custom board with STM32L475
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

I should be able to connect to CE server after connection is lost.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

If signal is lost due to some uncertain environment, update becomes problematic.

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:05:03.221,000] [1B][0m<inf> modem_ublox_sara_r4: RSSI: -81[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m*** Booting Zephyr OS build v2.3.0-rc1-392-g0eebd70718c6  ***
[1B][8D[1B][J[00:05:07.992,000] [1B][0m<inf> modem_ublox_sara_r4: Network is ready.[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:05:37.999,000] [1B][0m<inf> modem_ublox_sara_r4: RSSI: -83[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:06:08.007,000] [1B][0m<inf> modem_ublox_sara_r4: RSSI: -87[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:06:38.000,000] [1B][0m<inf> main: UpdateHub sample app started (hw02)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:06:38.000,000] [1B][0m<inf> main: Confirming the boot image[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:06:38.000,000] [1B][0m<inf> main: Starting UpdateHub polling mode[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:06:38.017,000] [1B][0m<inf> modem_ublox_sara_r4: RSSI: -93[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:06:39.332,000] [1B][0m<inf> updatehub: No update available[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:07:08.026,000] [1B][0m<inf> modem_ublox_sara_r4: RSSI: -1000[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:07:38.036,000] [1B][0m<inf> modem_ublox_sara_r4: RSSI: -1000[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:07:39.606,000] [1B][1;31m<err> modem_ublox_sara_r4: AT+USOCR=17 ret:-5[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:07:39.606,000] [1B][1;31m<err> updatehub: Cannot connect to UDP remote[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:08:08.044,000] [1B][0m<inf> modem_ublox_sara_r4: RSSI: -77[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:08:38.056,000] [1B][0m<inf> modem_ublox_sara_r4: RSSI: -75[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:08:39.615,000] [1B][1;31m<err> modem_ublox_sara_r4: AT+USOCR=17 ret:-5[1B][0m
**[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:08:39.615,000] [1B][1;31m<err> updatehub: Cannot connect to UDP remote[1B][0m**
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:09:08.065,000] [1B][0m<inf> modem_ublox_sara_r4: RSSI: -75[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:09:38.078,000] [1B][0m<inf> modem_ublox_sara_r4: RSSI: -77[1B][0m

**Environment (please complete the following information):**
 - OS: (e.g. Ubuntu 18.04.4 LTS )
 - Toolchain (e.g Zephyr SDK 0.11.2)
 - Commit SHA 3852e081261854849341f2cdd89b0da222089e96

**Additional context**
I expect something is wrong with socket connection management, as the problem is similar found with WiFi but not yet able to find that.

nandojve, 2020-06-02:
CC: @otavio

jukkar, 2020-06-02:
There was some discussion in slack about modems and PPP.
https://zephyrproject.slack.com/archives/CK4LW9NBB/p1590335212147600
So basically if PPP connection goes to PPP_DEAD state, it cannot recover itself but would require some entity to restart it. Are you using PPP with the modem or offloading driver with AT commands?

jukkar, 2020-06-03:
> modem_ublox_sara_r4: Cannot connect to UDP remote

I am assuming here, because of the PR description, that the offloading driver is being used. In that case my PPP comment is not valid.

mtahirbutt, 2020-06-03:
thanks for your remarks. I just tried to play with NETWOKR_TIMEOUT and recv(ctx.sock, data, MAX_DOWNLOAD_DATA, MSG_DONTWAIT) from socket function and I reduced the frequency of usage of recv() function, and the situation improved and do not see the problem. As for now important is the fact which aspects I need to handle with system reset and which I can handle without system reset. I will update #25558 addressing the situation where radio signal gets poor and the updatehub is still able to handle that situation.

closed 2020-09-20 by github-actions[bot]