zephyrproject-rtos/zephyr #25858: Net shell utilities do not work with NET_OFFLOAD or NET_OFFLOAD_SOCKET

aport, 2020-05-31:
**Describe the bug**
When using a SARA-R4 modem with the NET_OFFLOAD driver, I would expect the net shell utilities to work. Unfortunately they depend on NET_NATIVE to work.

**To Reproduce**
Steps to reproduce the behavior:
1. Build Zephyr with a NET_OFFLOAD=y, NET_NATIVE=n, NET_SHELL=y
2. Attempt to use any of the net shell utilities
3. Observe that utilities have no effect and are non functional

**Expected behavior**
Net shell utilities (ping, tcp) should be able to utilize NET_OFFLOAD interfaces

**Impact**
I must write a custom application to test/evaluate Zephyr's networking functionality when using a NET_OFFLOAD interface, or compile in the native network stack which significantly increases image size.

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain gnuarmemb
 - tag v2.3.0-rc1

jukkar, 2020-06-01:
Atm net-shell tcp command still uses net_context API instead of BSD socket API. So currently tcp command will not work with `CONFIG_NET_OFFLOAD_SOCKET` but should work with  `CONFIG_NET_OFFLOAD` so we could allow that. The best option would be to convert the tcp command to use normal BSD socket APIs so that it could work with all options.

jukkar, 2023-11-16:
The `ping` net-shell command was recently made to support offloading in commit e3fb634e7afa1a14121559e6a0b3ddbeb947c5c5
The `udp` and `tcp` are still using `net_context` and there is currently no plan convert them to use BSD socket API.