zephyrproject-rtos/zephyr #25859: mesh example not working with switched off dcdc?

NilsMinor, 2020-05-31:
**Describe the bug**

I am building the **onoff_level_lighting_vnd_app** example the the **nrf52dk_nrf52832** board setup. As I am using a custom board that does not implement the nrf52 dcdc I changed the Kconfig file and switched off the DCDC. 

```
config BOARD_ENABLE_DCDC
	bool "Enable DCDC mode"
	select SOC_DCDC_NRF52X
	default n
	depends on BOARD_NRF52DK_NRF52832
```

Building and flashing works fine. I am using the nrf mesh app for provisioning. I select the zephyr board, connecting, initialize and provisioning works (_provisioning complete_). The app then automatically is requesting the composition data that does not work properly. In 50% of cases I a get a _Reguest timed out_ error message. 

In case it worked, I can see the the models/elements but when I am not able to bind an application key as I always get an _Reguest timed out_ error message. 

[image]
[image]

**To Reproduce**
Steps to reproduce the behavior:
```
west build -b nrf52dk_nrf52832 samples/boards/nrf/mesh/onoff_level_lighting_vnd_app/
west flash
```
**Environment (please complete the following information):**
- OS Ubuntu 18.02
- Toolchain latest sdk and lates Zephyr

Any Idea what I am doing wrong ? I have troubles using meshctl so I decided to got with the nordic app that worked pretty stable in the past for me. But I am not able to get the example running on my board, a have not changed the code I only switched off the dcdc.

Thanks for help,
Nils

cvinayak, 2020-06-01:
@NilsMinor Please test the samples/bluetooth/peripheral sample on your custom board and check for its interoperability with a samples/bluetooth/central_hr on a nRF52DK board (and may be test with a phone with nRF Connect app). This is to rule out any clock and/or antenna matching issues in your custom boards.

NilsMinor, 2020-06-01:
HI @cvinayak I tested the _samples/bluetooth/peripheral_ example and connected my board to nrf Toolbox which worked fine :) I flashed the _onoff_level_lighting_vnd_app_ example again, on the same board but the behavior is the same. Requesting composition data takes very long and it is not possible to bind an application key.

carlescufi, 2020-06-01:
> HI @cvinayak I tested the _samples/bluetooth/peripheral_ example and connected my board to nrf Toolbox which worked fine :) I flashed the _onoff_level_lighting_vnd_app_ example again, on the same [...]

@trond-snekvik do you have experience with mesh and DC/DC disabled?
@cvinayak do you think advertising and scanning could somehow be affected by this?

cvinayak, 2020-06-03:
No, there is no dependency of DC-DC on advertising, scanning or connections.

@NilsMinor Is this reproducible on a nRF52DK with DC-DC turned off?

NilsMinor, 2020-06-04:
@cvinayak thanks for your response. I do not have an NRF52DK right know but will test it in the future.

closed 2020-06-04 by NilsMinor

cvinayak, 2020-06-06:
@NilsMinor You have closed the issue, was there a conclusion?