zephyrproject-rtos/zephyr #25887: legacy timeout API does not work

pabigot, 2020-06-01:
With `CONFIG_LEGACY_TIMEOUT_API=y` this:
```
void main(void)
{
	printk("Hello World! %s\n", CONFIG_BOARD);
	k_sleep(K_MSEC(3));
	printk("post 3 ms at uptime %u\n", k_uptime_get_32());
	k_sleep(K_USEC(3));
	printk("post 3 us at uptime %u \n", k_uptime_get_32());
	k_sleep(K_NSEC(3));
	printk("post 3 ns at uptime %u\n", k_uptime_get_32());
}
```
prints
```
*** Booting Zephyr OS build v2.3.0-rc1-401-gd7601a541d89  ***
Hello World! nrf52840dk_nrf52840
post 3 ms at uptime 10
post 3 us at uptime 3012 
```
and presumably would print
```
post 3 ns at uptime 3003012
```
if I waited long enough.

andyross, 2020-06-01:
K_USEC and K_NSEC are new macros, they didn't exist for legacy code so I don't undertand filing a bug report against them.  Is there old code that's going to break because the new macros don't honor old units correctly?

andyross, 2020-06-01:
For clarity: I mean, if you want you can stick those macros inside an "#ifndef CONFIG_LEGACY_TIMEOUT_API".  I just don't see the point.

pabigot, 2020-06-01:
Sure, but it's obviously confusing or we wouldn't have gotten that question on slack.  Making it work as expected seems trivial.

andyross, 2020-06-01:
I'd definitely prefer to wall off the new APIs from legacy usage vs. trying to add code to a legacy mode.  Making it work "right" only perpetuates a mode we're trying to deprecate.  The simple answer for anyone trying to do this (i.e. mix legacy mode with the new units) is "don't".

PR #25889 (2020-06-02): timeout: correct non-legacy legacy timeout API
The implementation of Z_TIMEOUT_US() and Z_TIMEOUT_NS() in the legacy
timeout API is incorrect in that it multiplies the input value by the
scale factor rather than dividing it, making K_USEC(3) equivalent to
K_SECONDS(3).  Replace with implementation that doesn't surprise a
user that happens to find and use them.

Fixes #25887
closed 2020-06-02 by carlescufi