zephyrproject-rtos/zephyr #25891: per-memory domain copies of library globals

andrewboie, 2020-06-02:
**Is your enhancement proposal related to a problem? Please describe.**
tl;dr: Library globals only have one instance even if the project has multiple logical applications sharing the same micro-controller.

On MPU-based systems, memory domains control user thread access to memory that isn't the thread's stack or areas marked at boot time for universal access, like program text.

If a Zephyr application uses a library, that library is statically linked into the combined application+kernel. If a Zephyr project contains multiple logical applications, all of these end up in a single statically linked binary. For any given library brought in we can.

It's possible to instantiate `k_mem_partition` to allow discrete access to library globals, but this still shares those globals with all users of the library, even if they would be otherwise completely isolated. Plus, having discrete memory partitions for every different library does not scale, because MPU regions are limited.

**Describe the solution you'd like**
- Link library code to be position independent. It's base addresses for data and bss will only be known at runtime. We're going to need some kind of mechanism to handle all the linking considerations but still include the object code in the zephyr kernel binary. It would make some things simpler if text/rodata is still statically linked with the rest of the kernel/app text, we just need unique data/bss.

- Define a grouping mechanism (`k_process`? This will eventually map directly to VM spaces) within the kernel for defining the scope of a logical application. Need to think about how this maps to memory domains (1:1?)

- Have a mechanism for allocating a properly sized and aligned block of contiguous memory for the data/bss of all shared libraries a `k_process` will use. It should be possible to program an MPU region at its boundaries. The naive way to do this is to always reserve memory assuming all the libraries linked are being used, but it might be possible to be smarter about this. Ideally we do this at build time and no heap.

- When a process is instantiated, do the necessary shared linking so that data and bss for each library will be carved out of a single block of memory that we have reserved in the previous step. The kernel will need its own library context as well.

- Reprogram this region on context switch to reflect the incoming `k_process`

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
This will really force us to distinguish between library code and subsystem code; only the latter needs to be a singleton entity in the kernel's space. I suspect a few items may switch places.

andrewboie, 2020-08-27:
This is going to require some ingenuity, if it's feasible at all.

Standard shared libraries in Linux presume that the GOT is located from a fixed offset from the program text. This works because the library program headers just get mapped into the address space with mmap(). But with a physical memory map, this doesn't fit well.

Ideally:

- We need all the program text for all libraries merged into a single contiguous region so a read-only, executable MPU region can be programmed into it. We wouldn't want any kind of procedure linkage table (PLT), just statically link all program text addresses.
- Same deal with read-only data. Either in another RO region or just part of the text region if we don't care about read-only data having an executable property.
- What has to be position-independent is the data/bss. Any access to such data needs to be relative to some runtime value per memory domain indicating the starting address of that library's data for the domain. We need to support multiple libraries in this way.

Shared libraries just don't work in this way.

What this *does* look sort of like is how thread-local storage works, where a register (or a segment descriptor on x86) is maintained representing a base address for the TLS, per thread. I'm thinking something along the lines of:

- Implement generic TLS https://github.com/zephyrproject-rtos/zephyr/issues/22185
- For each thread, the TLS area is some RAM carved out of the thread's stack.
- Part of the TLS area is an offset table for each library that is in use. The layout of this table is fixed for each thread and has an entry for every shared library in use. Each table entry has the base linear address of where the data for the library begins. These all point to the same memory for threads in the same memory domain. Other, actual thread-local data like errno is also in TLS along side it. But these offset tables are copies of each other for every thread in the same domain.

To reference a library global, there would be a lookup in TLS of the base address of the library data. Then add the offset within the library data for that global to get that global's address in our memory domain.

No linker supports this AFAIK. I don't know if others have needed to solve similar problems, and if so what they came up with. More research is needed.

andrewboie, 2020-08-27:
One of the problems with using actual shared libraries is that the compiler assumes a fixed offset to the GOT from the library program text. All GOT lookups are IP-relative. This works fine in Linux because everything just gets mapped in the process address space.

In zephyr, we would need a GOT for every memory domain. In a physical memory map, the memory for each GOT would be at a different linear address, and the fixed offset guarantee doesn't hold.

One idea is to have a fixed GOT location which gets updated on context switch to the incoming memory domain, a copy is made. This would solve accessing the GOT, but have its own complications:

- Extra memory partition for the fixed global GOT. Maybe not so bad, we're saving many memory partitions by doing this for nontrivial applications.
- Extra context switching overhead, although at least it's predictable
- We still want all the library text/rodata to be contiguous with kernel/app text/rodata, so that we don't need extra MPU regions for each library to enable access. Very unsure if this is feasible.
- This approach falls over on SMP, where different memory domains can be active on different CPUs. Starting to think this might be a dealbreaker.

Given a choice of "break SMP" or "break TLS" my preference still for the latter.

andrewboie, 2020-08-28:
There is a third option: link in multiple copies of libraries per memory domain.

Pluses:
- No funny tricks to get separate library data/bss per domain
- No extra indirection layers, PIC, etc
- Regular .a static libraries work
- Tighter control of what memory domains use what libraries, as they are more like separate applications
- Separately built applications are something that has come up in discussions from time to time for other reasons.

Minuses:
- Multiple copies of almost the same program text, using up either RAM or flash space (depending on XIP)
- For XIP systems, multiple copies of the data section, using up flash space
- Zephyr will need to be built and linked in a fundamentally different way, with each memory domain representing some different linking step
- Memory domains will be closely associated with the build and not the runtime constructions they are now

andrewboie, 2020-09-01:
Another complication: given that debugging sessions are system-wide, how do we handle the address of various shared library symbols if they change depending on what thread is scheduled?

For the TLS approach, these are all treated as thread-local and I expect this will not complicate GDB too much since AFAIK GDB already understands TLS, although this needs further investigation.
For using actual shared libraries, I am not sure how this will work from a debugger perspective.
For multiple copies of static libraries, this will result in naming collisions at final linking time of the combined kernel image.

andrewboie, 2020-09-01:
My thinking on this is converging towards the TLS-based approach being the most promising, despite its drawbacks; at least enough to put together a proof-of-concept. Regardless of method chosen, this will be optional, we'll still always support the current scheme of using a `k_mem_partition` for libary globals as a fallback.

andrewboie, 2020-09-18:
Further thoughts on this and a proposal:

- The first proposal, abusing TLS, is interesting but has serious caveats. It may be worth a proof-of-concept once we have a proper TLS implementation in place, but this is not a priority.
- The second proposal, using real .so files I think may be very complicated to implement and maintain.
- The third proposal, having multiple copies linked, ties into something that comes up in conversations from time to time to generally support a modular Zephyr build, where multiple logical applications built/linked separately are assembled into a final image. If we supported this kind of build, having separate library copies should be a straightforward extension.

I thought about why I was so keen on solving this problem in the first place, and it really boils down to the C library, particularly the heap:

- minimal libc is normally free of globals EXCEPT when the malloc() heap is enabled, in which case it is global and placed in a `z_malloc_partition`.
- newlib/newlib_nano have a bunch of globals, it's why we had to route `libc.a` or `libc_nano.a` into the `z_libc_partition`, presenting an opportunity for every thread in the system to interfere with each other. it also has a single global malloc() heap.

Taking the C library out of the picture makes this much less of a concern.

So this is what I propose to do:

- We should scope out and work on a mechanism for modular Zephyr builds where each Zephyr application is built separately as this is another thing that is generally useful. Add "each application can specify its own set of static libraries" as one of the requirements.
- Restrict the scope of other, more involved solutions to the C library ONLY, such that for any C library used:
  - We don't have multiple copies of its program text/rodata
  - the malloc() arena and any other process-global libc data are managed on a per-memory domain basis
  - `z_libc_partition` and `z_malloc_partition` that everyone currently has to share cease to exist

keith-packard, 2020-09-23:
The picolibc PR (#26545) provides support for TLS using the -ftls=local-exec model. That depends on static linking all objects defining TLS variables as the total size of the TLS space must be known at link time. Picolibc uses TLS for most global variables, including errno and signgam. It does not (currently) place variables controlling the heap in TLS as it is expected that the heap will be shared across threads. It sounds like what we need is a per-domain heap, which would require modifying the malloc code in picolibc and then providing some mechanism to identify the current heap.

Supporting TLS with the minimal C library included in Zephyr would require copying the picolibc TLS support functions that initialize the TLS segment from ROM and the code that sets the TLS base address for architectures not already in-lining that.

Supporting TLS from newlib would be challenging; the per-thread data resides in a rather large struct as it contains every possible per-thread value, rather than being limited to those related to APIs used in the application. I'd say that scoping this issue to require libc TLS support would be a reasonable approach, given that there would be two available.

dcpleung, 2024-02-08:
Using TLS (with compiler and arch support) is the least invasive. So opt for TLS approach unless the need for others arises.

closed 2024-02-08 by dcpleung