zephyrproject-rtos/zephyr #25893: Application syscalls in usermode gives bus fault with stacking error

rgujju, 2020-06-02:
**Describe the bug**
I am trying to follow the instructions about using application syscalls in usermode from [here](https://docs.zephyrproject.org/latest/reference/usermode/syscalls.html) and i am getting a bus fault with stacking error. 

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nrf52840dk_nrf52840
3. make
4. west flash

See minimal source to reproduce [here](https://github.com/rgujju/Dynamic_App_Loading/tree/minimal) (minimal branch)

**Expected behavior**
Expected to call the `z_impl_SetLed` function from usermode as well.

**Impact**
Not able implement syscall functions in the application

**Screenshots or console output**
```
*** Booting Zephyr OS build v2.3.0-rc1-399-g394ff7a28b6b  ***
Setting LED 0 to 1
[00:00:00.006,225] <err> os: ***** BUS FAULT *****
[00:00:00.006,225] <err> os:   Stacking error
[00:00:00.006,225] <err> os:   Precise data bus error
[00:00:00.006,256] <err> os:   BFAR Address: 0x8e8c1ec
[00:00:00.006,256] <err> os: ***** HARD FAULT *****
[00:00:00.006,256] <err> os:   Fault escalation (see below)
[00:00:00.006,256] <err> os: ***** BUS FAULT *****
[00:00:00.006,256] <err> os:   Precise data bus error
[00:00:00.006,256] <err> os:   BFAR Address: 0x8e8c1f0
[00:00:00.006,286] <err> os: r0/a1:  0x200000d4  r1/a2:  0x40000000  r2/a3:  0xe000ed0
[00:00:00.006,286] <err> os: r3/a4:  0x0000265b r12/ip:  0x20003890 r14/lr:  0x0000115
[00:00:00.006,286] <err> os:  xpsr:  0x01000005
[00:00:00.006,286] <err> os: Faulting instruction address (r15/pc): 0x000027a2
[00:00:00.006,286] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.006,317] <err> os: Fault during interrupt handling

[00:00:00.006,317] <err> os: Current thread: 0x20000038 (unknown)
[00:00:01.524,536] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - 394ff7a28b6baf2d41e1d84baae479ef55458f75

**Additional context**

rgujju, 2020-06-02:
This may be related to stack alignment. If i change the way i create a stack from `k_thread_stack_t blinky_stack[4096];` to `K_THREAD_STACK_DEFINE(blinky_stack, 4096);` it works.

andrewboie, 2020-06-02:
Hi,

Use of K_THREAD_STACK_DEFINE() is not optional. You *must* use it to declare stacks. You won't get the alignment required by the MPU right otherwise.

closed 2020-06-02 by andrewboie