zephyrproject-rtos/zephyr #25901: timer: nrf_rtc_timer: Subtraction underflow causing 8 minute time skips

JordanYates, 2020-06-02:
**Describe the bug**
Due to a logic/synchronisation error in `nrf_rtc_timer.c`, it is possible for the tick subtraction in `z_clock_set_timeout` to underflow.

The observed situation is that the value of `last_count` is GREATER than the current counter value. https://github.com/zephyrproject-rtos/zephyr/blob/394ff7a28b6baf2d41e1d84baae479ef55458f75/drivers/timer/nrf_rtc_timer.c#L233

The subtraction then leads to a negative value, and at that point all bets are off in terms of the internal logic. `set_protected_absolute_ticks` ends up getting called with around 2**32 as the absolute timeout, and the sanity of internal variables degrades from there.
This is obviously undesirable.

I am unable to work out exactly how `last_count` ends up less than the counter value, but the situation happens within several seconds of starting a bulk transmission of async UART data from my PC to the Zephyr device. There is a decent amount of work happening in the RX callback, which could be triggering whatever the timing edge condition happens to be.

Reverting the following commit resolves the issue: https://github.com/zephyrproject-rtos/zephyr/commit/10d15d185b63f6616bbb7c9975dff15d52464b11

I believe that one of the assumptions in the new implementation is incorrect, potentially around the fact that it is lockless.

**Expected behavior**
Subtraction underflow should not occur, tickless idle shouldn't break.

**Impact**
Major. The timeout API is what drives the tickless idle implementation, and it cannot be trusted after this condition.

**Screenshots or console output**

Here is the output of 
```
for(;;) {
    printk("%10llu\r\n", k_uptime_ticks());
    k_sleep(K_MSEC(1000));
}
```
Note the huge jumps in uptime.
```
*** Booting Zephyr OS build v2.3.0-rc1-1-g571d88d702c9  ***
[00:00:00.017,303] <inf> fs_nvs: alloc wra: 1, 830
[00:00:00.017,303] <inf> fs_nvs: data wra: 1, 3fc
[00:00:00.021,545] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.021,545] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.021,575] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.3 Build 0
[00:00:00.021,942] <inf> bt_hci_core: Identity: c3:3b:7b:e8:c8:de (public)
[00:00:00.021,942] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x05f1
[00:00:00.021,972] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0xffff
     33570
     66342
     99114
    131886
    164658
    197430
    230202
    262974
<----------- UART starts here
  17071882
  17104654
  33884754
  50685500
  67470565
  84251985
 101044373
<----------- UART stops here
 101077145
 101109917
 101142689
 101175461
 101208233
 101241005
 101273777
 101306549
 101339321
 101372093
```

**Environment (please complete the following information):**
 - Zephyr2.3 rc1

nordic-krch, 2020-06-03:
rc1 did not contain #25517 which was fixing something that looks similar. Can you try? You can just apply this single-liner https://github.com/zephyrproject-rtos/zephyr/pull/25517/commits/b6deff76b5ea876c88ef00ca98a7839c14879587

JordanYates, 2020-06-03:
Solves the issue, guess I should have looked forward in time as well as back :)
Thanks.

closed 2020-06-03 by JordanYates