zephyrproject-rtos/zephyr #25907: Request for Flash Logging feature

rhardik, 2020-06-02:
**Problem**
It's very difficult to connect UART/Jlink to get logs in-field devices.
and some of my devices getting crash after 3 to 4 days but I get no log to debug the issue.

**Solution**
I am looking for a feature where I can store the normal/crash logs in flash.
Similar to  [Nordic_SDK_Flash_Logging_feature]( https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v15.0.0%2Flib_nrf_log_backend_flash.html)

Please give some pointers So I can try it.

carlescufi, 2020-06-19:
See the additional information in #26295

koffes, 2020-06-19:
@rhardik I missed this issue in my search. As Carles pointed ut, my recent #26295 duplicates this issue. I have taken the liberty of posting the details from #26295 below:

**Is your enhancement proposal related to a problem? Please describe.**
All log mechanisms today requires the user/developer to have either RTT/Serial/SWO connected to the board to see log output. Also, if you desire to store logs, this has to be done on the host side of the log link.

In most cases this is fine, but in some situations it would be highly desirable to place logs in on-board persistent storage:
A: A customer experiences an error, but did not have a backend connected. With this solution the log can be retrieved using a debugger from internal flash, USB from external flash or an SD card can be read over USB or put into a PC.

B: Logging from small portable devices. In many cases you have a portable device you want logging where it is cumbersome or impossible to have USB/UART/RTT connected.
E.g. logging battery level over a long period of time.

**Describe the solution you'd like**
LOG_BACKEND_PERSISTENT_STORAGE or
LOG_BACKEND_EXT_FLASH
LOG_BACKEND_SD_CARD
(LOG_BACKEND_INT_FLASH)

If the corresponding KConfig settings have not been made, you will get an error.
You will of course also have to set a file system, have a filename for the log and some kind of full storage handling.

Alternatively
As a first step, provide a LOG_BACKEND_USER or LOG_BACKEND_BLANK. When enabled, logs will e.g. go to UART and the USER backend at the same time. The user can then handle logs in the USER log callback as desired and the specifics of storing to flash or SD card can be left to the user.

zagor, 2020-10-04:
I have submitted #28892 which implements one version of this. But due to the different nature of persistent flash logs versus normal ephemeral logs, I think we want to introduce some new APIs. Feel free to join the discussion in the PR.

manoj153, 2022-01-23:
https://github.com/zephyrproject-rtos/zephyr/pull/32973/commits/8e05a9bb0fa3e921370e753815f99722fc8ec2d9

Have that PR solved the aforementioned issue?