zephyrproject-rtos/zephyr #25910: Cross-arch IRQ priority get/set

carlescufi, 2020-06-02:
I'd like Zephyr to have, in `include/irq.h`:

```
 unsigned int irq_get_priority(unsigned int irq);
 int irq_set_priority(unsigned int irq, unsigned int priority);
```
So one can get and set the priority of an interrupt without having to resort to arch-specific code (eg. `NVIC_SetPriority()` which to make matter worse doesn't take into account the offset introduced by the kernel).

bwasim, 2020-06-02:
In addition of this, it would also be very valuable if there's a generic API for affining interrupts to particular CPUs. Unlike the priority API, the `get_irq_affinity` / `set_irq_affinity` will only be applicable for SMP / AMP scenarios..

stephanosio, 2020-06-03:
@carlescufi While this may not be exactly what you describe, dynamic IRQ priority configuration (i.e. `irq_set_priority` equivalent) can be done in an arch/intc-agnostic manner through `irq_connect_dynamic`.

ioannisg, 2020-06-03:
> @carlescufi While this may not be exactly what you describe, dynamic IRQ priority configuration (i.e. `irq_set_priority` equivalent) can be done in an arch/intc-agnostic manner through `irq_connect_dynamic`.

@stephanosio yeap, we've discussed that, too. I suppose @carlescufi would like to have this option for statically-configured IRQ lines (i.e. not dynamic), as well. This is, currently supported for Cortex-M via a kernel-private API, so I guess, the point is whether to expose this as a public API, and how exactly we can do it. (Might be desired, as well, for direct IRQs, @carlescufi , is this correct?)

Before going with an implementation, I 'd like to see if there are any hidden dangers of exposing this functionality to the app developer.

stephanosio, 2020-06-03:
> I suppose @carlescufi would like to have this option for statically-configured IRQ lines

`IRQ_CONNECT` accepts the `priority` parameter, so one can statically configure IRQ priority in arch/intc-agnostic manner too.

If the goal is to dynamically re-assign the IRQ priority of an interrupt (both static and dynamic) that is already configured (can't really think of any sane usage cases for this though), one could do that using `irq_connect_dynamic`.

ioannisg, 2020-06-03:
OK, so let me re-phrase this: this API should give the option to a user to dynamically change the IRQ priority of a non-dynamic IRQ. 

> If the goal is to dynamically re-assign the IRQ priority of an interrupt (both static and dynamic) that is already configured (can't really think of any sane usage cases for this though), one could do that using irq_connect_dynamic; just need to irq_disable the particular interrupt before doing so (as required by z_isr_install.

This is doable but you need to have `CONFIG_DYNAMIC_INTERRUPTS` enabled (correct?) - I suppose the point here is to be able to change IRQ priorities, without enabling this Kconfig, which affects memory footprint and run-time performance.

stephanosio, 2020-06-03:
> This is doable but you need to have `CONFIG_DYNAMIC_INTERRUPTS` enabled (correct?)

That is indeed correct.

> I suppose the point here is to be able to change IRQ priorities, without enabling this Kconfig, which affects memory footprint and run-time performance.

Ok, that makes more sense. The rationale given above was "So one can get and set the priority of an interrupt without having to resort to arch-specific code," so I assumed the primary goal was to provide an arch-agnostic way of configuring IRQ priority, which is already supported.

I am still not sure how practical and/or useful this API would be; also whether this would be considered a good practice. The question I would like to ask is: when would one need to re-configure IRQ priority on-the-fly?

ioannisg, 2020-06-03:
> I am still not sure how practical and/or useful this API would be; also whether this would be considered a good practice. The question I would like to ask is: when would one need to re-configure IRQ priority on-the-fly (other than connecting a new dynamic IRQ, that is)?

i agree, @carlescufi needs to answer this question, since he has requested this enhancement :)

(My assumption is that you want to have as minimal overhead as possible (i.e. CONFIG_DYNAMIC_INTERRUPTS=n), use the standard ISR wrapper, and have the option to just change the IRQ priority on the fly)

IMO, adding an additional API, suitable only for rare use-cases, is not a big deal: as I said above, it feels more important to investigate whether the usage of such API will not have negative impact(s) to the kernel.

andrewboie, 2020-06-03:
I think it should be possible to reconfigure priority even for static build-time interrupts.

I'd need to look at the code further, but on x86 IIRC the priority is used to determine the position in the IDT for the installed interrupt, it's not handled by the APIC. It's been a long time since I looked at this in detail though (and I was never fully clear how it worked). But it should be possible to write a little code to move entries to different IDT vectors if we want to, but we'll have to make sure the IDT is in RAM and not ROM.

carlescufi, 2020-06-19:
> @stephanosio yeap, we've discussed that, too. I suppose @carlescufi would like to have this option for statically-configured IRQ lines (i.e. not dynamic), as well. This is, currently supported for [...]

That's correct @ioannisg. Interrupt priority in DT or in Kconfig or in the code should be the initial value, but why not be able to query it and modify it later?

gregshue, 2022-08-25:
> I am still not sure how practical and/or useful this API would be; also whether this would be considered a good practice. The question I would like to ask is: when would one need to re-configure IRQ priority on-the-fly (other than connecting a new dynamic IRQ, that is)?

I'm not sure what @carlescufi had in mind, but here is a practical use case:
- As a mechatronic device developer, I need to build one FW image that supports the series of prototypes along with the minor variants that exist, so that I can minimize the variability in the results of testing done by other engineers.
- As a result, the FW needs to determine the electro-mechanical prototype revision and variant at "start" time and adjust behavior as appropriate.  This adjustment may require lowering the interrupt priority of a hardware source for later revisions (e.g., EVT requires a servo to run at 4KHz, while DVT needs the servo to run at only 700Hz).  The electro-mechanical prototype revision may only be discoverable through exercising a mechanical sequence.

Now this introduces complications in non-interrupt code accessing data shared with the interrupt code.  In order to keep the periodic interrupt code running with minimal latency impact, the non-interrupt code must lock out all interrupts **at the priority of the periodic interrupt**, do the minimal necessary access, and restore the previous interrupt lockout level.  When the priority of the interrupt changes dynamically, it is more difficult to determine what the interrupt lockout priority needs to be.

The required logic is a bit more complicated than described above, but it points out the critical constraints.

andyross, 2022-09-19:
Just saw this come by.  FWIW: the reason this API doesn't exist is because the concept isn't easily portable.  Every architecture does priority assignment differently, and most interrupt controller drivers have their own schemes that would need to be integrated.  Some systems like Xtensa have hard-wired priorities that can't be configured at all (though in many cases the systems then include a second level interrupt controller that does have features like this).  Systems like x86 encode the priority in the vector number, so an API like the one requested would have to decouple vectors from "interrupts" in a recoverable way (i.e. have some kind of internal table to do the mapping).  It gets even weirder for setups like MSI, where the "priority" is encoded in the **target** vector and thus involves the driver generating the interrupt (i.e. if you want to change your USB xhci interrupt priority, you need to talk to the xhci controller!).

Add to that the fact that most systems are able to emulate part of this (via masking off interrupts at shared priorities inside the ISR so that they can only interrupt each other in known ways) and things get harder still.  All that code has size and performance overhead.  Do we enable it for all these platforms just because some of them want runtime-mutable interrupts?

I'm not saying that no clean API is possible, just that it requires a bunch of thought that no one has attempted yet.  And IMHO it doesn't seem like a huge hardship to have platform-local APIs for problems like this.  Interrupt priority tuning is always going to be low-level and probably device-specific anyway.