zephyrproject-rtos/zephyr #25918: qemu_nios2 crash when enabled icount

wentongwu, 2020-06-03:
seems when qemu runes instruction WRCTL of nios2 arch with icount enabled, there will be crash.

wentongwu, 2020-06-03:
```
diff --git a/home/wentongw/tmp/qemu-4.2.0/hw/nios2/cpu_pic.c b/./hw/nios2/cpu_pic.c
index 1c1989d..ada8d5b 100644
--- a/home/wentongw/tmp/qemu-4.2.0/hw/nios2/cpu_pic.c
+++ b/./hw/nios2/cpu_pic.c
@@ -21,6 +21,7 @@
 #include "qemu/osdep.h"
 #include "cpu.h"
 #include "hw/irq.h"
+#include "sysemu/cpus.h"
 
 #include "qemu/config-file.h"
 
@@ -54,9 +55,11 @@ static void nios2_pic_cpu_handler(void *opaque, int irq, int level)
 
 void nios2_check_interrupts(CPUNios2State *env)
 {
-    if (env->irq_pending) {
-        env->irq_pending = 0;
-        cpu_interrupt(env_cpu(env), CPU_INTERRUPT_HARD);
+    if (!use_icount) {
+        if (env->irq_pending) {
+            env->irq_pending = 0;
+            cpu_interrupt(env_cpu(env), CPU_INTERRUPT_HARD);
+        }
     }
 }
```

experimental patch deferring interrupt handle to cpu_exec seems work, need more debug.

```
int cpu_exec(CPUState *cpu)
{
    CPUClass *cc = CPU_GET_CLASS(cpu);
    int ret;
    SyncClocks sc = { 0 };

    /* replay_interrupt may need current_cpu */
    current_cpu = cpu;

    if (cpu_handle_halt(cpu)) {
        return EXCP_HALTED;
    }

    rcu_read_lock();

    cc->cpu_exec_enter(cpu);

    /* Calculate difference between guest clock and host clock.
     * This delay includes the delay of the last cycle, so
     * what we have to do is sleep until it is 0. As for the
     * advance/delay we gain here, we try to fix it next time.
     */
    init_delay_params(&sc, cpu);

    /* prepare setjmp context for exception handling */
    if (sigsetjmp(cpu->jmp_env, 0) != 0) {
#if defined(__clang__) || !QEMU_GNUC_PREREQ(4, 6)
        /* Some compilers wrongly smash all local variables after
         * siglongjmp. There were bug reports for gcc 4.5.0 and clang.
         * Reload essential local variables here for those compilers.
         * Newer versions of gcc would complain about this code (-Wclobbered). */
        cpu = current_cpu;
        cc = CPU_GET_CLASS(cpu);
#else /* buggy compiler */
        /* Assert that the compiler does not smash local variables. */
        g_assert(cpu == current_cpu);
        g_assert(cc == CPU_GET_CLASS(cpu));
#endif /* buggy compiler */
#ifndef CONFIG_SOFTMMU
        tcg_debug_assert(!have_mmap_lock());
#endif
        if (qemu_mutex_iothread_locked()) {
            qemu_mutex_unlock_iothread();
        }
        qemu_plugin_disable_mem_helpers(cpu);

        assert_no_pages_locked();
    }

    /* if an exception is pending, we execute it here */
    while (!cpu_handle_exception(cpu, &ret)) {
        TranslationBlock *last_tb = NULL;
        int tb_exit = 0;

        while (!cpu_handle_interrupt(cpu, &last_tb)) {
            uint32_t cflags = cpu->cflags_next_tb;
            TranslationBlock *tb;
```

wentongwu, 2020-06-03:
@andrewboie

wentongwu, 2020-06-04:
```
(gdb) info breakpoints
Num     Type           Disp Enb Address    What
1       breakpoint     keep y   0x00426408 in z_set_timeout_expiry at /home/wentongw/zephyr/zephyrproject/zephyr/include/arch/nios2/arch.h:52
	breakpoint already hit 2 times
2       breakpoint     keep y   0x00426408 in z_set_timeout_expiry at /home/wentongw/zephyr/zephyrproject/zephyr/include/arch/nios2/arch.h:52
	breakpoint already hit 1 time
(gdb) si
0x0042640c	52		__asm__ volatile (
(gdb) si
0x00426410	52		__asm__ volatile (
(gdb) 
0x00426414	52		__asm__ volatile (
(gdb) 
0x00426418	52		__asm__ volatile (
(gdb) 
0x0042641c	52		__asm__ volatile (
(gdb) 
0x00426420	52		__asm__ volatile (
(gdb) 
0x00426424 in arch_irq_lock () at /home/wentongw/zephyr/zephyrproject/zephyr/include/arch/nios2/arch.h:52
52		__asm__ volatile (
(gdb) 
0x00426428	52		__asm__ volatile (
(gdb) 
0x0042642c	52		__asm__ volatile (
(gdb) 
0x00426430	52		__asm__ volatile (
(gdb) 
Remote connection closed
```
@andrewboie @andyross @nashif after wrctl instruction qemu(icount enabled) will abort itself, but first I wonder if there is other way to lock irq on nios2 arch? Thanks

```
void z_set_timeout_expiry(s32_t ticks, bool idle)
{
  426408:       defffc04        addi    sp,sp,-16
  42640c:       dc400115        stw     r17,4(sp)
  426410:       dc000015        stw     r16,0(sp)
  426414:       dfc00315        stw     ra,12(sp)
  426418:       dc800215        stw     r18,8(sp)
  42641c:       2021883a        mov     r16,r4
  426420:       2823883a        mov     r17,r5
        __asm__ volatile (
  426424:       0025303a        rdctl   r18,status
  426428:       00bfff84        movi    r2,-2
  42642c:       9084703a        and     r2,r18,r2
  426430:       1001703a        wrctl   status,r2
        __ASSERT(z_spin_lock_valid(l), "Recursive spinlock %p", l);
  426434:       d1201504        addi    r4,gp,-32684

```

```
*** Booting Zephyr OS build zephyr-v2.1.0-5757-g394ff7a28b6b  ***
Running test suite test_fifo_timeout
===================================================================
starting test - test_timeout_empty_fifo
qemu: fatal: Raised interrupt while not in I/O function
IN: PC=426430 z_set_timeout_expiry
     zero=00000000        at=00000000        r2=00000000        r3=00400b6c 
       r4=00000002        r5=00000001        r6=00000000        r7=00000000 
       r8=00000000        r9=00000000       r10=00400c94       r11=00400c00 
      r12=00000000       r13=00000000       r14=00000000       r15=00000000 
      r16=00000002       r17=00000001       r18=00000000       r19=00000000 
      r20=00000000       r21=00000000       r22=00000000       r23=00000000 
       et=00000000        bt=00000000        gp=004083d0        sp=004035a8 
       fp=00000000        ea=00000000        ba=00000000        ra=00426958 
   status=00000000   estatus=00000000   bstatus=00000000   ienable=00000004 
 ipending=00000004     cpuid=00000000 reserved0=00000000 exception=00000000 
  pteaddr=00000000    tlbacc=00000000   tlbmisc=00000000 reserved1=00000000 
  badaddr=00000000    config=00000000   mpubase=00000000    mpuacc=00000000 
reserved2=00000000 reserved3=00000000 reserved4=00000000 reserved5=00000000 
reserved6=00000000 reserved7=00000000 reserved8=00000000 reserved9=00000000 
reserved10=00000000 reserved11=00000000 reserved12=00000000 reserved13=00000000 
reserved14=00000000 reserved15=00000000 reserved16=00000000 reserved17=00000000 
      rpc=00426430  mmu write: VPN=00000 PID 00 TLBACC 00000000

Aborted (core dumped)
```

wentongwu, 2020-06-04:
checked nios2 arch doc, the behavior locking irq is correct in zephyr code.

wentongwu, 2020-06-05:
will communicate below patch with qemu community
```
diff --git a/hw/nios2/cpu_pic.c b/hw/nios2/cpu_pic.c
index 1c1989d5..b04db4d7 100644
--- a/hw/nios2/cpu_pic.clog_core/src/log_core_test.c" 544L, 13181C                                                                                                                               247,8-15      44%
+++ b/hw/nios2/cpu_pic.c
@@ -42,7 +42,9 @@ static void nios2_pic_cpu_handler(void *opaque, int irq, int level)
         } else if (!level) {
             env->irq_pending = 0;
             cpu_reset_interrupt(cs, type);
-        }
+        } else {
+            cs->interrupt_request |= type;
+       }
     } else {
         if (level) {
             cpu_interrupt(cs, type);
```

wentongwu, 2020-07-13:
patch has been merged to qemu master, 5.1 version of which will include them.

wentongwu, 2020-07-21:
@galak any plan to upgrade zephyr sdk qemu based on qemu 5.1? Thanks

stephanosio, 2020-07-23:
> @galak any plan to upgrade zephyr sdk qemu based on qemu 5.1? Thanks

@wentongwu QEMU 5.1 will (probably) be released on 18 Aug (see https://wiki.qemu.org/Planning/5.1).

wentongwu, 2020-07-23:
@stephanosio thanks, not sure Zephyr's release plan, but if next Zephyr SDK release(probably 0.11.5) doesn't follow QEMU 5.1, I think we need cherry-pick above patches...

wentongwu, 2020-09-21:
@galak just see Zephyr SDK has already upgraded QEMU to 5.1 version, but is there any plan to do a new release for Zephyr SDK recently? Thanks a lot！

wentongwu, 2020-09-25:
@galak ping.

wentongwu, 2020-09-28:
@galak @nashif @andrewboie ping

PR #27123 (2021-01-26): boards: nios2: enable icount for qemu_nios2 platform
Enable icount for qemu_nios2 platform.

Fixes: #25918.
commit 27a43808 (2021-01-26): boards: nios2: enable icount for qemu_nios2 platform

Enable icount for qemu_nios2 platform.

Fixes: #25918.
closed 2021-01-26 by nashif