zephyrproject-rtos/zephyr #25919: dhcpv4 or rx ethernet packets not working on nucleo_f429zi

Blobby19, 2020-06-03:
Hi,
I flashed dhcpv4_client sample application on my nucleo_f429zi board and unfortunately the board is not able to get I address from my DHCP server.

**Describe the bug**
I made wireshark analysis, and see that the board is able to send dhcp_discover packets, and get a response from DHCP Server with a dhcp_offer packet, but the board seems to ignore them.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_f429zi samples/net/dhcpv4_client/

**Expected behavior**
The board should get an IP address from dhcp server.

**Screenshots or console output**
```
uart:~$ *** Booting Zephyr OS build zephyr-v2.2.0-1897-gd59ac4f0d50a  ***
[00:00:01.517,000] <dbg> eth_stm32_hal.eth_initialize: MAC 02:80:e1:d8:b3:a2
[00:00:01.517,000] <dbg> net_if.net_if_init: (0x20003ce0): 
[00:00:01.517,000] <dbg> net_if.init_iface: (0x20003ce0): On iface 0x2000aca0
[00:00:01.517,000] <dbg> net_if.net_if_post_init: (0x20003ce0): 
[00:00:01.517,000] <dbg> net_dhcpv4.net_dhcpv4_init: (0x20003ce0): 
[00:00:01.523,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:01.523,000] <dbg> net_dhcpv4.net_dhcpv4_start: (0x20003ce0): iface 0x2000aca0 state=init
[00:00:01.523,000] <dbg> net_dhcpv4.net_dhcpv4_start: (0x20003ce0): wait timeout=7s
[00:00:02.010,000] <dbg> net_if.net_if_up: (0x20003d9c): iface 0x2000aca0
[00:00:02.010,000] <dbg> net_dhcpv4.dhcpv4_iface_event_handler: (0x2000371c): Interface 0x2000aca0 coming up
[00:00:02.010,000] <dbg> net_dhcpv4.dhcph4_manage_timers: (0x20003d9c): iface 0x2000aca0 state=init
[00:00:02.010,000] <dbg> net_dhcpv4.dhcpv4_enter_selecting: (0x20003d9c): enter state=selecting
[00:00:02.010,000] <dbg> net_dhcpv4.dhcpv4_send_discover: (0x20003d9c): send discover xid=0xb7d0fdf5 timeout=4s
[00:00:02.010,000] <dbg> net_dhcpv4.dhcpv4_timeout: (0x20003d9c): Waiting for 4s
[00:00:02.010,000] <dbg> net_if.net_if_tx: (0x20003a6c): Processing (pkt 0x20007700, prio 0) network packet iface 0x2000aca0/1
[00:00:06.010,000] <dbg> net_dhcpv4.dhcph4_manage_timers: (0x20003d9c): iface 0x2000aca0 state=selecting
[00:00:06.010,000] <dbg> net_dhcpv4.dhcpv4_send_discover: (0x20003d9c): send discover xid=0xb7d0fdf6 timeout=8s
[00:00:06.010,000] <dbg> net_dhcpv4.dhcpv4_timeout: (0x20003d9c): Waiting for 8s
[00:00:06.010,000] <dbg> net_if.net_if_tx: (0x20003a6c): Processing (pkt 0x20007700, prio 0) network packet iface 0x2000aca0/1
```
[image]

**Environment (please complete the following information):**
 - OS: Debian 10
 - Toolchain: Zephyr SDK
 - 2.2.99

jukkar, 2020-06-03:
I do not have exactly that hw, but tried this in nucleo_f767zi. The dhcpv4_sample client worked for me in Zephyr OS build v2.3.0-rc1-422-g76d8f8f1e268 just fine.

Blobby19, 2020-06-03:
Hi Jukkar,
Thanks a lot for your help, it's working using the version `v2.3.0-rc1-422-g76d8f8f1e268`.
```
uart:~$ *** Booting Zephyr OS build v2.3.0-rc1-422-g76d8f8f1e268  ***
[00:00:01.607,000] <dbg> eth_stm32_hal.eth_initialize: MAC 02:80:e1:49:12:d5
[00:00:01.607,000] <dbg> net_if.net_if_init: (0x200047e4): 
[00:00:01.607,000] <dbg> net_if.init_iface: (0x200047e4): On iface 0x20000a80
[00:00:01.607,000] <dbg> net_if.net_if_post_init: (0x200047e4): 
[00:00:01.607,000] <dbg> net_dhcpv4.net_dhcpv4_init: (0x200047e4): 
[00:00:01.613,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:01.613,000] <dbg> net_dhcpv4.net_dhcpv4_start: (0x200047e4): iface 0x20000a80 state=init
[00:00:01.613,000] <dbg> net_dhcpv4.net_dhcpv4_start: (0x200047e4): wait timeout=6s
[00:00:02.029,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -62
[00:00:02.029,000] <dbg> net_if.net_if_up: (0x200048a8): iface 0x20000a80
[00:00:02.029,000] <dbg> net_dhcpv4.dhcpv4_iface_event_handler: (0x20004210): Interface 0x20000a80 coming up
[00:00:02.029,000] <dbg> net_dhcpv4.dhcph4_manage_timers: (0x200048a8): iface 0x20000a80 state=init
[00:00:02.029,000] <dbg> net_dhcpv4.dhcpv4_enter_selecting: (0x200048a8): enter state=selecting
[00:00:02.029,000] <dbg> net_dhcpv4.dhcpv4_send_discover: (0x200048a8): send discover xid=0x6f558966 timeout=4s
[00:00:02.029,000] <dbg> net_dhcpv4.dhcpv4_timeout: (0x200048a8): Waiting for 4s
[00:00:02.029,000] <dbg> net_if.net_if_tx: (0x20004570): Processing (pkt 0x20008208, prio 0) network packet iface 0x20000a80/1
[00:00:05.038,000] <dbg> net_dhcpv4.net_dhcpv4_input: (0x200044dc): Received dhcp msg [op=0x2 htype=0x1 hlen=6 xid=0x6f558966 secs=0 flags=0x80 chaddr=02:80:E1:49:12:D5
[00:00:05.038,000] <dbg> net_dhcpv4.net_dhcpv4_input: (0x200044dc):   ciaddr=0.0.0.0
--- 19 messages dropped ---
[00:00:05.038,000] <dbg> net_dhcpv4.net_dhcpv4_input: (0x200044dc):   yiaddr=192.168.1.74
[00:00:05.038,000] <dbg> net_dhcpv4.net_dhcpv4_input: (0x200044dc):   siaddr=192.168.1.254
[00:00:05.038,000] <dbg> net_dhcpv4.net_dhcpv4_input: (0x200044dc):   giaddr=0.0.0.0]
[00:00:05.038,000] <dbg> net_dhcpv4.dhcpv4_parse_options: (0x200044dc): options_server_id: 192.168.1.254
[00:00:05.038,000] <dbg> net_dhcpv4.dhcpv4_parse_options: (0x200044dc): options_lease_time: 86400
[00:00:05.038,000] <dbg> net_dhcpv4.dhcpv4_parse_options: (0x200044dc): options_renewal: 43200
[00:00:05.038,000] <dbg> net_dhcpv4.dhcpv4_parse_options: (0x200044dc): options_rebinding: 75600
[00:00:05.038,000] <dbg> net_dhcpv4.dhcpv4_parse_options: (0x200044dc): option unknown: 28
[00:00:05.038,000] <dbg> net_dhcpv4.dhcpv4_parse_options: (0x200044dc): option unknown: 6
[00:00:05.038,000] <dbg> net_dhcpv4.dhcpv4_parse_options: (0x200044dc): options_router: 192.168.1.254
[00:00:05.043,000] <dbg> net_dhcpv4.dhcpv4_parse_options: (0x200044dc): option unknown: 15
[00:00:05.043,000] <dbg> net_dhcpv4.dhcpv4_parse_options: (0x200044dc): options_router: <log_strdup alloc failed>
[00:00:05.043,000] <dbg> net_dhcpv4.dhcpv4_parse_options: (0x200044dc): options_subnet_mask <log_strdup alloc failed>
[00:00:05.043,000] <dbg> net_dhcpv4.dhcpv4_parse_options: (0x200044dc): options_end
[00:00:05.043,000] <dbg> net_dhcpv4.dhcpv4_handle_reply: (0x200044dc): state=requesting msg=ack
[00:00:05.043,000] <inf> net_dhcpv4: Received: <log_strdup alloc failed>
[00:00:05.043,000] <dbg> net_if.net_if_ipv4_addr_add: (0x200044dc): [0] interface 0x20000a80 address <log_strdup alloc failed> type DHCP added
[00:00:05.043,000] <dbg> net_dhcpv4.dhcpv4_enter_bound: (0x200044dc): enter state=bound renewal=43200s rebinding=75600s
[00:00:05.043,000] <inf> net_dhcpv4_client_sample: Your address: 192.168.1.74
[00:00:05.043,000] <inf> net_dhcpv4_client_sample: Lease time: 86400 seconds
[00:00:05.043,000] <inf> net_dhcpv4_client_sample: Subnet: <log_strdup alloc failed>
[00:00:05.043,000] <inf> net_dhcpv4_client_sample: Router: <log_strdup alloc failed>
[00:00:06.029,000] <dbg> net_dhcpv4.dhcph4_manage_timers: (0x200048a8): iface 0x20000a80 state=bound
```

chrta, 2020-06-22:
I tested on the same board on today's master revision 72ebf70543fca78740888bc58319a19c2a3d636b at it worked (at least for my dhcp server):
```
$ west build -p auto -b nucleo_f429zi samples/net/dhcpv4_client/

...

*** Booting Zephyr OS build zephyr-v2.3.0-499-g72ebf70543fc  ***
[00:00:03.903,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:04.902,000] <inf> net_dhcpv4: Received: 192.168.170.175
[00:00:04.902,000] <inf> net_dhcpv4_client_sample: Your address: 192.168.170.175
[00:00:04.902,000] <inf> net_dhcpv4_client_sample: Lease time: 600 seconds
[00:00:04.902,000] <inf> net_dhcpv4_client_sample: Subnet: 255.255.255.0
[00:00:04.902,000] <inf> net_dhcpv4_client_sample: Router: 192.168.170.1
```

erwango, 2020-08-10:
@Blobby19 can it be closed now, as you reported it was working ?

erwango, 2020-09-14:
Closing as no response for more than one month

closed 2020-09-14 by erwango