zephyrproject-rtos/zephyr #25920: Compilation error when CONFIG_BOOTLOADER_MCUBOOT=y specified

amarkevich, 2020-06-03:
**Describe the bug**
Compilation error when CONFIG_BOOTLOADER_MCUBOOT=y specified

**To Reproduce**
Steps to reproduce the behavior:
1. go to zephyr\samples\subsys\nvs 
2. add to prj.conf
CONFIG_BOOTLOADER_MCUBOOT=y
3. build the project

**Expected behavior**
build finished without errors

**Impact**
unable to compile the project

**Console output**
```
1> D:/nfr/v1.3.0-rc1/zephyr/include/storage/flash_map.h:235:11: error: 'PM_PM_PM_storage_ID_LABEL_OFFSET' undeclared (first use in this function)
1> D:/nfr/v1.3.0-rc1/zephyr/include/sys/util.h:510:36: note: in definition of macro 'UTIL_PRIMITIVE_CAT'
1> D:/nfr/v1.3.0-rc1/zephyr/include/storage/flash_map.h:235:2: note: in expansion of macro 'UTIL_CAT'
1> ../src/main.c:77:14: note: in expansion of macro 'FLASH_AREA_OFFSET'
1> D:/nfr/v1.3.0-rc1/zephyr/include/storage/flash_map.h:235:11: note: each undeclared identifier is reported only once for each function it appears in
1> D:/nfr/v1.3.0-rc1/zephyr/include/sys/util.h:510:36: note: in definition of macro 'UTIL_PRIMITIVE_CAT'
1> D:/nfr/v1.3.0-rc1/zephyr/include/storage/flash_map.h:235:2: note: in expansion of macro 'UTIL_CAT'
1> ../src/main.c:77:14: note: in expansion of macro 'FLASH_AREA_OFFSET'

```

**Environment (please complete the following information):**
 - OS: Windows 10 Enterprise (10.0.18363)
 - Toolchain nRF Connect SDK v1.3.0-rc1
 - Version 2.3.0 rc1

carlescufi, 2020-06-03:
@amarkevich from what I can see you are using NCS, so the proper forum for questions about it is the Nordic [devzone](https://devzone.nordicsemi.com/)

carlescufi, 2020-06-03:
That said, if @nvlsianpu wants to comment on this issue, he can still do it while it is still closed.

closed 2020-06-03 by carlescufi

nvlsianpu, 2020-06-03:
@amarkevich This will be fixed in coming NCS 1.3.0 relese

nvlsianpu, 2020-06-05:
@amarkevich Fixed in NCS by https://github.com/nrfconnect/sdk-nrf/pull/2402

amarkevich, 2020-06-05:
@nvlsianpu Thanks for notification!