zephyrproject-rtos/zephyr #25925: tests: net: socket: socketpair: fails due to empty message header name

ABOSTM, 2020-06-03:
**Describe the bug**
Running test tests/net/socket/socketpair/ fails.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p -b nucleo_f429zi tests/net/socket/socketpair/
2. west flash
3. See error

**Expected behavior**
test passed

**Screenshots or console output**
```
Running test suite socketpair
===================================================================
starting test - test_socketpair_AF_LOCAL__SOCK_STREAM__0
E: syscall z_user_alloc_from_copy failed check: Memory region 0x00000000 (size 0) read access denied

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/socketpair/src/test_socketpair_happy_path.c:141: happy_path: (res equal to -1)
sendmsg(2) failed: 12
FAIL - test_socketpair_AF_LOCAL__SOCK_STREAM__0
```

**Analysis**
This is a regression introduced with PR #25706
Due to new check on messages parameters, test fails because in function `happy_path()` , 
`msghdr.msg_name`  is not set and thus equal 0  (and the same for msg_namelen)

But in net_ip.h, in the structure comment, it seems this name is optional:
```
struct msghdr {
	void         *msg_name;       /* optional socket address */
	socklen_t     msg_namelen;    /* size of socket address */
...
```
So I don't know whether
*  test should be updated and structure comment ?

or
* PR #25706 should be reworked ?

Note: probably the same will happend with `msg_control` field of the structure

ABOSTM, 2020-06-03:
^^ @jukkar

jukkar, 2020-06-03:
>     * PR #25706 should be reworked ?

As those couple of fields are optional for connected socket, the checks introduced in #25706 need to be fixed. Will send a fix asap.

jukkar, 2020-06-03:
I just wonder why this issue was not caught by sanitychecker.

commit d3dde1ad (2020-06-03): net: sockets: Fix sendmsg() user mode param checks for dst address

If we are calling sendmsg() for a connected socket, then msg_namelen
is 0 and msg_name is NULL. Check these allowed values properly.

Also modify unit tests so that we test this scenario.

Fixes #25925
closed 2020-06-03 by carlescufi