zephyrproject-rtos/zephyr #25926: API:k_cycle_get_32() can't get value in specific platform

Zhaoningx, 2020-06-03:
Describe the bug
seems when API:k_cycle_get_32() runes platforms of native_posix, there will just get return value 0.
and it's OK to run at others platforms.

Running at qemu_x86 platform result as follow:
[image]

Running at native_posix platform result as follow:
[image]

Implement Code:
`	/*for-loop to get some average operations time spent*/

	for (i = 0; i < const_times; i++) {
		start_time =  k_cycle_get_32();
		printk("start_time1 = %u\n", start_time);
		rb_get_min(&test_tree1);
		start_time = k_cycle_get_32() - start_time;
		spent_time[0][0] += start_time;

		start_time =  k_cycle_get_32();
		printk("start_time2 = %u\n", start_time);
		rb_get_max(&test_tree1);
		start_time = k_cycle_get_32() - start_time;
		spent_time[0][1] += start_time;

		start_time = k_cycle_get_32();
		printk("start_time3 = %u\n", start_time);
		rb_insert(&test_tree1, &tx_node[0]);
		start_time = k_cycle_get_32() - start_time;
		spent_time[0][2] += start_time;

		start_time = k_cycle_get_32();
		printk("start_time4 = %u\n", start_time);
		rb_remove(&test_tree1, &tx_node[0]);
		start_time = k_cycle_get_32() - start_time;
		spent_time[0][3] += start_time;
	}`

Configuration files:
[image]

and by the way, it's unstable on both platforms of qemu_nois2 and qemu_x86_64.

aescolar, 2020-06-03:
@Zhaoningx Most likely what you are seeing is that while the CPU does not go to sleep, time does not pass in native_posix:
https://docs.zephyrproject.org/latest/boards/posix/native_posix/doc/index.html#important-limitations

I'm guessing the reason why you are getting 0 out of `k_cycle_get_32()` is because you are running that code right after boot and before having ever set the CPU to sleep.

If that is the case, that is not a bug. If you call `k_sleep()`, `k_busy_wait()` or in any other way perform some operation which does make that thread wait for some time, you should see that `k_cycle_get_32()` returns you a later time.

Please tell if this clarified the issue for you, and if that is the case, close this issue.

Zhaoningx, 2020-06-04:
Hi @aescolar, thanks for your review, I have read zephyr doc about native_posix carefully, and I test the condition you mentioned, and I used k_msleep() before invoking the API k_cycle_get_32, as you say, it can returns a later time instead. But, there are something confused me, can I get the time of running several lines code in native_posix without CPU sleeping?
And I'll close this issue later.

aescolar, 2020-06-04:
>  can I get the time of running several lines code in native_posix without CPU sleeping?

Hi @Zhaoningx . No unfortunately that is not possible in native_posix due to fundamental reasons: Code itself takes "0" time to execute.

youhuazhu, 2020-06-05:
Hi @aescolar, for native posix part,   Can we get cpu cycle with `rdtsc` like below:

```
static inline uint64_t get_cycle_count()
{
unsigned int lo,hi;
__asm__ __volatile__ ("rdtsc" : "=a" (lo), "=d" (hi));
return ((uint64_t)hi << 32) | lo;
}
```

andyross, 2020-06-05:
@youhuazhu  - Yes, fundamentally your process is x86_64 linux userspace code and it can execute the rdtsc instruction and it will return a cycle count the reflects real world time.  But you still can't spin on that to "wait" for anything to happen in the Zephyr native_posix implementation.  That's not how it works.

What, exactly, is the problem you are trying to solve?

youhuazhu, 2020-06-05:
oh,  yes, we can get the native host cycel count from that way,  you are right,  this cpu cycle not only for Zephyr native_posix implemention, so not accurate, I tried this.

and up to now, I have no good way to solve this issue,  
I think we can close this **question**.  @Zhaoningx

Zhaoningx, 2020-06-05:
hi, @aescolar  @youhuazhu @andyross thanks, I know that and close it now.

closed 2020-06-05 by Zhaoningx

aescolar, 2020-06-05:
@youhuazhu @Zhaoningx 
If  what you are trying to do is profile your code (when compiled for your x86 host), you can use normal desktop tools. For example, a relatively simple (FOSS) tool you can use is valgrind's callgrind.
```
valgrind --tool=callgrind --dump-instr=yes --collect-jumps=yes ./zephyr/zephyr.exe
kcachegrind callgrind.out.<pid> &
```