zephyrproject-rtos/zephyr #25942: Bluetooth: Scanning + Non-connectable advertising broken on nRF5340

trond-snekvik, 2020-06-03:
Non-connectable advertising fails if scanning is already enabled on the nRF5340. It does not provide any error messages, but advertising never starts. Removing the call to `bt_le_scan_start` fixes the issue. Changing to connectable advertising also fixes the issue. The problem is the same regardless of whether advertising or scanning is started first.

The behavior is only observable on the open source split LL Nordic controller, not the Nordic proprietary controller. 

**To Reproduce**
Flash samples/bluetooth/scan_adv on the app core (`nrf5340pdk_nrf5340_cpuapp`) and samples/bluetooth/hci_rpmsg on the net core (`nrf5340pdk_nrf5340_cpunet`). The device is supposed to advertise with a 100ms interval, but doesn't. Remove the call to `bt_le_scan_start`, and it starts working. The expected behavior can be observed if the sample is flashed on an nRF52832 DK.

Tested on latest master and v2.2.0 tag. Same behavior on both.

**Impact**
This is a showstopper for the Bluetooth Mesh on nRF5340, which relies heavily on these two features being combined.

trond-snekvik, 2020-06-03:
cc @carlescufi @joerchan

cvinayak, 2020-06-06:
Root cause and fix mentioned in #26018.

PR #26018 (2020-06-08): Bluetooth: controller: Address nRF5340 Engineering A Errata 16
Address the nRF5340 Engineering A Errata 16, RADIO: POWER
register is not functional.

This affects the dependency of controller's HAL
implementation that expected correct reset values being
set in the Radio peripheral.

As the SUBSCRIBE_TXEN and SUBSCRIBE_RXEN where not in their
reset value, Radio transmission and reception start using
DPPI was broken.

Fixes #25942.
commit 3cabacce (2020-06-08): Bluetooth: controller: Address nRF5340 Engineering A Errata 16

Address the nRF5340 Engineering A Errata 16, RADIO: POWER
register is not functional.

This affects the dependency of controller's HAL
implementation that expected correct reset values being
set in the Radio peripheral.

As the SUBSCRIBE_TXEN and SUBSCRIBE_RXEN where not in their
reset value, Radio transmission and reception start using
DPPI was broken.

Fixes #25942.
closed 2020-06-08 by carlescufi