zephyrproject-rtos/zephyr #25945: devicetree: support generating symbols for -gpios properties w/o compatible

pabigot, 2020-06-03:
A recent change to devicetree tooling allowed generation of nodes for which there was no `compatible`, but that generation did not include any properties on such nodes (which I still think is weird).

Do we have enough special-casing in the tooling to support something like:
```
/ {
  app {
    power-gpios = <&gpioa 5 GPIO_ACTIVE_HIGH>;
  };
};
```
then look it up by `DT_PATH(app, power_gpios)`?  At least for GPIO we can reasonably say that if the property ends in `-gpios` we can assume it's a GPIO phandle+specifier.  @galak?

_Originally posted by @pabigot in https://github.com/zephyrproject-rtos/zephyr/pull/25596_

pabigot, 2020-06-03:
From https://github.com/zephyrproject-rtos/zephyr/pull/24769#pullrequestreview-402765100 re converting other properties where the corresponding representation can be inferred w/o recourse to a binding would be useful: 
I added these to `/`:

```
       special = "this";
       subspecial {
          bar = "that";
       };
```

The subordinate node was generated, but neither `/special` nor `/subspecial/bar` were generated

Is that what was intended?  Generating the node without its properties seems odd; will this satisfy the need identified in https://github.com/zephyrproject-rtos/zephyr/pull/24460#discussion_r414774455

pabigot, 2020-06-17:
Also needed for things like:

    servo-pwms = <&pwm0 17>;

per a slack discussion.

sslupsky, 2020-07-01:
Yeah, I ran into this issue a while ago too.  This would be useful.

pabigot, 2020-07-20:
This keeps coming up.  @galak Do you have fundamental objections to this that will prevent us from implementing it, or can we proceed?

@mbolivar-nordic It would probably be more efficient for you to do this, but if you're fully booked I'm willing to take it on.

mbolivar-nordic, 2020-07-22:
I don't think I'll have cycles for this anytime soon

carlescufi, 2020-08-04:
API meeting 4th August 2020

- The functionality in itself is welcome by most
- The namespace is however under debate:
  - Arbitrary namespace (`app`. `mygpios`, etc): not gathering any support
  - Fixed namespace (`zephyr-aux`, `zephyr-app`, `zephyr-user`). The tooling can then limit the generation of these (nodes without a compatible) when they are under the fixed namespace: `/zephyr,user/<user-chosen-name>-{gpios, pwms, ..} `
- The use of these nodes should be limited to samples/tests but not drivers
- The use of `chosen` for this does not seem to match the original intent of `chosen` in Zephyr (limited set of known chosen nodes)

carlescufi, 2020-08-04:
This proposal would effectively replace the one in #18544

PR #27937 (2020-09-04): edtlib: support inferring binding from node content
Clean up of devicetree tooling removed generation of information present in devicetree in nodes that have no `compatible`, or for extra properties not defined by a binding.  Discussion proposed that these properties should be allowed, but only in a defined node `/zephyr,user`. For that node infer bindings based on the presence of properties.

Fixes: #25945
Supersedes: #18544, #15020
closed 2020-09-04 by galak