zephyrproject-rtos/zephyr #25952: STM32 LPTIM driver doesn't restart counter after sleeping K_TICKS_FOREVER

aurel32, 2020-06-03:
**Describe the bug**
The `z_clock_set_timeout()` function of the STM32 LPTIM driver checks if it is being passed `K_TICKS_FOREVER` and in that case just resets and stops the LPTIM. However passing such a value doesn't mean the system has to halt forever. The system might just wait for an interrupt for an undetermined amount of time, for example the end of a I2C or SPI transfer. It means that the next time `z_clock_set_timeout()` is called, the LPTIM should be re-enabled and reconfigured. Otherwise the system just hang.

Note that it is a minor issue as the scheduler only passes `K_TICKS_FOREVER` in case `CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE=y` (which is not the default), otherwise INT_MAX is passed instead.

**To Reproduce**
Build `samples/basic/blinky` on a board supporting LPTIM with the following options:
```
CONFIG_DEVICE_POWER_MANAGEMENT=y
CONFIG_STM32_LPTIM_TIMER=y
CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE=n
```
Check that the LED blinks correctly.

Then set `CONFIG_SYSTEM_CLOCK_SLOPPY_IDLE=y`. The LED doesn't blink anymore as the system just hangs.

aurel32, 2020-06-04:
> In case the timeout is set 'FOREVER', the system stops the lptim clock and sleeps until another wakeUp source comes. In the blinky application, there is no such a wakeUp source, like user button.

No there is no need for another wake-up source in the blinky case. The timeout being set to 'FOREVER' doesn't mean the CPU can sleep. It just mean that for now the timer is not needed anymore.

In the blinky case, it happens that after the first executed `k_sleep` the scheduler calls `z_clock_set_timeout` with `K_TICKS_FOREVER` as there is no scheduled event anymore. Then the GPIO is toggled, and the next `k_sleep` is executed with the 1s sleep time, causing `z_clock_set_timeout` to be called with the number of ticks corresponding to 1 s.

Blinky is a very simple case, what could happen in more complex code is that the CPU in a busy loop waiting for a serial, I2C or SPI transfer to finish.

FRASTM, 2020-06-05:
My understanding is : once the 'scheduler calls z_clock_set_timeout with K_TICKS_FOREVER as there is no scheduled event anymore'  consequently there is no next k_sleep. 
So the GPIO is not toggling anymore. In the blinky application, no event or interrupt is likely to restore the scheduling.

aurel32, 2020-06-05:
> My understanding is : once the 'scheduler calls z_clock_set_timeout with K_TICKS_FOREVER as there is no scheduled event anymore'

I agree

> consequently there is no next k_sleep.
> So the GPIO is not toggling anymore. In the blinky application, no event or interrupt is likely to restore the scheduling.

I disagree, at least when `z_clock_set_timeout` is passed `idle = false`. There is no event schedule anymore, but that might change a few CPU cycles later, as `K_TICKS_FOREVER` doesn't imply that the CPU is idle. From what I understand, this is actually determined by the `idle` boolean value passed to `z_clock_set_timeout`.