zephyrproject-rtos/zephyr #25958: Concept Overview for improving support for serial flash devices via SPI and QSPI

pabigot, 2020-06-04:
It's becoming increasingly clear that the old theory that we should provide SPI NOR flash parameters via devicetree is becoming unsustainable; cf  #23322, #22965, and some slack conversations about newer chips with much larger sizes.  That original policy was based on a perception that runtime extraction of device parameters was too costly in flash and RAM.

While one approach is prototyped in #25806 I was looking for something that's a little more comprehensive, including being prepared to support other parameter blocks.  This issue provides a retroactive RFC for how to support a wider range of serial flash devices by using the JEDEC Serial Flash Discoverable Parameters tables that can be read from JESD216-compatible devices, including most SPI NOR serial flashes supported by the Zephyr spi-nor flash driver and other in-tree and pending flash interfaces (primarily QSPI).

The general concept is:

* Provide data structures and decoders that operate on JESD216 headers and parameter tables.
* Use [`jedec,spi-nor`](https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/mtd/jedec%2Cspi-nor.txt) as the compatible, with an extension to define the SFDP Basic Flash Parameters via an `sfdp-bfp` property for cases where runtime determination of characteristics is not desired.
* Follow Linux examples ([stm32](https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/spi/st%2Cstm32-qspi.yaml), [freescale](https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/spi/spi-fsl-qspi.txt)) for describing controllers.  Note that these bindings may be associated with Zephyr flash devices rather than SPI devices, and that the child nodes should be `jedec,spi-nor` compatibles (with additional properties using a more specific compatible that extends `jedec,spi-nor`).
* Provide a way to re-use in utility applications the SFDP and JEDEC ID functions implemented by specific drivers.  A specific application is one that reads a device and generates the `sfdp-bfp` and `jedec-id` property values.
* Where the current bindings provide parameters that can be obtained from SFDP the corresponding properties should be deprecated/removed and the SFDP encodings used.
* Where driver-specific devicetree properties are required those may be added via a compatible that extends the `jedec-spi-common.yaml` binding set.

The information from SFDP enables a lot of other potential enhancements including better control of deep power down and use of timing information to reduce time spent busy-waiting for operations to complete.

These APIs and functionality are prototoyped in #25851, currently only for the legacy SPI NOR driver.  As we identify functional commonalities, such as identifying erase commands for a specific offset and span, these can be pulled out into a shared implementation for use by specific drivers.

pabigot, 2020-06-04:
@mnkp @arvid-r @galak @rckTom

mnkp, 2020-06-04:
> * Use [`jedec,spi-nor`](https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/mtd/jedec%2Cspi-nor.txt) as the compatible, with an extension to define the SFDP Basic Flash [...]

The device tree specification states:
_The compatible property value consists of one or more strings that define the specific programming model for the device. This list of strings should be used by a client program for device driver selection. The property value consists of a concatenated list of null terminated strings, from most specific to most general. They allow a device to express its compatibility with a family of similar devices, potentially allowing a single device driver to match against several devices._

The compatible property is used to associate DTS node with the device driver. While we don't use this mechanism in Zephyr (yet), we shouldn't purposefully violate the specification. Using `compatible = "jedec,spi-nor"` for flash devices attached to QSPI controller would be fine if the drivers/flash/spi_nor.c was able to handle QSPI devices.

In Linux the spi-nor driver is able to control both SPI and QSPI flash devices and they consistently use jedec,spi-nor compatible for the flash nodes.

pabigot, 2020-06-04:
> The compatible property is used to associate DTS node with the device driver. While we don't use this mechanism in Zephyr (yet), we shouldn't purposefully violate the specification. Using [...]

That isn't how I understand what Linux does.  In the [stm32](https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/spi/st%2Cstm32-qspi.yaml) and [freescale](https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/spi/spi-fsl-qspi.txt) it isn't the generic spi-nor driver that is used for those controllers.  That's exactly the situation we have in Zephyr.

Linux is [pretty clear](https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/mtd/jedec%2Cspi-nor.txt#L9) that *a* compatible for the flash memory itself, regardless of its controller, should be `jedec,spi-nor`.  If necessary

galak, 2020-06-04:
Dev-review (Jun 4, 2020): Discussed how we should represent this in DTS, and semantic differences about programming model vs driver.  For now we should have at least `jedec,spi-nor` on any flash device node that conforms to that programming model.  In practice the combination of the `jedec,spi-nor` node and its parent are what really tell us how or which driver should come into play.

galak, 2020-06-04:
We need to look at implementing improvements on DT_CHILD_* related to this functionality.

arvid-r, 2020-06-04:
How does this relate to the chosen nodes in Zephyr?
Do we need both `zephyr, flash` and `zephyr, flash-controller`?

pabigot, 2020-06-04:
Personally I've never knowingly made use of either of those chosen properties.

At least one use of them in the tree seems to be wrong, as it assumes a partition is on the "chosen" controller, which is unreliable.  The proper controller can be found from the partition node itself (now, at least, maybe it couldn't in legacy devicetree).

arvid-r, 2020-06-05:
Yes, exactly. This caused much confusion when I started to look at this. If I remember correctly, it is MCUBOOT that uses the `zephyr, flash-controller` part. It should probably be changed to grab the information it needs through the partitions instead, as it won't need to do anything to the controller itself I guess.

closed 2020-08-17 by nashif