zephyrproject-rtos/zephyr #25960: tests/net/socket/socketpair failed on mimxrt1050_evk and sam_e70_xplained.

chen-png, 2020-06-04:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b sam_e70_xplained -p auto tests/net/socket/socketpair
2. west flash
3. See error

**Screenshots or console output**
*** Booting Zephyr OS build v2.3.0-rc1-422-g76d8f8f1e268  ***
Running test suite socketpair

**starting test - test_socketpair_AF_LOCAL__SOCK_STREAM__0
E: syscall z_user_alloc_from_copy failed check: Memory region 0x00000000 (size 0) read access denied
Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/socketpair/src/test_socketpair_happy_path.c:141: happy_path: (res equal to -1)
sendmsg(2) failed: 12**
FAIL - test_socketpair_AF_LOCAL__SOCK_STREAM__0

starting test - test_socketpair_AF_UNIX__SOCK_STREAM__0
E: syscall z_user_alloc_from_copy failed check: Memory region 0x00000000 (size 0) read access denied
Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/socketpair/src/test_socketpair_happy_path.c:141: happy_path: (res equal to -1)
sendmsg(2) failed: 12
FAIL - test_socketpair_AF_UNIX__SOCK_STREAM__0

starting test - test_socketpair_expected_failures
E: syscall z_vrfy_zsock_socketpair failed check: Memory region 0x00000000 (size 8) write access denied
PASS - test_socketpair_expected_failures

starting test - test_socketpair_unsupported_calls
PASS - test_socketpair_unsupported_calls
...
starting test - test_socketpair_poll_immediate_data
PASS - test_socketpair_poll_immediate_data

starting test - test_socketpair_poll_delayed_data
PASS - test_socketpair_poll_delayed_data

starting test - test_socketpair_poll_close_remote_end_POLLIN
PASS - test_socketpair_poll_close_remote_end_POLLIN

starting test - test_socketpair_poll_close_remote_end_POLLOUT
PASS - test_socketpair_poll_close_remote_end_POLLOUT

Test suite socketpair failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.11.3
 - Commit ID: 76d8f8f1e

jukkar, 2020-06-04:
Duplicate of #25925

jukkar, 2020-06-04:
Please use 2.3.0-rc2 for testing, this issue is fixed there.

closed 2020-06-04 by jukkar