zephyrproject-rtos/zephyr #25964: Bluetooth: <err> bt_att: ATT Timeout

cvinayak, 2020-06-04:
**Describe the bug**
Local device performing ATT write cmds, generates `<wrn> bt_att: Ignoring unexpected request` when peer device send an ATT request. The peer then gets `<err> bt_att: ATT timeout`

**To Reproduce**
Use the branch: https://github.com/cvinayak/zephyr/commit/b4939f526a9c1c86a88fd55810ed85e1ecaaf46d

Steps to reproduce the behavior:
Build and flash peripheral:
  1. mkdir -p build/peripheral; cd build/peripheral
  2. cmake -GNinja -DBOARD=nrf52dk_nrf52832 ../../samples/bluetooth/peripheral
  3. ninja
  4. ninja flash

Build and flash central_hr:
  5. mkdir -p build/central_hr; cd build/central_hr
  6. cmake -GNinja -DBOARD=nrf52dk_nrf52832 ../../samples/bluetooth/central_hr
  7. ninja
  8. ninja flash

  9. Open two terminals, `minicom -D /dev/ttyACMx`
  10. Observe both peripheral and central_hr fail with respective warning and error messages in the console.

**Expected behavior**
ATT request from peer device shall not be dropped. ATT bearer shall not timeout.

**Impact**
showstopper

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
- OS: Linux
 - Toolchain: Zephyr SDK 0.11.2
 - Commit SHA or Version used: https://github.com/cvinayak/zephyr/commit/b4939f526a9c1c86a88fd55810ed85e1ecaaf46d

**Additional context**
Add any other context about the problem here.

alexsven, 2020-06-04:
I'm seeing the same thing for our application. When using bt_gatt_unsubscribe() we get "bt_att: ATT timeout" on one side and "bt_att: Ignoring unexpected request" on the other. That also means we can not subscribe again, so we need to reset the kits before subscribing. This happened after updating to v2.3.0-rc1 from v2.2.0

JordanYates, 2020-06-06:
I am seeing what I believe is the same issue.
The problem occurs when two transmissions of different types (REQUEST, CONFIRMATION, etc) on the same ATT channel occur within a very short window.
In my case at least, this occurs when the GATT server publishes an Indication, which the host schedules an acknowledgement for. If the GATT client is also performing writes to the server, two writes can go through the host in a very short window.

https://github.com/zephyrproject-rtos/zephyr/blob/e77985a2f0531a8fbefdb9ae2e5ea7c5e3dbd70f/subsys/bluetooth/host/att.c#L152
This line sets the ATT callback, `chan_cfm_sent`, `chan_rsp_sent`, etc.
However there is no protection around this setting. If `chan_send` is called again before `bt_att_sent` runs, the original callback is lost, and the callback associated with the second transmission is instead run twice.

In my situation, because the callback is lost, `chan_cfm_sent` is never called and thus the `ATT_PENDING_CFM` bit is never cleared on the flags. The next indication from the server is therefore ignored at https://github.com/zephyrproject-rtos/zephyr/blob/e77985a2f0531a8fbefdb9ae2e5ea7c5e3dbd70f/subsys/bluetooth/host/att.c#L2309

Vudentz, 2020-06-08:
We will probably need a list/queue so that we can process the sent event in order.

joerchan, 2020-06-08:
@Vudentz This is looking like a regression, no-one has reported it before. Any ideas on why it would need a queue/list now when we didn't before?

Vudentz, 2020-06-09:
> @Vudentz This is looking like a regression, no-one has reported it before. Any ideas on why it would need a queue/list now when we didn't before?

This was introduced with EATT support since bt_l2cap_chan_send cannot rely on callbacks like bt_l2cap_send_cb, anyway I working on a fix for this.

Counterfeiter, 2020-06-09:
back to v2.2.0 works for me...
Fix 26063 on v2.3.0 works for me also...

Thanks!

joerchan, 2020-06-09:
With the propopsed fix from 26063 and the steps in this bug-report:

The warning is gone: `<wrn> bt_att: Ignoring unexpected request when peer device send an ATT request.`

The error is still there: `<err> bt_att: ATT timeout`

joerchan, 2020-06-09:
@JordanYates and @Counterfeiter Thanks for reporting issue and status. I have created a new bug-report for the problem that you guys saw. Which appears to be a different issue than the ATT timeout seen here.

Did you guys see an ATT timeout, and has it been fixed? Remember to wait the ATT timeout, which is 30 seconds.

JordanYates, 2020-06-09:
@joerchan I may be wrong, but I believe it is the same issue, or rather, that this issue is caused by the lost callbacks.
When the `<wrn> bt_att: Ignoring unexpected request` occurs, no response is sent back to peer device. This will cause ATT timeouts.
Either way, the linked pull request will resolve my issue.

joerchan, 2020-06-09:
@JordanYates They are definitely two separate issues. Your issue may result in ATT timeout as well.
Can you check if your issue results in ATT timeout and update the new issue I created if that is the case?

joerchan, 2020-06-09:
@Vudentz 

The issue is caused by ATT not sending the responses to Service discovery requestse.
In vinayaks setup the peripheral is sending write-requests as fast as possible, while the central is doing device-discovery and sending notifications as fast as possible.
In this case the service discovery requests by the central is what is timing out. This is because the peripheral fails to queue the response packets, and simply drop them.
This occurs here:
```
static u8_t att_find_type_rsp(struct bt_att_chan *chan, u16_t start_handle,
			      u16_t end_handle, const void *value,
			      u8_t value_len)
{
	/* ... */

	int send_err = bt_att_chan_send(chan, data.buf, chan_rsp_sent);
	if (send_err) {
		BT_WARN("%d send_err %d", __LINE__, send_err);
	}

	return 0;
}
```

The error code is -11, which is it appears is coming from the  `ATT_PENDING_SENT` now.
But could also be the tx_sem.

mfiumara, 2020-06-09:
Not sure if relevant but I see simialr same behavior on 2.2.1 when connecting to a peripheral using nrfconnect:
```
[00:24:57.234,680] <wrn> bt_ctlr_hci: type: 0x14
[00:24:57.332,244] <wrn> bt_att: Unhandled ATT code 0x1d
[00:24:57.430,297] <wrn> bt_att: Ignoring unexpected request
```
Then after 30 seconds the peripheral disconnects (I suspect the ATT time-out).

mhx, 2020-06-11:
I've tried again, rebasing #26063 onto the latest master, just to be sure, and this time immediately got it to reproduce (albeit still with the 50ms connection interval, but I've been using that for the last month without any issues):

nRF Connect: [2020-06-11T07_04_33.079Z-log.txt](https://github.com/zephyrproject-rtos/zephyr/files/4763245/2020-06-11T07_04_33.079Z-log.txt)
Zephyr: [2020-06-11T07_04_33.079Z-zephyr.log](https://github.com/zephyrproject-rtos/zephyr/files/4763246/2020-06-11T07_04_33.079Z-zephyr.log)

Procedure was as decribed above:

- Connect to peripheral
- Enter passkey
- Update connection interval to 50ms
- Sequentially enable notifications for all 4 attributes
- Upon enabling the last attribute's notifications (seen in the Zephyr log in line `[00:00:45.571,411] <inf> main: Notification enabled`), updates from the peripheral stop and after half a minute, there's a GATT timeout in nRF Connect
- A disconnect seems to be the only way to interact with the peripheral

joerchan, 2020-06-12:
@mhx The procedure you described, is this with the peripheral sample without modifications? If you have modifications would you be able to share these so that we can reproduce the same scenario as you?

Counterfeiter, 2020-06-12:
I still facing problems with v2.3.0 and Fix 26063, if I disable the logging.

@joerchan The problem to reproduce this seems to be the timing. I have to disable the whole logging to see connection problems with some iOS Apps.

But I am very new to zephyr, so I haven't the deeper understanding to help you out.

joerchan, 2020-06-12:
@mhx and @Counterfeiter Yeah, I'm also seeing issues. Can you enable CONFIG_ASSERT=y, see if we all get `ASSERTION FAIL [buf] @ WEST_TOPDIR/zephyr/subsys/net/buf.c:574`

mfiumara, 2020-06-12:
@joerchan  I have seen the same problem on 2.2.0. I enabled `CONFIG_ASSERT=y` but I get an assertion somewhere else: k_panic when I try to perform an image upload using mcumgr:
```
0x0003ad42 in assert_post_action (file=0x4 "5'", file@entry=0x49123 "WEST_TOPDIR/zephyr/kernel/include/ksched.h", 
    line=line@entry=262) at /home/mattia/bitbucket/node_firmware_v9/zephyr/lib/os/assert.c:45
45              k_panic();
(gdb) thread apply all bt

Thread 1 (Thread 57005):
#0  0x0003ad42 in assert_post_action (file=0x4 "5'", file@entry=0x49123 "WEST_TOPDIR/zephyr/kernel/include/ksched.h", line=line@entry=262) at /home/mattia/bitbucket/node_firmware_v9/zephyr/lib/os/assert.c:45
#1  0x00034a56 in z_sched_lock () at /home/mattia/bitbucket/node_firmware_v9/zephyr/kernel/include/ksched.h:262
#2  z_impl_k_mutex_unlock (mutex=mutex@entry=0x20009b8c <rtt_term_mutex>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/kernel/mutex.c:232
#3  0x00018f58 in k_mutex_unlock (mutex=<optimized out>) at zephyr/include/generated/syscalls/kernel.h:657
#4  data_out_block_mode (data=0x20002992 <char_buf> "e 0x10)\033[0m\r\ntyp\001", length=0, ctx=<optimized out>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/subsys/logging/log_backend_rtt.c:214
#5  0x0003c22e in buffer_write (outf=0x18f1d <data_out_block_mode>, buf=0x20002992 <char_buf> "e 0x10)\033[0m\r\ntyp\001", len=0, ctx=0x0) at /home/mattia/bitbucket/node_firmware_v9/zephyr/subsys/logging/log_output.c:149
#6  0x0003c502 in log_output_flush (log_output=log_output@entry=0x4233c <log_output>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/subsys/logging/log_output.c:158
#7  0x00018ec4 in log_backend_std_panic (log_output=0x4233c <log_output>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/subsys/logging/log_backend_std.h:51
#8  panic (backend=<optimized out>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/subsys/logging/log_backend_rtt.c:260
#9  0x00018844 in log_backend_panic (backend=0x41d6c <log_backend_rtt>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/include/logging/log_backend.h:178
#10 z_impl_log_panic () at /home/mattia/bitbucket/node_firmware_v9/zephyr/subsys/logging/log_core.c:576
#11 0x00033eac in log_panic () at zephyr/include/generated/syscalls/log_ctrl.h:34
#12 k_sys_fatal_error_handler (reason=reason@entry=0, esf=esf@entry=0x20006308 <_interrupt_stack+1784>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/kernel/fatal.c:41
#13 0x00033f5c in z_fatal_error (reason=reason@entry=0, esf=esf@entry=0x20006308 <_interrupt_stack+1784>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/kernel/fatal.c:123
#14 0x0001acb4 in z_arm_fatal_error (reason=reason@entry=0, esf=esf@entry=0x20006308 <_interrupt_stack+1784>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/arch/arm/core/aarch32/fatal.c:47
#15 0x0001b17a in z_arm_fault (msp=<optimized out>, psp=<optimized out>, exc_return=<optimized out>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:968
#16 0x0001ac58 in z_arm_usage_fault () at /home/mattia/bitbucket/node_firmware_v9/zephyr/arch/arm/core/aarch32/fault_s.S:108
#17 <signal handler called>
#18 assert_post_action (file=0x4 "5'", file@entry=0x49123 "WEST_TOPDIR/zephyr/kernel/include/ksched.h", line=line@entry=262) at /home/mattia/bitbucket/node_firmware_v9/zephyr/lib/os/assert.c:45
#19 0x00034a56 in z_sched_lock () at /home/mattia/bitbucket/node_firmware_v9/zephyr/kernel/include/ksched.h:262
#20 z_impl_k_mutex_unlock (mutex=mutex@entry=0x20009b8c <rtt_term_mutex>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/kernel/mutex.c:232
#21 0x00018f58 in k_mutex_unlock (mutex=<optimized out>) at zephyr/include/generated/syscalls/kernel.h:657
#22 data_out_block_mode (data=0x20002992 <char_buf> "e 0x10)\033[0m\r\ntyp\001", length=0, ctx=<optimized out>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/subsys/logging/log_backend_rtt.c:214
#23 0x0003c22e in buffer_write (outf=0x18f1d <data_out_block_mode>, buf=0x20002992 <char_buf> "e 0x10)\033[0m\r\ntyp\001", len=0, ctx=0x0) at /home/mattia/bitbucket/node_firmware_v9/zephyr/subsys/logging/log_output.c:149
#24 0x0003c502 in log_output_flush (log_output=log_output@entry=0x4233c <log_output>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/subsys/logging/log_output.c:158
#25 0x00018ec4 in log_backend_std_panic (log_output=0x4233c <log_output>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/subsys/logging/log_backend_std.h:51
#26 panic (backend=<optimized out>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/subsys/logging/log_backend_rtt.c:260
#27 0x00018844 in log_backend_panic (backend=0x41d6c <log_backend_rtt>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/include/logging/log_backend.h:178
#28 z_impl_log_panic () at /home/mattia/bitbucket/node_firmware_v9/zephyr/subsys/logging/log_core.c:576
#29 0x00033eac in log_panic () at zephyr/include/generated/syscalls/log_ctrl.h:34
#30 k_sys_fatal_error_handler (reason=reason@entry=4, esf=esf@entry=0x20003a10 <_k_thread_stack_sensing_thread_tid+1128>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/kernel/fatal.c:41
#31 0x00033f5c in z_fatal_error (reason=4, esf=0x20003a10 <_k_thread_stack_sensing_thread_tid+1128>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/kernel/fatal.c:123
#32 0x0001aa66 in _oops () at /home/mattia/bitbucket/node_firmware_v9/zephyr/arch/arm/core/aarch32/swap_helper.S:462
#33 <signal handler called>
[... cut ...]
#36 0x00035ab6 in z_thread_single_abort (thread=0x35cb7 <pend+274>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/kernel/sched.c:475
#37 0x00037110 in k_spin_lock (l=0x20002880 <timeout_lock>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/include/spinlock.h:85
#38 z_clock_announce (ticks=<optimized out>) at /home/mattia/bitbucket/node_firmware_v9/zephyr/kernel/timeout.c:205
#39 0x00037210 in z_impl_k_timer_start (timer=0x0, duration=-536810240, period=536880316) at /home/mattia/bitbucket/node_firmware_v9/zephyr/kernel/timer.c:110
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

joerchan, 2020-06-12:
@mfiumara If I'm reading that stack-trace right it appears the ASSERT came from `zephyr_i2c_read` and then got a usage-fault in the `assert_post_action_handler`.
The fact that you get a similar ATT timeout  on 2.2.0 must mean that it is a different problem, because this one is a regression since 2.2.0.

Can you create a separate bug-report for it?

mhx, 2020-06-14:
> @mhx The procedure you described, is this with the peripheral sample without modifications? If you have modifications would you be able to share these so that we can reproduce the same scenario as you?

@joerchan No, the sample has quite a few modifications and it's rather specific to the custom board it's running on (e.g. attribute updates are triggered by the light sensor interrupts). I could try to narrow it down to a smaller, board independent piece of code, but it's unlikely that I'll find the time to do this in the next two weeks unfortunately.

I'll definitely try `CONFIG_ASSERT=y`, though.

mhx, 2020-06-14:
No luck with `CONFIG_ASSERT=y`. I can still reproduce the problem, but there are no assertions in the log.

Vudentz, 2020-06-15:
[00:02:03.218,658] [1;31m<err> bt_conn: not connected![0m
[00:02:03.218,688] [0m<dbg> bt_att.chan_send: code 0x1b[0m

This has nothing to do with the initial issue it seems, it appears to get disconnected in the meantime.

mhx, 2020-06-16:
> [00:02:03.218,658] �[1;31m bt_conn: not connected!�[0m
> [00:02:03.218,688] �[0m bt_att.chan_send: code 0x1b�[0m
> 
> This has nothing to do with the initial issue it seems, it appears to get disconnected in the meantime.

That was a deliberate disconnect I triggered from the nRF Connect app. Before the disconnect, nothing happened for more than one minute:

    [00:00:45.571,411] [0m<dbg> bt_att.chan_send: code 0x13[0m
    [00:02:03.218,627] [0m<dbg> bt_att.att_sent: conn 0x2000135c chan 0x20003084[0m

mhx, 2020-06-17:
Okay, I actually have a repro now with just very basic changes to the code in `samples/bluetooth/peripheral` that at least for me produces the issue 100% of the time. I've also confirmed that the code works fine with zephyr @ d148f8648b.

The change I've made is that, in order to simulate the interrupt-driven updates in my application, I've added a timer that triggers `interrupt_worker`. `interrupt_worker` then performs the attribute updates and re-arms the timer to trigger again after 100ms. This is pretty much what's happening in my application, only that instead of a timer trigger, I'm using an external interrupt that triggers after 100ms.

The issue appears to be that the code ends up in a state where `interrupt_worker` actually gets stuck while performing the attribute updates. I don't actually know where exactly it gets stuck, though, as I can't find a call stack that includes the `interrupt_worker` (probably gdb doesn't understand zephyr threads). But it can clearly be seen in the console that `interrupt_worker` is called, but never returns.

Here's the procedure to reproduce the failure:
- apply [this diff](https://gist.github.com/mhx/694d73b8e6a144a0bbea25e7d15e56f0) to `samples/bluetooth/peripheral` and flash to peripheral; I've repro'd this with an `adafruit_feather_nrf52840` board
- connect to peripheral from nRF Connect
- update connection interval to 50ms
- enable notifications for Battery Level, Current Time and Heart Rate Measurement
- usually at this point, the console logging will stop and after half a minute, nRF Connect will show a Gatt timeout

carlescufi, 2020-06-17:
> Okay, I actually have a repro now with just very basic changes to the code in `samples/bluetooth/peripheral` that at least for me produces the issue 100% of the time. I've also confirmed that the code works fine with zephyr @ [d148f86](https://github.com/zephyrproject-rtos/zephyr/commit/d148f8648b4f9a06ee888221609c568919770599).
> 
> The change I've made is that, in order to simulate the interrupt-driven updates in my application, I've added a timer that triggers `interrupt_worker`. `interrupt_worker` then performs the attribute updates and re-arms the timer to trigger again after 100ms. This is pretty much what's happening in my application, only that instead of a timer trigger, I'm using an external interrupt that triggers after 100ms.
> 
> The issue appears to be that the code ends up in a state where `interrupt_worker` actually gets stuck while performing the attribute updates. I don't actually know where exactly it gets stuck, though, as I can't find a call stack that includes the `interrupt_worker` (probably gdb doesn't understand zephyr threads). But it can clearly be seen in the console that `interrupt_worker` is called, but never returns.
> 
> Here's the procedure to reproduce the failure:
> 
> * apply [this diff](https://gist.github.com/mhx/694d73b8e6a144a0bbea25e7d15e56f0) to `samples/bluetooth/peripheral` and flash to peripheral; I've repro'd this with an `adafruit_feather_nrf52840` board
> * connect to peripheral from nRF Connect
> * update connection interval to 50ms
> * enable notifications for Battery Level, Current Time and Heart Rate Measurement
> * usually at this point, the console logging will stop and after half a minute, nRF Connect will show a Gatt timeout

Does https://github.com/zephyrproject-rtos/zephyr/pull/26063 fix the issue @mhx?

mhx, 2020-06-17:
> Does #26063 fix the issue @mhx?

Ah, I didn't see there was an additional change added to #26063 recently. No, this new change actually makes things *a lot* worse. When connecting to the peripheral from nRF Connect, it never gets to the point where I actually see the attribute tree. Without any interaction, I'm getting a Gatt timeout after half a minute. I'm seeing

    [00:00:03.662,231] <wrn> bt_att: Ignoring unexpected request

in the log when attempting to connect.

carlescufi, 2020-06-17:
@mhx yep, that is exactly what I see as well with the latest commit.

Vudentz, 2020-06-18:
Ive update the set, it is now possible to flood the stack with write without response like before and as a bonus I adds support for doing it using the shell.

mhx, 2020-06-18:
> Ive update the set, it is now possible to flood the stack with write without response like before and as a bonus I adds support for doing it using the shell.

This looks really good! I'm unable to reproduce the problem with the latest changes. I saw a bus fault once, but can't reproduce it:

    [00:00:38.866,516] <inf> hrs: HRS notifications enabled
    [00:00:38.922,058] <err> bt_conn: Unable to allocate TX context
    [00:00:38.924,499] <err> os: ***** BUS FAULT *****
    [00:00:38.924,499] <err> os:   Imprecise data bus error
    [00:00:38.924,499] <err> os: r0/a1:  0xfffffffc  r1/a2:  0x0000000c  r2/a3:  0x00000008
    [00:00:38.924,499] <err> os: r3/a4:  0x20005a60 r12/ip:  0x00000000 r14/lr:  0x0001f11b
    [00:00:38.924,530] <err> os:  xpsr:  0x21000000
    [00:00:38.924,530] <err> os: Faulting instruction address (r15/pc): 0x0001f120
    [00:00:38.924,530] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
    [00:00:38.924,530] <err> os: Current thread: 0x20002794 (unknown)
    [00:00:38.994,171] <err> os: Halting system

joerchan, 2020-06-18:
@Vudentz It's looking good now. Throughput sample is working well. Also tested the fixes on top of vinayaks original bug-report branch (https://github.com/joerchan/zephyr/tree/att-timeout) and it is also working without issues now.

carlescufi, 2020-06-18:
> This looks really good! I'm unable to reproduce the problem with the latest changes. I saw a bus fault once, but can't reproduce it:

No, that doesn't seem related. We have a couple of deadlocks that need fixing (they have PRs) which might fix the issue you are seeing.

mhx, 2020-06-18:
> > This looks really good! I'm unable to reproduce the problem with the latest changes. I saw a bus fault once, but can't reproduce it:
> 
> No, that doesn't seem related. We have a couple of deadlocks that need fixing (they have PRs) which might fix the issue you are seeing.

Yep, don't think it's related. The new code has been running for hours now without any issues.

PR #26063 (2020-06-18): Bluetooth: ATT: Multiple fixes
ATT channel sent callback shall not be overwritting until the
operation completes as it can result in breaking flow control when
CONFIG_BT_ATT_ENFORCE_FLOW is enabled.

Fixes #25964
Fixes #26070
Fixes #26071
commit 4418ba76 (2020-06-18): Bluetooth: ATT: Fix overwritting sent callback

ATT channel sent callback shall not be overwritting until the
operation completes as it can result in breaking flow control when
CONFIG_BT_ATT_ENFORCE_FLOW is enabled.

Fixes #25964
Fixes #26071
PR #26431 (2020-11-16): [v2.3-branch] Bluetooth: ATT: Multiple fixes
ATT channel sent callback shall not be overwritting until the
operation completes as it can result in breaking flow control when
CONFIG_BT_ATT_ENFORCE_FLOW is enabled.

Fixes #25964
Fixes #26070
Fixes #26071
commit 6821e0b4 (2020-11-16): Bluetooth: ATT: Fix overwritting sent callback

ATT channel sent callback shall not be overwritting until the
operation completes as it can result in breaking flow control when
CONFIG_BT_ATT_ENFORCE_FLOW is enabled.

Fixes #25964
Fixes #26071
closed 2020-06-18 by carlescufi