zephyrproject-rtos/zephyr #25965: HCI-UART not responding at higher baudrates on NRF52810

jjamesson, 2020-06-04:
Hi , Zephyr community,

I am having an issue with the HCI-UART example, where I am not getting a response from the controller for the HCI RESET command on higher UART bandwidth, namely with the "sudo btattach -B /dev/ttyACM1 -S 1000000 -P h4" command. However, I do get the response when using 115200 as the baudrate.

The problem exists with the "nrf52810_pca10040" west build, but the problem does NOT exist on the "nrf52_pca10040" west build (meant for using with the 52832 chip). The custom hardware we plan on deplaying hosts the 52810 controller, so this is an issue for us.

I have tried both the newest Zephyr and Zephyr 1.14 to solve this problem. 

Hope you can help. Feel free to ask additional information about this issue.

With regards,
James

carlescufi, 2020-06-04:
@jjamesson I assume you are using an nRF52 DK (PCA10040)?

jjamesson, 2020-06-04:
Yes, I am. Sorry for not mentioning.

carlescufi, 2020-06-04:
Alternatively, just change the baud rate directly to `1000000` here:
https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/nrf52dk_nrf52810/nrf52dk_nrf52810.dts#L95

jjamesson, 2020-06-05:
@carlescufi Indeed, I figured shortly after writing this post that there was no overlay file for the DK hosting the nrf52810 chip, and modifying the .dts file solved this issue. Thanks for looking into this. This topic can be closed. 

With regards,

carlescufi, 2020-06-06:
Thanks for letting us know @jjamesson

closed 2020-06-06 by carlescufi