zephyrproject-rtos/zephyr #25973: lib: posix: dynamic stack support for pthread_create()

cfriedt, 2020-06-04:
**Is your enhancement proposal related to a problem? Please describe.**
Currently, Zephyr's `pthread_create()` implementation requires that a `pthread_attr_t` be passed in whereas the POSIX specification explicitly allows `NULL`.

The rationale behind it is clear; in Zephyr, memory is a precious commodity, and we use the `pthread_attr_setstack()` function to attach a (typically statically allocated) stack to a specific `pthread_attr_t`.

For the majority of use cases, static stack allocation should continue to be the preferred approach. However, there are some specific uses cases where this might not be ideal. For example, supporting `std::thread` using libstdc++ and the [default gthr-posix configuration](https://github.com/gcc-mirror/gcc/blob/master/libgcc/gthr-posix.h#L663).

**Describe the solution you'd like**
I think it's possible to sort of have the best of both worlds.

Additional Kconfig values. E.g
```
CONFIG_PTHREAD_DYNAMIC_STACK=y
CONFIG_PTHREAD_DYNAMIC_STACK_SIZE=512
CONFIG_PTHREAD_MAX_DYNAMIC_STACKS=2
```
If `CONFIG_PTHREAD_DYNAMIC_STACK` is enabled, that number of "dynamic" stacks would be statically allocated, at compile time via `K_THREAD_STACK_ARRAY_DEFINE`. Thread permissions would be inherited from the calling thread and set with `k_object_access_grant()`.

Morever, the thread stack could be automatically recycled upon completion by specifying a `pthread_key_t` and "destructor" for the key, passing in a `dynamic_stack` argument to `zephyr_thread_wrapper()` [here](https://github.com/zephyrproject-rtos/zephyr/blob/master/lib/posix/pthread.c#L186), and then in `zephyr_thread_wrapper()`, `pthread_setspecific()` could be used to associate the dynamic stack with the `pthread_key_t`.

**Describe alternatives you've considered**
Initially, I had considered heap-allocation for stacks, but on devices with only 10s of kB's of SRAM, it might be a slippery-slope.

Using Kconfig values for the maximum number of dynamic stacks, the dynamic stack size, etc, push the responsibility of sensible memory management to the application.

The only downside to static allocation is that that memory is no longer available, generally, for the heap to use if no threads are using dynamic stacks.

Technically, it may also be possible to support a hybrid approach, with two different `pthread_key_t` and destructors, so that if the statically allocated dynamic stacks are exhausted, Zephyr could attempt to heap-allocate more. 

**Additional context**
I'm hoping to use this for #25572.
CC: @pfalcon, @pabigot

pfalcon, 2020-06-04:
@cfriedt: Are you aware of https://github.com/zephyrproject-rtos/zephyr/pull/16435 ?

cfriedt, 2020-06-04:
> @cfriedt: Are you aware of #16435 ?

Aha! I was not aware of it. Seems to be nearly complete. Looks like purely a heap implementation.

andrewboie, 2020-06-04:
> Initially, I had considered heap-allocation for stacks, but on devices with only 10s of kB's of SRAM, it might be a slippery-slope.

Not sure I understand the downsides of using a heap? You won't need to know how many stacks to reserve in the array in advance, nor would you have to come up with a CONFIG_PTHREAD_DYNAMIC_STACK_SIZE that works for all dynamic threads. It all depends on the app but threads tend to vary greatly in how much stack space they need.

> Obviously, it would also be possible to use entirely heap-allocated dynamic stacks as well.

This is almost possible, but not quite. If you wanted to go the heap route I think there are two things standing in the way of this:

- Stacks have specific alignment constraints. We need an allocator that can specify alignment. The old `k_mem_pool` couldn't do it. We have a new `k_heap` implementation, I remember @andyross told me it wouldn't be too hard to extend it to support aligned allocation.
- Need to know the alignment characteristics for stack objects of a given in a generic way. I'm working on a stack overhaul PR #24714 which adds generic macros to know how to align the base address and round up the size properly, so these should be available soon.

cfriedt, 2020-06-04:
> > Initially, I had considered heap-allocation for stacks, but on devices with only 10s of kB's of SRAM, it might be a slippery-slope.
> 
> Not sure I understand the downsides of using a heap? You won't need to know how many stacks to reserve in the array in advance, nor would you have to come up with a CONFIG_PTHREAD_DYNAMIC_STACK_SIZE that works for all dynamic threads. It all depends on the app but threads tend to vary greatly in how much stack space they need. Ultimately it's up to you though.

I would prefer to use the heap as well, the only downside is that if you *do* know the (maximum) 
 number of "dynamic" threads in advance, then you would still be subject to memory availability on the heap.

If we want to go the heap route, that's fine with me though.

cfriedt, 2020-10-09:
Really rough draft atm @andrewboie, using the kernel thread macros mentioned [in your comment here](https://github.com/zephyrproject-rtos/zephyr/issues/26999#issuecomment-673605373).

The PR that's linked to this has my changes.

```
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.4.0-278-g4709ac9f23d2  ***
Running test suite posix_apis
===================================================================
START - test_posix_pthread_execution
allocated 00001400@0x000000000011cb60
stack     00000400@0x000000000011d000
allocated 00001400@0x000000000011df78
stack     00000400@0x000000000011e000
allocated 00001400@0x000000000011f390
stack     00000400@0x0000000000120000
Thread 0 starting with scheduling policy 1 & priority 1
Thread 1 starting with scheduling policy 1 & priority 1
Thread 2 starting with scheduling policy 1 & priority 1
Bounce test OK
free   >= 00000400@0x000000000011f390
ASSERTION FAIL [z_spin_lock_valid(l)] @ ../include/spinlock.h:92
	Recursive spinlock 0x00000000001078cf
E: RAX: 0x0000000000000004 RBX: 0x0000000000000000 RCX: 0x000000000011c2c0 RDX: 0x000000000011c2c0
E: RSI: 0x000000000000005c RDI: 0x0000000000114d5f RBP: 0x0000000000163db0 RSP: 0x0000000000163db0
E:  R8: 0x0000000000000001  R9: 0x0000000000000030 R10: 0x0000000000000000 R11: 0x0000000000000000
E: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x0000000000163db0 RFLAGS: 0x0000000000000046 CS: 0x0018 CR3: 0x000000000016b000
E: call trace:
E: RIP: 0x0000000000103c86
E:      0x0000000000110f53
E:      0x00000000001110c7
E:      0x000000000011096a
E:      0x0000000000107f44
E:      0x000000000010c9e7
E:      0x0000000000110b03
E:      0x0000000000102aba
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x000000000011c2c0 (idle 00)
E: Halting system
```

PR #44727 (2023-07-24): lib: posix: dynamic stack support for pthread_create
Support automatic thread stack allocation for `pthread_create()`, which is required as part of IEEE Std. 1003.2017 (and all earlier versions).

For more info, see
https://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_create.html

Fixes #25973

(Note: This depends on #44379 and #59773)
closed 2023-07-24 by cfriedt