zephyrproject-rtos/zephyr #25979: Need root LICENSE files in hal_stm32 module

jettr, 2020-06-04:
**Describe the bug**
The https://github.com/zephyrproject-rtos/hal_stm32 repository does not have a root LICENSE file. Most, but not all of the source files within the repository have license information.

**Impact**
 This is blocker to locally replicating this repository for builds.

erwango, 2020-06-04:
@jettr, Txs for reporting this issue.
I'm not sure how we can fix this, more explicitly I'm not sure what should a root LICENSE file contain in case of a module like hal_stm32.

Here is the status of the package:
Most of the files (the ones coming from cube packages) are BSD-3 Licensed in this module, when this is not the case, they are:
-CMake related files (some of them being Apache 2.0, few not licensed)
-2 python scripts (ST Copyright, missing Apache 2.0 that should be added)
-README (not licensed)

What can be done easily is to update the CMake and python scripts to Apache 2.0.
I don't think there is a need to do anything on README.

Once this is done:
- is the root License file still required ?
- What could it mention knowing that the module contains mix of BSD3 and few Apache 2.0  ?

cc @MaureenHelm, as it seems more modules could be impacted

jettr, 2020-06-05:
I am talking with our internal teams on which potential solution would work best. I will let you know when I hear back. Thank you for responding so quickly!

MaureenHelm, 2020-06-05:
> cc @MaureenHelm, as it seems more modules could be impacted

@ioannisg Can you add this to the list of items to consider for module repos?

ioannisg, 2020-06-05:
@MaureenHelm I think we already have a discussion item for license files and license checks.
Will make sure we discuss this soon.

jettr, 2020-06-16:
Is there any update on this? This is blocking our forward progress on replicating our needed Zephyr dependencies and building it in an official way.

Do we have an estimate on when we think we can resolve this for the stm32 repository?

erwango, 2020-06-16:
@jettr sorry for the delay, we'll try to get this fixed by next week.

jettr, 2020-06-17:
Thank you! I really appreciate it!

jettr, 2020-06-25:
Do we have an idea of how we are going to fix this with a potential timeline? This really is blocking us.

erwango, 2020-06-25:
@jettr, please have a look to https://github.com/zephyrproject-rtos/hal_stm32/pull/61 and provide feedback

jettr, 2020-06-25:
I can't add my feedback on that pull request (I don't think I have permissions to leave a comment). It looks like it will be okay, but I am getting it reviewed internally just to make sure I am not missing anything. I will let you know soon. Thank you!

closed 2020-06-26 by galak