zephyrproject-rtos/zephyr #25981: Support storing mcuboot images on serial flash accessed through Nordic QSPI

pabigot, 2020-06-04:
The Nordic QSPI driver requires that all transfers be 32-bit aligned to/from SRAM, and that the transfer size be a multiple of 4 bytes.  This requirement is not satisfied by mcuboot/mcumgr which:
* need to be able to read and write sub-word lengths
* need to be able to write records that are stored in flash

This is a specific case of violations of #16439.

javier-moreno-tridonic-com, 2020-06-05:
Coming from here https://github.com/zephyrproject-rtos/zephyr/issues/25294

I'm also facing the issue of not being able to use properly the external flash on a nrf52840dk_nrf52840 board.

I've been using the modifications suggested in the mentioned link and thank to those, managed to get it to run partially.

I rewrote the partitions by using the following overlay. Idea behind it was to have as much internal flash as possible for the running image.

```
/delete-node/ &slot0_partition;
/delete-node/ &slot1_partition;
/delete-node/ &scratch_partition;

&flash0 {
    partitions {
        compatible = "fixed-partitions";
        #address-cells = <1>;
        #size-cells = <1>;

        slot0_partition: partition@c000 {
            label = "image-0";
            reg = <0x0000C000 0x0000EC000>;
        };
    };
};

&mx25r64 {
    partitions {
        compatible = "fixed-partitions";
        #address-cells = <1>;
        #size-cells = <1>;

        slot1_partition: partition@0 {
            label = "image-1";
            reg = <0x00000000 0x0000EC000>;
        };
        scratch_partition: partition@ec000 {
            label = "image-scratch";
            reg = <0x0000EC000 0x0000EC000>;
        };
    };
};
```

Also feed the following overlay-nrf52840dk_nrf52840-bootloader.conf to the mcuboot build as suggested

```
##########################################
### Configure required bootloader settings
##########################################
CONFIG_MULTITHREADING=y
CONFIG_BOOT_MAX_IMG_SECTORS=256
CONFIG_NORDIC_QSPI_NOR=y
CONFIG_NORDIC_QSPI_NOR_FLASH_LAYOUT_PAGE_SIZE=4096
```

I've prepared 2 signed images of our app using west, and shifted the one for the external flash to its virtual memory address (**0x12000000**  note that since the default app builds for slot 0 -0xC000- the difference is applied as offset -0x11FF4000-).

```
$ west sign -t imgtool -d build -- --version 0.0.1+0 --key ../applications/mcuboot/root-rsa-2048.pem
$ cp /build/zephyr.signed.hex /build/zephyr.signed.0.hex
$ west sign -t imgtool -d build -- --version 0.0.1+1 --key ../applications/mcuboot/root-rsa-2048.pem
$ cp /build/zephyr.signed.hex /build/zephyr.signed.1.hex
$ srec_cat zephyr.signed.1.hex -intel -offset 0x11FF4000 -o zephyr.signed.1.ext.hex -intel
```
Flashed everything in with nrjprog and dumped everything back also with nrjprog to ensure that the images are written and in the proper location.

In my case, the problem is that looks like slot 1 is completely ignored. This is the uart output I see:

```
uart:~$ *** Booting Zephyr OS build zephyr-v2.0.0-8127-g8a14d94a54ca ***
[00:00:00.004,699] <inf> mcuboot: Starting bootloader
[00:00:00.005,706] <inf> mcuboot: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.005,737] <inf> mcuboot: Boot source: primary slot
[00:00:00.010,162] <inf> mcuboot: Swap type: none

#### JUMP TO APP CODE HAPPENS HERE ####

[00:00:00.006,439] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
```

I can't see any reference in the output to slot1, also as far as I understand it, mcuboot should select the image in slot1 as it has the higher version number.

I saw new change suggestions on https://github.com/zephyrproject-rtos/zephyr/pull/25971#issuecomment-639017038 but as far as I understand it, the extra changes there should only affect mcumgr usage.

Is there anything I might be missing here? The only difference that I see with the mentioned tested and running scenario is the lack of usage on mcumgr and the fact that I also located the scratch partition in the external flash. Is mcumgr a must?

I've based everything on mcuboot 1.6.0 and current zephyr master.

javier-moreno-tridonic-com, 2020-07-27:
@pabigot Many thanks for the reply and sorry for the delay (due to corona times, working hours are... well weird to say the least). As far as I can tell, one coworker gave mcumgr a try through UDP and we got ourselves an update mechanism :)

pabigot, 2020-09-07:
AFAICT all functional changes necessary to make this work have been merged.  There's still a documentation gap, but that'd be a different issue.

closed 2020-09-07 by pabigot