zephyrproject-rtos/zephyr #25982: [Coverity CID :210020] Explicit null dereferenced in drivers/usb/device/usb_dc_mcux_ehci.c

zephyrbot, 2020-06-04:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/c61dfdc680615a1d636d1a676897815860ca4da5/drivers/usb/device/usb_dc_mcux_ehci.c#L210

Category: Null pointer dereferences
Function: `usb_dc_ep_configure`
Component: Drivers
CID: [210020](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210020)

Details:

```
204    
205         if (k_mem_pool_alloc(&ep_buf_pool, block, cfg->ep_mps, K_MSEC(10))) {
206             LOG_ERR("Memory allocation time-out");
207             return -ENOMEM;
208         }
209    
>>>     CID 210020:  Null pointer dereferences  (FORWARD_NULL)
>>>     Passing null pointer "(*block).data" to "memset", which dereferences it. [Note: The source code implementation of the function has been overridden by a builtin model.]
210         memset(block->data, 0, cfg->ep_mps);
211         dev_data.eps[ep_abs_idx].ep_mps = cfg->ep_mps;
212         status = dev_data.interface->deviceControl(dev_data.controllerHandle,
213                 kUSB_DeviceControlEndpointInit, &ep_init);
214         if (kStatus_USB_Success != status) {
215             LOG_ERR("Failed to initialize endpoint");
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jfischer-phytec-iot, 2020-06-05:
_k_mem_pool_alloc_ would allocate or fail, so I think it is false positive.

carlescufi, 2020-06-10:
Classified as false positive in Coverity

closed 2020-06-10 by carlescufi