zephyrproject-rtos/zephyr #25988: [Coverity CID :210687] Argument cannot be negative in tests/net/socket/socketpair/src/test_socketpair_happy_path.c

zephyrbot, 2020-06-04:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/c61dfdc680615a1d636d1a676897815860ca4da5/tests/net/socket/socketpair/src/test_socketpair_happy_path.c#L126

Category: Error handling issues
Function: `happy_path`
Component: Tests
CID: [210687](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=210687)

Details:

```
120                 NULL, &len);
121             zassert_true(res >= 0, "recvfrom(2) failed: %d", errno);
122             actual_msg_len = res;
123             zassert_equal(actual_msg_len, expected_msg_len,
124                       "wrong return value");
125    
>>>     CID 210687:  Error handling issues  (NEGATIVE_RETURNS)
>>>     "actual_msg_len" is passed to a parameter that cannot be negative. [Note: The source code implementation of the function has been overridden by a builtin model.]
126             zassert_true(strncmp(expected_msg, actual_msg,
127                 actual_msg_len) == 0,
128                 "the wrong message was passed through the socketpair");
129    
130             /*
131              * Test with sendmsg(2) / recv(2) - Zephyr lacks recvmsg atm
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

cfriedt, 2020-06-09:
There are a few false positives. This is one of them.

jukkar, 2020-06-09:
Ok, closing.

closed 2020-06-09 by jukkar