zephyrproject-rtos/zephyr #25990: tests/net/socket/select failed on sam_e70_xplained board.

chen-png, 2020-06-05:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b sam_e70_xplained -p auto tests/net/socket/select
2. west flash
3. See error

**Screenshots or console output**

*** Booting Zephyr OS build v2.3.0-rc2  ***
Running test suite socket_select

starting test - test_fd_set
PASS - test_fd_set

starting test - test_select
E: 0x00000000 is not a valid NET_SOCKET
E: address is not a known kernel object
E: 0x00000000 is not a valid NET_SOCKET
E: address is not a known kernel object
**Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/select/src/main.c:160: test_select: (k_uptime_get_32() - tstamp <= FUZZ is false)
FAIL - test_select**

Test suite socket_select failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.11.3
 - Commit ID: v2.3.0-rc2

ABOSTM, 2020-06-05:
The same for STM32 boards.
It looks like a regression introduced by #25804

dleach02, 2020-06-05:
@andrewboie , this is being traced back to your PR

PR #26001 (2020-06-05): net: sockets: Fix socket ctx check in usermode
When `z_get_fd_obj_and_vtable()` function returns NULL (no valid entry
in the FD table for the socket), there is no need for further usermode
checks on the `ctx` pointer, as there is nothing to invalidate in that
case.

Fixes #25990
Fixes #25991
commit 370dfe08 (2020-06-05): net: sockets: Fix socket ctx check in usermode

When `z_get_fd_obj_and_vtable()` function returns NULL (no valid entry
in the FD table for the socket), there is no need for further usermode
checks on the `ctx` pointer, as there is nothing to invalidate in that
case.

Fixes #25990
Fixes #25991
closed 2020-06-05 by carlescufi