zephyrproject-rtos/zephyr #25991: [net][net.socket.select][imx-rt series] test fails  (k_uptime_get_32() - tstamp <= FUZZ is false)

hakehuang, 2020-06-05:
**Describe the bug**
Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/select/src/main.c:160: test_select: (k_uptime_get_32() - tstamp <= FUZZ is false)

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1060_evk ..
3. make flash

**Expected behavior**
all test pass.

**Impact**
net socket function

**Screenshots or console output**
'''
*** Booting Zephyr OS version 2.3.0-rc2  ***
I: ETH_0 enabled 100M full-duplex mode.
Running test suite socket_select
===================================================================
starting test - test_fd_set
PASS - test_fd_set
===================================================================
starting test - test_select
E: 0x00000000 is not a valid NET_SOCKET
E: address is not a known kernel object
E: 0x00000000 is not a valid NET_SOCKET
E: address is not a known kernel object

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/select/src/main.c:160: test_select: (k_uptime_get_32() - tstamp <= FUZZ is false)

FAIL - test_select
===================================================================
Test suite socket_select failed.
===================================================================
PROJECT EXECUTION FAILED
'''

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (Zephyr SDK, ...)
 - v.2.3.0-rc2

hakehuang, 2020-06-05:
@dleach02 , @MaureenHelm

carlescufi, 2020-06-05:
Probably duplicate of #25990

dleach02, 2020-06-05:
yep... looks like we introduced a bug in the last week... given that it is across multiple (all?) architectures maybe we should fix this before doing final?

closed 2020-06-05 by carlescufi

dleach02, 2020-06-05:
I can confirm that #26001 fixes this problem. I reproduced the problem before and then confirmed that the test passed after applying the patch

PR #26001 (2020-06-05): net: sockets: Fix socket ctx check in usermode
When `z_get_fd_obj_and_vtable()` function returns NULL (no valid entry
in the FD table for the socket), there is no need for further usermode
checks on the `ctx` pointer, as there is nothing to invalidate in that
case.

Fixes #25990
Fixes #25991
commit 370dfe08 (2020-06-05): net: sockets: Fix socket ctx check in usermode

When `z_get_fd_obj_and_vtable()` function returns NULL (no valid entry
in the FD table for the socket), there is no need for further usermode
checks on the `ctx` pointer, as there is nothing to invalidate in that
case.

Fixes #25990
Fixes #25991