zephyrproject-rtos/zephyr #26021: Problems compiling for Measuring Time

jaslob, 2020-06-06:
Hello I'm pretty new to Zephyr and have Problems compiling my application for measuring Time.  Not sure if it's something I did wrong. Thanks for your help.

main.cpp
#include <zephyr.h>
#include <math.h>
#include <stdio.h>
#include <iostream>
#include <iomanip>  
#include <kernel.h>

int main()
{
    const double Pi = 3.141592653;
    
    double angle;
    double rad;
    double sine;
    
    u32_t start_time;
    u32_t stop_time;
    u32_t cycles_spent;
    u32_t nanoseconds_spent;
    
    ...
    	angle = sys_rand32_get() % 360;
    	rad = angle * Pi / 180;
	start_time = k_cycle_get_32();
    	sine = sin(rad);
	stop_time = k_cycle_get_32();
	
	cycles_spent = stop_time - start_time; // time in miliseconds
	nanoseconds_spent = SYS_CLOCK_HW_CYCLES_TO_NS(cycles_spent);
        
      ...
    
    return 0;
}
prj.conf
CONFIG_NEWLIB_LIBC=y
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_STD_CPP17=y

CONFIG_MAIN_STACK_SIZE=2048
CONFIG_FPU=y
CONFIG_SSE=y

CONFIG_TEST_RANDOM_GENERATOR=y

Error:
[1/7] Linking CXX executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf 
: && /home/user/Zephyr/zephyr-sdk-0.11.2/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-g++  -m32   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -Wl,-T  zephyr/linker.cmd  -Wl,-Map=/home/user/zephyrproject/zephyr/project/benchmarking/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/arch/x86/core/libarch__x86__core.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/x86/ia32/libsoc__x86__ia32.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/serial/libdrivers__serial.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/x86/core/offsets/offsets.c.obj  -L"/home/user/Zephyr/zephyr-sdk-0.11.2/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/32"  -L/home/user/zephyrproject/zephyr/project/benchmarking/build/zephyr  -lgcc  -Wl,--print-memory-usage  -m32  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lstdc++  -lm  -lc  -lgcc  -L"/home/user/Zephyr/zephyr-sdk-0.11.2/x86_64-zephyr-elf/x86_64-zephyr-elf"/lib/32 && :
Memory region         Used Size  Region Size  %age Used
             RAM:      563672 B         4 MB     13.44%
        IDT_LIST:          80 B         2 KB      3.91/home/user/Zephyr/zephyr-sdk-0.11.2/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/../../../../x86_64-zephyr-elf/bin/ld: app/libapp.a(main.cpp.obj): in function `arch_k_cycle_get_32':
/home/user/zephyrproject/zephyr/include/arch/x86/arch.h:226: undefined reference to `z_timer_cycle_get_32()'
/home/user/Zephyr/zephyr-sdk-0.11.2/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/../../../../x86_64-zephyr-elf/bin/ld: /home/user/zephyrproject/zephyr/include/arch/x86/arch.h:226: undefined reference to `z_timer_cycle_get_32()'
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/user/zephyrproject/zephyr/project/benchmarking/build

pabigot, 2020-06-08:
What version of Zephyr are you using, and what is your target platform?  Presumably some Intel one since `CONFIG_SSE=y`.

After porting to current master (replace `SYS_CLOCK_HW_CYCLES_TO_NS` with `k_cyc_to_ns_ceil32` and int types this links on pca10056.

jaslob, 2020-06-09:
I'm using the zephyr sdk 0.11.2 and am simulating with qemu_x86.

pabigot, 2020-06-09:
Not the SDK, the Zephyr version.   Are you on the master branch, or using 2.0.2, or 1.14?

jaslob, 2020-06-09:
Master branch (I followed the getting started guide)

pabigot, 2020-06-09:
Thanks, reproduced, fix in #26082.

PR #26082 (2020-06-09): arch: x86: add extern "C" to arch.h
Without this builds for qemu_x86 can't invoke k_cycle_get_32()
because z_timer_cycle_get_32() is installed with a mangled name.

Fixes #26021
closed 2020-06-09 by nashif