zephyrproject-rtos/zephyr #26026: RFC: drivers: pwm: add functions for capturing pwm pulse width and period

henrikbrixandersen, 2020-06-06:
## Introduction

Extend the Zephyr RTOS PWM API with functions for capturing PWM pulse width and period time.

### Problem description

Many PWM controllers supports capturing PWM pulse width and period time/frequency, but Zephyr currently lacks an API for exposing this functionality.

### Proposed change

Add two API functions for capturing the PWM pulse width and period:
- `pwm_pin_capture_cycles()`
- `pwm_pin_capture_cycles_async()´

## Detailed RFC

Please see PR ##26025 for proposed API.

### Proposed change (Detailed)

Please see PR ##26025 for proposed API.

### Dependencies

None.

### Concerns and Unresolved Questions

Not sure if the `pwm_pin_capture_cycles_async()´ function will be useful to anybody. I do not have a use for it, but since this is how other asynchronous Zephyr RTOS APIs are done, I have added it for completeness.

## Alternatives

PWM capture could also be implemented as a sensor driver, but seeing that many PWM controllers allow channels/pins to be used for PWM generation while others are used for PWM capture, it would be difficult to share the device between these two functions.

The Linux kernel PWM subsystem allows for PWM controllers to provide a capture function as well.

pabigot, 2020-07-07:
API 2020-07-07: Moved from triage to in-progress per discussion.

closed 2021-01-12 by carlescufi

FERASMANASEER, 2025-12-28:
> ## مقدمة
> قم بتوسيع واجهة برمجة تطبيقات PWM الخاصة بنظام التشغيل Zephyr RTOS بوظائف لالتقاط عرض نبضة PWM وفترة الزمن.
> 
> ### وصف المشكلة
> تدعم العديد من وحدات التحكم PWM التقاط عرض نبضة PWM وفترة/تردد الدورة، لكن Zephyr تفتقر حاليًا إلى واجهة برمجة تطبيقات لعرض هذه الوظيفة.
> 
> ### التغيير المقترح
> أضف وظيفتين من وظائف واجهة برمجة التطبيقات (API) لالتقاط عرض نبضة PWM وفترتها:
> 
> * `pwm_pin_capture_cycles()`
> * `pwm_pin_capture_cycles_async()`
> 
> ## طلب التعليقات المفصل
> ### التغيير المقترح (مفصل)
> يرجى الاطلاع على طلب السحب [رقم 26025](https://github.com/zephyrproject-rtos/zephyr/pull/26025) للاطلاع على واجهة برمجة التطبيقات المقترحة.
> 
> ### التبعيات
> لا أحد.
> 
> ### المخاوف والأسئلة التي لم يتم حلها
> لست متأكدًا مما إذا كانت هذه `pwm_pin_capture_cycles_async()`الوظيفة ستكون مفيدة لأحد. أنا شخصيًا لا أستخدمها، ولكن بما أن هذه هي الطريقة التي تُنفذ بها واجهات برمجة تطبيقات Zephyr RTOS غير المتزامنة الأخرى، فقد أضفتها من باب الإكمال.
> 
> ## البدائل
> يمكن أيضًا تنفيذ التقاط PWM كمحرك مستشعر، ولكن نظرًا لأن العديد من وحدات التحكم PWM تسمح باستخدام القنوات / الدبابيس لتوليد PWM بينما يتم استخدام قنوات أخرى لالتقاط PWM، فسيكون من الصعب مشاركة الجهاز بين هاتين الوظيفتين.
> 
> يسمح النظام الفرعي PWM في نواة لينكس لوحدات التحكم PWM بتوفير وظيفة التقاط أيضًا.

PR #26025 (2021-01-12): drivers: pwm: add functions for capturing pwm pulse width and period
Extend the PWM API with optional functions for capturing the PWM pulse width and period time.

Fixes #26026.
commit 77b8440f (2021-01-12): drivers: pwm: add API for capturing pwm pulse width and period

Extend the PWM API with optional API functions for capturing PWM pulse
width and period cycles.

Fixes #26026.