zephyrproject-rtos/zephyr #26034: menuconfig target aborts when Kconfig warnings are present

aport, 2020-06-07:
**Describe the bug**
`west build -t menuconfig` aborts when Kconfig warnings are present. This prevents the user from resolving the Kconfig warnings.

**To Reproduce**
For this purpose I'll use an exising Kconfig conflict. Any existing Kconfig symbol conflict will work to reproduce.
Steps to reproduce the behavior:
1. Enable CONFIG_POSIX_API=y
2. Enable CONFIG_NET_SOCKETS_POSIX_API=y (this symbol conflicts with the previous)
3. Save configuration
4. Attempt to build, see build fails due to Kconfig warning
5. Issue `west build -t menuconfig` to resolve the Kconfig warning
6. See that the Kconfig warning prevents `menuconfig` from running

**Expected behavior**
Kconfig warnings should not prevent `menuconfig` from executing. A user would expect to use `menuconfig` as the primary strategy for resolving Kconfig warnings.

**Impact**
I must delete .config and start over

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain gnuarmemb
 - v2.3.0

tejlmand, 2020-06-08:
This is another variant of this existing bug: https://github.com/zephyrproject-rtos/zephyr/issues/9573

carlescufi, 2020-06-08:
Closing as duplicate of #9573

closed 2020-06-08 by carlescufi