zephyrproject-rtos/zephyr #26038: build zypher with llvm fail

bhe4, 2020-06-08:
**Describe the bug**
build zypher with llvm fail

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
[7/114] ccache /usr/bin/clang --target=i686-pc-none-elf -DKERNEL -D__ZEPHYR__=1 -I../kernel/include -I../arch/x86/include -I../include -Izephyr/include/generated -I../soc/x86/ia32 -isystem ../lib/libc/minimal/include -isystem /usr/lib/llvm-9/lib/clang/9.0.0/include -isystem /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/include-fixed -Os -imacros/mnt/c/iafw/ngfw/zephyrproject/zephyr_ngfw/build_pre/zephyr/include/generated/
autoconf.h -ffreestanding -fno-common -g -imacros/mnt/c/iafw/ngfw/zephyrproject/zephyr_ngfw/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign
-Wpointer-arith -Wno-sometimes-uninitialized -Wno-shift-overflow -Wno-missing-braces -Wno-self-assign -Wno-address-of-packed-member -Wno-unused-function -Wno-initializer-overrides -Wno-section -Wno-unknown-warn$
ng-option -Wno-unused-variable -Wno-format-invalid-specifier -Wno-gnu -Wno-tautological-compare -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -ffunction-sections -f$
ata-sections -Qunused-arguments -m32 -gdwarf-2 -mno-sse -march=pentium -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/x86/cor$
/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj   -c /mnt/c/iafw/ngfw/zephyrproject/zephyr_ngfw/arch/x86/core/offsets/offsets.c
FAILED: zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj
ccache /usr/bin/clang --target=i686-pc-none-elf -DKERNEL -D__ZEPHYR__=1 -I../kernel/include -I../arch/x86/include -I../include -Izephyr/include/generated -I../soc/x86/ia32 -isystem ../lib/libc/minimal/include -$
system /usr/lib/llvm-9/lib/clang/9.0.0/include -isystem /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/include-fixed -Os -imacros/mnt/c/iafw/ngfw/zephyrproject/zephyr_ngfw/build_pre/zephyr/include/generated/autocon$
.h -ffreestanding -fno-common -g -imacros/mnt/c/iafw/ngfw/zephyrproject/zephyr_ngfw/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpoint$
r-arith -Wno-sometimes-uninitialized -Wno-shift-overflow -Wno-missing-braces -Wno-self-assign -Wno-address-of-packed-member -Wno-unused-function -Wno-initializer-overrides -Wno-section -Wno-unknown-warning-opti$
n -Wno-unused-variable -Wno-format-invalid-specifier -Wno-gnu -Wno-tautological-compare -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -ffunction-sections -fdata-sec$
ions -Qunused-arguments -m32 -gdwarf-2 -mno-sse -march=pentium -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/x86/core/offset$
/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj   -c /mnt/c/iafw/ngfw/zephyrproject/zephyr_ngfw/arch/x86/core/offsets/offsets.c
In file included from /mnt/c/iafw/ngfw/zephyrproject/zephyr_ngfw/arch/x86/core/offsets/offsets.c:9:
../kernel/include/kernel_offsets.h:26:1: error: invalid operand in inline asm: '.globl  ___cpu_t_current_OFFSET .equ    ___cpu_t_current_OFFSET,${0:p}  .type   ___cpu_t_current_OFFSET,@object'
GEN_OFFSET_SYM(_cpu_t, current);
^
../kernel/include/gen_offset.h:80:2: note: expanded from macro 'GEN_OFFSET_SYM'
        GEN_ABSOLUTE_SYM(__##S##_##M##_##OFFSET, offsetof(S, M))
        ^
../include/toolchain/gcc.h:405:10: note: expanded from macro 'GEN_ABSOLUTE_SYM'
        __asm__(".globl\t" #name "\n\t.equ\t" #name \
                ^
In file included from /mnt/c/iafw/ngfw/zephyrproject/zephyr_ngfw/arch/x86/core/offsets/offsets.c:9:
../kernel/include/kernel_offsets.h:26:1: error: unknown token in expression in '.equ' directive
../kernel/include/gen_offset.h:80:2: note: expanded from macro 'GEN_OFFSET_SYM'
        GEN_ABSOLUTE_SYM(__##S##_##M##_##OFFSET, offsetof(S, M))
        ^
../include/toolchain/gcc.h:405:30: note: expanded from macro 'GEN_ABSOLUTE_SYM'
        __asm__(".globl\t" #name "\n\t.equ\t" #name \
                                    ^
<inline asm>:2:31: note: instantiated into assembly here
        .equ    ___cpu_t_current_OFFSET,
                                        ^
In file included from /mnt/c/iafw/ngfw/zephyrproject/zephyr_ngfw/arch/x86/core/offsets/offsets.c:9:
../kernel/include/kernel_offsets.h:26:1: error: missing expression in '.equ' directive
GEN_OFFSET_SYM(_cpu_t, current);
^
../kernel/include/gen_offset.h:80:2: note: expanded from macro 'GEN_OFFSET_SYM'
        GEN_ABSOLUTE_SYM(__##S##_##M##_##OFFSET, offsetof(S, M))
        ^
../include/toolchain/gcc.h:405:30: note: expanded from macro 'GEN_ABSOLUTE_SYM'
        __asm__(".globl\t" #name "\n\t.equ\t" #name \
                                    ^
<inline asm>:2:31: note: instantiated into assembly here
        .equ    ___cpu_t_current_OFFSET,

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

carlescufi, 2020-06-08:
Building Zephyr is not currently supported with LLVM. You can follow the issue tracking this and comment there:
https://github.com/zephyrproject-rtos/zephyr/issues/3689

closed 2020-06-08 by carlescufi

AmPaschal, 2022-04-28:
Hi,

While the issue #3689 is still open, it says llvm currently works with the board set to native_posix. But when I try this, I still run into this same error specified here.