zephyrproject-rtos/zephyr #26041: Replay attack reset fix

johnsmithecon, 2020-06-08:
**Is your enhancement proposal related to a problem? Please describe.**
When I reset my publisher and keep the same node Id, my publisher becomes unusable to the subscriber because the sequence number is now lower.

**Describe the solution you'd like**
When I reset my publisher, I should be able to reprovision it to any address and have normal functionality.

**Describe alternatives you've considered**
I know you can reset whole mesh network or you can change address of publisher, but neither are efficient solutions

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

trond-snekvik, 2020-06-08:
I'm afraid this is by design. Reallocating the same address requires wiping the RPL of all other nodes for that particular address, which breaks replay protection completely. 

Consider for instance a third party that records all your mesh traffic, and your publisher sending an "unlock front door" message with seqnum 10000. When you reset the RPL of all the receivers and reset your publisher, the publisher starts sending from seqnum 0. If the third party then resends the "unlock front door" message, it will not only allow someone to walk right in and steal your TV, it'll also block your publisher's messages. When the publisher sends its next 9999 messages, the other devices will already have seen message number 10000 and will (ironically) reject the legit messages as replays.

You only really have two options, and I realize they might both suck: 
1. Don't reprovision your device.
2. Don't reuse the address.

I don't see how we could implement this without breaking the security of the mesh stack

carlescufi, 2020-06-18:
One option would be to add an unsafe Kconfig option that disables storing the replay protection list (RPL) in non-volatile storage. At that point, power-cycling the receivers would allow you to work with a reprovisioned publisher.

@johnsmithecon please feel free to send a proposal in that direction, but you will need to add a big warning that the option is insecure.

carlescufi, 2020-06-18:
> @trond-snekvik is this a bug or an enhancement?

Definitely an enhancement

PavelVPV, 2025-02-10:
> When I reset my publisher, I should be able to reprovision it to any address and have normal functionality.

This is against the Bluetooth Mesh specification:

Section 3.11.7, MshPRTv1.1:

```
After a node is removed from a network, its unicast addresses may be reused by a Provisioner. A
Provisioner shall only reuse these addresses after the current IV Index (at the time of removal) has been
updated (see Section 3.11.5) in order to enable the sequence numbers to be reused.
```

Plus, this can't be done easily because other nodes in the network should know somehow that that particular node was reset so they need to remove the RPL entry of that node.
This could be done through a vendor model, but it will open the network for replay attacks.
It is possible to build the Zephyr mesh without persistent storage support (`CONFIG_BT_SETTINGS=n`) so that a device boots up unprovisioned. However, the whole network needs to be reset then.

> I know you can reset whole mesh network or you can change address of publisher, but neither are efficient solutions

This is not correct. A user doesn't need to reset the whole network, only wait for a new IV index or give any unused address in the mesh network.

Closing this issue as it against the spec. If someone still wants exactly this behavior, then, either:
- Avoid using hardcoded unicast addresses and use group or virtual addresses instead
- Disable `CONFIG_BT_SETTINGS` and reset the entire network
- Implement a vendor model to reset a certain entry in the RPL list.

closed 2025-02-10 by PavelVPV