zephyrproject-rtos/zephyr #26049: False multiple define of irq with IRQ_CONNECT

lsangild, 2020-06-08:
**Describe the bug**
The bug prevents me from setting up an interrupt with a certain interrupt number.

I am testing the irq on the STM32F4-Discovery. My first setup would be connecting an interrupt line to the switch on the board, turning on/off the led, to test the possibilities of the interrupt options provided by Zephyr.
When using the expected interrupt line 6, according to the [Reference Manual](https://www.st.com/resource/en/reference_manual/dm00031020-stm32f405415-stm32f407417-stm32f427437-and-stm32f429439-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) page 374, I get the error, but I am able to compile using channel 11 though, but this does not seem to turn on the led anyways. 

    Generating ISR table .pio/build/disco_f407vg/zephyr/isr_tables.c
    gen_isr_tables.py: multiple registrations at table_index 6 for irq 6 (0x6)
    Has IRQ_CONNECT or IRQ_DIRECT_CONNECT accidentally been invoked on the same irq multiple times?
    Traceback (most recent call last):
      File "/home/XXX/.platformio/packages/framework-zephyr/arch/common/gen_isr_tables.py", line 315, in <module>
        main()
      File "/home/XXX/.platformio/packages/framework-zephyr/arch/common/gen_isr_tables.py", line 307, in main
        + "\nHas IRQ_CONNECT or IRQ_DIRECT_CONNECT accidentally been invoked on the same irq multiple times?"
      File "/home/XXX/.platformio/packages/framework-zephyr/arch/common/gen_isr_tables.py", line 36, in error
        raise Exception()
    Exception
    *** [.pio/build/disco_f407vg/zephyr/isr_tables.c] Error 1

**To Reproduce**
Using the following code
```
#include <zephyr.h>

void led_isr ( void * arg ) {
    return;
}

int main(void)
{
    IRQ_CONNECT(6, 13, led_isr, NULL, 0);

    while (1)
    {
    k_sleep(1000);
    }
    return 0;
};
```

Steps to reproduce the behavior:
1. Create empty project
2. Build with platformio

**Expected behavior**
This builds, and I am able to add stuff to the led_isr routine at a later time.

**Impact**
Showstopper. I wil be running tight regulation which would at best be hardware interrupts, but I am unable to play with the options at the moment, and therefore unable to benchmark interrupt reaction speeds.

**Environment (please complete the following information):**
 - OS: Fedora 31, Linux 5.6.15
 - Toolchain Zephyr SDK
 - Version 2.2 provided by platformio

carlescufi, 2020-06-08:
@lsangild could it be that you have a driver enabled that already uses that interrupt line?

lsangild, 2020-06-08:
Could that be caused by the usage of `gpios = <&gpioa 0 GPIO_INT_ACTIVE_LOW>;` in the .dts file?(I cannot test this until tomorrow).
I am building my own board with the STM32F407 and has used the Zephyr v2.1 version of the dts file as template.

If this is indeed the issue, is there another way of attaching to an interrupt created in the device tree?

erwango, 2020-06-08:
So indeed,  interrupt line 6 is used for Ext IRQ PX0 lines which is matching button configuration (PA0).
Unfortunately we don't have an API yet to do what you want.
What you can try is removing the button from board configuration.

lsangild, 2020-06-09:
I have created a new project based of off Zephyr v2.2, which uses the original board file for the stm32f4_disco and has `gpios = <&gpioa 0 GPIO_ACTIVE_LOW>;` for the button.
Removing the gpio-key and the alias does NOT resolve the issue. On the other hand I have found that I am allowed to build when using IRQ_DIRECT_CONNECT, which seems odd to me.

erwango, 2020-06-09:
@Isangild sorry for quick and bad answer yesterday.
Removing gpio button definition alone can't solve the issue.
Reason is current implementation of external interrupt controller is hardcoded and configure all lines  at compile time. This is at this step that isr_table is build and the double definition is generating the issue you can see when using IRQ_CONNECT.
I guess you don't see this issue with IRQ_DIRECT_CONNECT because it doesn't rely on compile time defined isr_table. But it won't work because of the conflicting configuration.

So, in order to proceed with your benchmark, you'll need to hack deep in `drivers/interrupt_controller/intc_exti_stm32.c` and comment code related to `EXTI0_IRQn` handling.

> The reason I am doing this, is that I am searching for a more transparent way of connecting to hardware interrupts for e.g. pins, I2C, UART, DMA a.o.
Is the right step to use the CMSIS implementations of setting interrupt handlers?

IIRC, this means that you don't want to use current drivers in the way they are implemented.
Do you mean you intend only zephyr kernel and re-implement your own set of drivers ?
This is of course possible, but since zephyr idea is to provide a full framework, there is no specific guideline on how to achieve this.

lsangild, 2020-06-09:
I believe it might be my understanding of the framework which is wrong.
Looking at the API (again), I see that the functionality I want is indeed already available. The reason for my confusion is probably due to me feeling the abstraction from the hardware Zephyr offers is too far from the hardware, which gives cause to confusion on my part when going through the reference manual and attempting to find Zephyr implementations of the functionality I want to use.

I do believe that the error message given, should also attempt to explain why the interrupt is not available (who took it before me?). How should anyone know that Zephyr makes claims on the interrupts, even though they are not configured anywhere in my code (no usage of e.g. gpio_init_callback)

closed 2020-06-09 by lsangild