zephyrproject-rtos/zephyr #26051: shell: uart: Allow a change in the shell initalisation to let routing it through USB UART

KubaFYI, 2020-06-08:
**Is your enhancement proposal related to a problem? Please describe.**
When wanting to route the shell through USB CDC ACM there is a problem with the shell getting possibly initialised before the CDC ACM driver. 

**Describe the solution you'd like**
I managed to successfully accomplish what I need by adding a call to `usb_enable()` in the `enable_shell_uart()` function and changing the literal `0` shell initialisation priority to `CONFIG_APPLICATION_INIT_PRIORITY` (so that it happens after the USB device's `CONFIG_KERNEL_INIT_PRIORITY_DEVICE`.

However, I'm not sure if this is the correct way to do it with me being possibly unaware of the reason why shell's initialisation needs to be `0`, an the best place to `usb_enable()`.

KubaFYI, 2020-06-08:
I've done the work on my fork at https://github.com/KubaFYI/zephyr/tree/feat_usb_acm_shell.

carlescufi, 2020-06-09:
> I've done the work on my fork at https://github.com/KubaFYI/zephyr/tree/feat_usb_acm_shell.

Why don't you send an RFC PR with the code you have and we then discuss it there?

KubaFYI, 2020-06-09:
@carlescufi Oh ok. I always kinda assumed it's annoying to bother everyone with PRs which have not been discussed to be the right way forward, but if that's the preferred workflow I'll go for it!

jfischer-phytec-iot, 2020-10-14:
The solution in current Zephyr tree is to configure the shell (`CONFIG_UART_SHELL_ON_DEV_NAME="CDC_ACM_0"`) and enable USB device support in user application.
See also https://github.com/zephyrproject-rtos/zephyr/pull/26122#issuecomment-689417216

closed 2020-10-14 by jfischer-phytec-iot

mountaindrew1959, 2021-09-09:
I just updated my Zephyr libraries and SDK to 0.13.0.  I know the CDC ACM configuration method using CONFIG_UART_SHELL_ON_DEV_NAME="CDC_ACM_0" in the prj.conf and defconfig files has been deprecated to the device tree.

I've added this code to my board .dts file:

zephyr_udc0: &usbotg_fs {
    pinctrl-0 = <&usb_otg_fs_dm_pa11 &usb_otg_fs_dp_pa12>;
    status = "okay";
};

&zephyr_udc0 {
    cdc_acm_uart0 {
        compatible = "zephyr,cdc-acm-uart";
        label = "CDC_ACM_0";
    };
};

I can get the power shell pointed to my serial TTY (ST Link) using:

 zephyr,shell-uart = &usart3;

But I can no longer get it to my USB.  I'm pretty new to this stuff so obviously there's something I don't understand.  Can anybody help me? Please and thanks in advance.