zephyrproject-rtos/zephyr #26059: Potentially incorrect interrupt handling in nRF SoC .dtsi for GPIO

mbolivar-nordic, 2020-06-08:
I am implementing #26050 and I noticed that the GPIO nodes (not GPIOTE nodes) on nRF SoC .dtsi files do not have `interrupts` properties. This causes a potential issue with dependency ordering.

In particular, it means that GPIO nodes can come before the GPIOTE and NVIC nodes when listed in dependency order, e.g.  `/soc/interrupt-controller@e00e100` (ordinal 21) and `/soc/gpiote@40006000` (ordinal 26) both come after `/soc/gpio@50000000` (ordinal 8) here:

```
$ west build -b nrf52840dk_nrf52840 -s samples/hello_world --cmake-only
$ head -n 37 build/zephyr/include/generated/devicetree_unfixed.h 
/*
 * Generated by gen_defines.py
 *
 * DTS input file:
 *   nrf52840dk_nrf52840.dts.pre.tmp
 *
 * Directories with bindings:
 *   $ZEPHYR_BASE/dts/bindings
 *
 * Nodes in dependency order (ordinal and path):
 *   0   /
 *   1   /aliases
 *   2   /analog-connector
 *   3   /chosen
 *   4   /connector
 *   5   /sw-pwm
 *   6   /buttons
 *   7   /soc
 *   8   /soc/gpio@50000000
 *   9   /buttons/button_0
 *   10  /buttons/button_1
 *   11  /buttons/button_2
 *   12  /buttons/button_3
 *   13  /cpus
 *   14  /cpus/cpu@0
 *   15  /leds
 *   16  /leds/led_0
 *   17  /leds/led_1
 *   18  /leds/led_2
 *   19  /leds/led_3
 *   20  /pwmleds
 *   21  /soc/interrupt-controller@e000e100
 *   22  /soc/pwm@4001c000
 *   23  /pwmleds/pwm_led_0
 *   24  /soc/adc@40007000
 *   25  /soc/clock@40000000
 *   26  /soc/gpiote@40006000
```

I'm not sure that's right from a dependency perspective given that we want to guarantee that nodes depend on the nodes which generate their interrupts.

I understand that the GPIOTE peripheral is what actually has the NVIC IRQ number, not the GPIO peripheral itself, but I would have expected the GPIO node to come after GPIOTE, and GPIOTE to come after NVIC, above.

Would it make more sense to mark the GPIOTE nodes with the `interrupt-controller;` property and add `interrupt-parent = <&gpiote>` and `interrupts` properties to each GPIO node? Or am I missing something?

mbolivar-nordic, 2020-06-08:
cc @pabigot

pabigot, 2020-06-08:
I'm not sure, but I think this is partly correct: gpiote depends on nvic, and shows up as requiring interrupt-controller.
```
 * Binding (compatible = nordic,nrf-gpiote):
 *   $ZEPHYR_BASE/dts/bindings/gpio/nordic,nrf-gpiote.yaml
 *
 * Dependency Ordinal: 26
 *
 * Requires:
 *   7   /soc
 *   21  /soc/interrupt-controller@e000e100
```

What isn't recorded is the dependency of gpio on gpiote: there's no `Supports` blob under GPIOTE..  So maybe that could be done by also treating gpiote as an interrupt controller, but I don't think it is one.  I don't know what the `interrupts` properties of the GPIO nodes would be.

It probably makes sense to represent the gpiote < gpio dependency in devicetree; a `also-requires = <&gpiote>;` property would do it though it's not clear what to call it.

Alternatively we will need the ability for specific drivers to record dependencies on things that they need, even if it isn't an obvious hardware dependency.  The obvious case is dependencies on sys-init functions, but other drivers might be necessary as well.