zephyrproject-rtos/zephyr #26064: tests/kernel/timer/timer_api failed on mec15xxevb_assy6853 board.

chen-png, 2020-06-09:
**To Reproduce**
Steps to reproduce the behavior:
1.  sanitycheck --hardware-map ~/mec1.file --device-testing
     -T tests/kernel/timer/timer_api  -vv
2. see error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.3.0  ***
Running test suite timer_api

starting test - test_time_conversions
PASS - test_time_conversions

starting test - test_timer_duration_period
PASS - test_timer_duration_period

starting test - test_timer_period_0
PASS - test_timer_period_0

starting test - test_timer_expirefn_null
PASS - test_timer_expirefn_null
......
starting test - test_timer_k_define
PASS - test_timer_k_define

starting test - test_timer_user_data
PASS - test_timer_user_data

**starting test - test_timer_remaining
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:603: test_timer_remaining: (abs(delta_ticks) <= MAX(slew_ticks, 1U) is false)
tick/busy slew -2 larger than test threshold 0**
FAIL - test_timer_remaining

starting test - test_timeout_abs
PASS - test_timeout_abs

Test suite timer_api failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.11.3
 - Commit ID: zephyr-v2.3.0

FRASTM, 2020-06-09:
@chen-png  Maybe, this is similar to  #26039

chen-png, 2020-06-16:
>Maybe, this is similar to #26039

yes,  it's fixed by #26045 too, so closed this.

closed 2020-06-16 by chen-png