zephyrproject-rtos/zephyr #26065: regex in sample.yaml does not work

yingmingx, 2020-06-09:
**Describe the bug**
When I test an example in the directory "samples/hello_world", I changed the regex in sample.yaml file. For example, change as follows:
-        - "Hello World! (.*)"
+        - "intel shanghai! (.*)"
this means the regex does not match the character string which should be print in main.c, after I run the sanitycheck tool, it passed without report errors.
This situation also happened when I changed the samples/hello_world/src/main.c  without changed the sample.yaml.

**To Reproduce**
Steps to reproduce the behavior:
1. open samples/hello_world/sample.yaml
2. change as follows:
-        - "Hello World! (.*)"
+        - "intel shanghai! (.*)"
3. sanitycheck -p qemu_x86 -T samples/hello_world/

yingmingx, 2020-06-16:
Steven Wang is busy these days，please others help to take a look at this problem, thanks.

nashif, 2020-06-16:
introduced in 7ec57b4666c6b724ce4b349fab10884a6fb05929 <-- @wentongwu

andrewboie, 2020-06-17:
Upgrade to high. This may let code get merged which actually breaks the build.

PR #26223 (2020-06-25): sanitycheck: handle timeouts correctly with qemu
In some cases we were not handling return code correctly and evaluating
timeouts as a pass. Report failure correctly.

Fixes #26065
commit 6fae53c0 (2020-06-25): sanitycheck: handle timeouts correctly with qemu

In some cases we were not handling return code correctly and evaluating
timeouts as a pass. Report failure correctly.

Fixes #26065
closed 2020-06-25 by nashif