zephyrproject-rtos/zephyr #26070: Bluetooth: ATT request not processed

KAGA164, 2020-06-09:
**Describe the bug**
`bt_gatt_read` after several `bt_gatt_write_without_response` call, exit without error but the BLE packet is not sent over the air. After `bt_gatt_read` sample works but characteristic is never read and sniffer does not detect any GATT Read packet. I did some dubbuging and I found that if L2CAP channel is bussy by previous write operation the Read ATT request is queued:
```
	/* Queue the request to be send later */
	sys_slist_append(&att->reqs, &req->node);
```

But after Write operation is finished this list is never proccessed.

When write operation is outgoing the `tx_sem` is taken and Read request return on ` k_sem_take` with error here:
```
static int bt_att_chan_req_send(struct bt_att_chan *chan,
				struct bt_att_req *req)
{
	int err;

	__ASSERT_NO_MSG(chan);
	__ASSERT_NO_MSG(req);
	__ASSERT_NO_MSG(req->func);
	__ASSERT_NO_MSG(!chan->req);

	BT_DBG("req %p", req);

	if (chan->chan.tx.mtu < net_buf_frags_len(req->buf)) {
		return -EMSGSIZE;
	}

	if (k_sem_take(&chan->tx_sem, K_NO_WAIT) < 0) {
		return -EAGAIN;
	}

	BT_DBG("chan %p req %p len %zu", chan, req,
	       net_buf_frags_len(req->buf));

	chan->req = req;

	/* Save request state so it can be resent */
	net_buf_simple_save(&req->buf->b, &req->state);

	/* Keep a reference for resending in case of an error */
	err = chan_send(chan, net_buf_ref(req->buf), NULL);
	if (err < 0) {
		net_buf_unref(req->buf);
		req->buf = NULL;
		k_sem_give(&chan->tx_sem);
		chan->req = NULL;
	}

	return err;
}
```
In this case after `k_sem_take` returned with error. Request are queued here:

```
	/* Queue the request to be send later */
	sys_slist_append(&att->reqs, &req->node);
```
As I said earlier this is never processed.
In previous revision of the `att.c` this mechanism looks a little different. When tx_sem was taken then
the att_requests are also added to `tx_queue` like other att data, now they are not there and they are never processed.

Previous behaviour:
```
	if (k_sem_take(&att->tx_sem, K_NO_WAIT) < 0) {
		k_fifo_put(&att->tx_queue, req->buf);
		return 0;
	}
```

Conclusions.
If the BLE Host is bussy by sending the ATT data, sending ATT Request is queued but after the HOST finish processing previous ATT send the queued ATT request are never send.

**To Reproduce**
Sample code to reproduce
```
	while (prog < IMG_SIZE) {
		bt_gatt_write_without_response(throughput->conn,
					      throughput->char_handle,
					      data, len, false)
		if (err) {
			printk("GATT write failed (err %d)\n", err);
			break;
		}

		/* print graphics */
		printk("%c", img[prog / IMG_X][prog % IMG_X]);
		data += 244;
		prog++;
	}

	delta = k_uptime_delta_32(&stamp);

	printk("\nDone\n");
	printk("[local] sent %u bytes (%u KB) in %u ms at %llu kbps\n",
	       data, data / 1024, delta, ((u64_t)data * 8 / delta));

	/* read back char from peer */
	throughput->read_params.single.handle = throughput->char_handle;
	throughput->read_params.single.offset = 0;
	throughput->read_params.handle_count = 1;
	throughput->read_params.func = read_fn;

	err = bt_gatt_read(throughput->conn, &throughput->read_params);
	if (err) {
		LOG_ERR("Characteristic read failed.");
	}
	if (err) {
		printk("GATT read failed (err %d)\n", err);
	}
```

**Expected behavior**
GATT Read Request should be sent over air or `bt_gatt_read` should return an error code.

Vudentz, 2020-06-09:
@KAGA164 Please let me know if the following patch works for you as well:

https://github.com/Vudentz/zephyr/commit/cf3d16af089eab86ea8cc91ec11cc17eedf9aa00

KAGA164, 2020-06-10:
Hi @Vudentz , thanks for quick response. I tested yours commit, it seems to solve ATT Request sending problem.

PR #26063 (2020-06-18): Bluetooth: ATT: Multiple fixes
ATT channel sent callback shall not be overwritting until the
operation completes as it can result in breaking flow control when
CONFIG_BT_ATT_ENFORCE_FLOW is enabled.

Fixes #25964
Fixes #26070
Fixes #26071
commit bc7ce86a (2020-06-18): Bluetooth: ATT: Fix not processing pending requests

Since the TX semaphore is used for all types of PDUs a request may have
to be put on the request list while there is no pending request pending
which means no response will be generated to trigger att_process,
previously this condition was handled by setting the request as
currently pending and append its buffer to tx_queue but this is no
longer efficient since there could be more than one channel active the
code should try all of them before queueing back to request list.

To fix this the request list will now be processed each time a PDU h [... cut]
PR #26431 (2020-11-16): [v2.3-branch] Bluetooth: ATT: Multiple fixes
ATT channel sent callback shall not be overwritting until the
operation completes as it can result in breaking flow control when
CONFIG_BT_ATT_ENFORCE_FLOW is enabled.

Fixes #25964
Fixes #26070
Fixes #26071
commit b0cecce7 (2020-11-16): Bluetooth: ATT: Fix not processing pending requests

Since the TX semaphore is used for all types of PDUs a request may have
to be put on the request list while there is no pending request pending
which means no response will be generated to trigger att_process,
previously this condition was handled by setting the request as
currently pending and append its buffer to tx_queue but this is no
longer efficient since there could be more than one channel active the
code should try all of them before queueing back to request list.

To fix this the request list will now be processed each time a PDU h [... cut]
closed 2020-06-18 by carlescufi