zephyrproject-rtos/zephyr #26076: bug: cortex-m0: vector table base address is set to zero when soc has control over where to put vector table.

Aukstkalnis, 2020-06-09:
**Describe the bug**
BlueNRG-2 has cortex-m0 architecture. As we know, this core does not have vector table reallocation hw support. Usually, MCUs with this core can select which section (RAM or Flash) map to 0 address, but BlueNRG-2 soc has register FLASH-CONFIG, that allows selecting where is vector table: in RAM or flash FLASH.

Now, zephyr set vector address to 0:
```c
#ifdef CONFIG_CPU_CORTEX_M_HAS_VTOR
...
#else
#if defined(CONFIG_SW_VECTOR_RELAY)
Z_GENERIC_SECTION(.vt_pointer_section) void *_vector_table_pointer;
#endif

#define VECTOR_ADDRESS 0

void __weak relocate_vector_table(void)
{
#if defined(CONFIG_XIP) && (CONFIG_FLASH_BASE_ADDRESS != 0) || \
    !defined(CONFIG_XIP) && (CONFIG_SRAM_BASE_ADDRESS != 0)
	size_t vector_size = (size_t)_vector_end - (size_t)_vector_start;
	(void)memcpy(VECTOR_ADDRESS, _vector_start, vector_size);
#elif defined(CONFIG_SW_VECTOR_RELAY)
	_vector_table_pointer = _vector_start;
#endif
}
...
#endif /* CONFIG_CPU_CORTEX_M_HAS_VTOR */
```
Because of this, device can not work with bootloader.

**To Reproduce**
currently zephyr does not support BlueNRG soc so there is no way to reproduce.

Aukstkalnis, 2020-06-10:
Maybe we can say that is not a bug because relocate_vector_table function is declared as __weak and it can be overwritten in soc.c like with stm32f0 SOCs. Thats what I did. I am a little bit confused about _vector_table_pointer = _vector_start; and copying vector table to RAM beggining. Both methods works... If I just do this:
```c
void relocate_vector_table(void)
{
#ifdef CONFIG_SW_VECTOR_RELAY
	size_t vector_size = (size_t)_vector_end - (size_t)_vector_start;
	memcpy((void*)CONFIG_SRAM_BASE_ADDRESS, _vector_start, vector_size);
	FLASH->CONFIG = FLASH_REMAP_RAM;
#endif
}
```
or do this:
```c
void relocate_vector_table(void)
{
#ifdef CONFIG_SW_VECTOR_RELAY
    _vector_table_pointer = _vector_start;
#endif
}
```
everything works... When I just set _vector_table_pointer to _vector_start everything works without selecting where is vector table with FLASH->CONFIG register. Why do I get same results?

ioannisg, 2020-06-12:
Not sure I understand, fully, all points, but just want to try clarifying some things.

I assume, @Aukstkalnis that BlueNRG-2 is an stm32f0 SoC, is this correct?
If so, STM32F0 enables the following setting:
```
config CPU_CORTEX_M0_HAS_VECTOR_TABLE_REMAP
	bool
	help
	  This option signifies the Cortex-M0 has some mechanisms that can map
	  the vector table to SRAM
```
This setting makes the ARCH code understand that the SoC will handle the vector table relocation in  SoC configuration, so SW_VECTOR_RELAY does not need to be enabled (i.e. you can see that SW_VECTOR_RELAY depends on !CPU_CORTEX_M0_HAS_VECTOR_TABLE_REMAP). Therefore, SW_VECTOR_RELAY cannot be enabled on this SoC.

I think you are right about the weak implementation. If you see the soc.c of stm32f0x, it seems to do exactly what you refer to, that is copying the tables in SRAM and doing the necessary SoC configuration. This is done, in case the FW is loaded with a bootloader, so there does not seem to be a problem.

Relavant code is here:

```
/**
 * @brief Relocate vector table to SRAM.
 *
 * On Cortex-M0 platforms, the Vector Base address cannot be changed.
 *
 * A Zephyr image that is run from the mcuboot bootloader must relocate the
 * vector table to SRAM to be able to replace the vectors pointing to the
 * bootloader.
 *
 * A zephyr image that is a bootloader does not have to relocate the
 * vector table.
 *
 * Replaces the default function from prep_c.c.
 *
 * @note Zephyr applications that will not be loaded by a bootloader should
 *       pretend to be a bootloader if the SRAM vector table is not needed.
 */
void relocate_vector_table(void)
{
#ifndef CONFIG_IS_BOOTLOADER
	extern char _ram_vector_start[];

	size_t vector_size = (size_t)_vector_end - (size_t)_vector_start;

	memcpy(_ram_vector_start, _vector_start, vector_size);
	LL_SYSCFG_SetRemapMemory(LL_SYSCFG_REMAP_SRAM);
#endif
}

```

Aukstkalnis, 2020-06-16:
Thanks. In helped and now works. But I down understand why it worked when I SW_VECTOR_RELAY was declared and I used 
```
void relocate_vector_table(void)
{
#ifdef CONFIG_SW_VECTOR_RELAY
    _vector_table_pointer = _vector_start;
#endif
}
```
?

ioannisg, 2020-06-16:
@Aukstkalnis I read the following:
```
config SW_VECTOR_RELAY
	bool "Enable Software Vector Relay"
	default y if BOOTLOADER_MCUBOOT
	depends on ARMV6_M_ARMV8_M_BASELINE && !(CPU_CORTEX_M0_HAS_VECTOR_TABLE_REMAP || CPU_CORTEX_M_HAS_VTOR)
	help
	  Add Vector Table relay handler and relay vector table, to
	  relay interrupts based on a vector table pointer. This is only
	  required for Cortex-M0 (or an Armv8-M baseline core) with no hardware
	  vector table relocation mechanisms or for Cortex-M0+
	  (or an Armv8-M baseline core) with no VTOR and no other hardware
	  relocation table mechanisms.
```

So I understand that SW_VECTOR_RELAY cannot be defined on a platform that has CPU_CORTEX_M0_HAS_VECTOR_TABLE_REMAP  enabled.

Aukstkalnis, 2020-06-16:
Yes, but at that moment I did not enabled CPU_CORTEX_M0_HAS_VECTOR_TABLE_REMAP.

ioannisg, 2020-06-16:
> Yes, but at that moment I did not enabled CPU_CORTEX_M0_HAS_VECTOR_TABLE_REMAP.

@Aukstkalnis this one is enabled by default on SOC_SERIES_STM32F0X

Aukstkalnis, 2020-06-17:
> @Aukstkalnis this one is enabled by default on SOC_SERIES_STM32F0X

This is my custom soc for bluenrg because current zephyr do not have BlueNRG support, and I did not select CPU_CORTEX_M0_HAS_VECTOR_TABLE_REMAP. So the SW_VECTOR_RELAY was declared. Soc configuration:
```
config SOC_BLUENRG
	bool "BlueNRG"
	select ARM
	select CPU_CORTEX_M0
	select SOC_FAMILY_BLUENRG
	select BLUENRG
	select HAS_BLUENRG_HAL
	select CPU_CORTEX_M_HAS_SYSTICK
	select CORTEX_M_SYSTICK
	select SERIAL
	select CLOCK_CONTROL
	select HAS_SYS_POWER_STATE_DEEP_SLEEP_1
	select ARCH_HAS_CUSTOM_SWAP_TO_MAIN
```

ioannisg, 2020-06-17:
@Aukstkalnis right, I see.
So back to your original comment:

>Thanks. In helped and now works. But I down understand why it worked when I SW_VECTOR_RELAY was declared and I used

```
void relocate_vector_table(void)
{
#ifdef CONFIG_SW_VECTOR_RELAY
    _vector_table_pointer = _vector_start;
#endif
}
```
>Is it possible, that application used bootloader's vector table?

If there is a bootloader, then I suppose the bootloader's vector table is used, anyway, and if this bootloader table contains the logic to route the interrupts to some area in Flash/RAM and if your cector_table_pointer is in that area, then I guess you are achieving this relaying :)

Aukstkalnis, 2020-06-18:
OK. Thanks @ioannisg. Since the problem is solved, then we can close this case :)

closed 2020-06-18 by Aukstkalnis