zephyrproject-rtos/zephyr #26084: 2.4 Release Checklist

carlescufi, 2020-06-09:
- [ ] Major enhancements:
   - See https://github.com/zephyrproject-rtos/zephyr/projects/9

- [ ] Pre RC1 Steps
   - [ ] Create v2.5.0 milestone so that features that don't make it to the release can be assigned a milestone
   - [ ] Verify that all external components and external dependencies are up to date, for example 
      - mbedTLS
      - tinycrypt
      - FatFS
   - [ ] Check known vulnerabilities, and fix exploitable vulnerabilities or verify them as un-exploitable (Security working group).

See https://docs.zephyrproject.org/latest/development_process/release_process.html#tagging for tagging details

--------
- [ ] Release Notes WIP
  - [ ] Make sure to tell everybody to use the **past tense** (i.e. "X has been added" or "X is now Y")
  - [ ] Create draft `doc/releases/release-notes-2.4.rst` with feature summary from https://www.zephyrproject.org/developers/#releases-overview #22923
  - [ ] Update `doc/releases/release-notes-2.4.rst` detailed sections (relative to 2.3.0 release) as important issues are addressed (subsystem owners). See #25869
--------
- [ ] Finalize Release Notes
  - [ ] Add list of GitHub issues (spell check issue list and fix in GitHub as needed) using `scripts/release/list_issues.py`
  - [ ] Review `doc/releases/release-notes-2.4.rst` overview summary and details sections (see Sign-off below)
    - [ ] Request summary details from code owners
- [ ] update `doc/reference/overview.rst`
- [ ] Update doc/conf.py (add version to pick list)
- [ ] Update doc/LICENSING.rst for new components that do not have a license
--------
- [ ]  Final Release Notes Sign-off
  - Subsystems
    - [ ] Kernel - @andyross, @andrewboie
    - [ ] Build and infrastructure - @tejlmand @nashif @galak  
    - [ ] Devicetree - @galak, @mbolivar-nordic 
    - [ ] Drivers and sensors - @MaureenHelm, @tbursztyka, @pabigot, @mnkp  
    - [ ] Filesystems - @nvlsianpu, @pabigot  
    - [ ] Networking - @jukkar 
    - [ ] Bluetooth - @jhedberg , @joerchan, @carlescufi 
    - [ ] USB - @jfischer-phytec-iot @emob-nordic 
    - [ ] CAN - @alexanderwachter  
  - Architectures
    - [ ] ARC - @ruuddw 
    - [ ] ARM - @ioannisg , @galak, @MaureenHelm 
    - [ ] X86 - @nashif, @andrewboie 
    - [ ] RISCV - @pgielda @nategraff-sifive
    - [ ] Xtensa, NIOS-II - @nashif, @andrewboie 
  - Board an Soc support - many
  - [ ] Security - @d3zd3z 
--------
- [ ] Update version in VERSION
- [ ] Tag both v2.4.0 and zephyr-v2.4.0
- [ ] Create release and add notes on https://github.com/zephyrproject-rtos/zephyr/releases. Make sure you use the `zephyr-v2.4.0` tag so that the `.zip` that GitHub offers has a reasonable name. Add the release notes to the GitHub release.
--------
- [ ] After Release
  - [ ] Update patchlevel in Makefile (PATCHLEVEL = 99)
  - [ ] Update doc generation (in future should be automatic based on tag) ping @nashif
  - [ ] Update https://www.zephyrproject.org/developer-resources/#current-release (email Brett)
  - [ ] Create `v2.4-branch` branch for 2.4.x releases (from v2.4.0 tag, don't include the switch to patchlevel 99)
  - [ ] Merge all post-release PRs in https://github.com/zephyrproject-rtos/zephyr/projects/27
  - [ ] Move all PRs/Issues in the v2.4.0 milestone to the next milestone
  - [ ] Close v2.4.0 milestone 
  - [ ] Email devel@lists.zephyrproject.org & announce@lists.zephyrproject.org lists about release and announce the opening of the merge window
  - [ ] Create checklist issue for next release
  - [ ] Add a comment to this issue linking to the next release checklist issue
  - [ ] Marketing Blog for announcing release (https://www.zephyrproject.org/blog)
  - [ ] Update https://www.zephyrproject.org with latest stats from GitHub (email Brett)
  - [ ] Update current release dates in https://github.com/zephyrproject-rtos/zephyr/wiki/Program-Management, move current release column to "past releases table", and add next release timeline if missing (v2.6.0)

MaureenHelm, 2020-09-27:
Next release checklist: #28716

MaureenHelm, 2020-10-14:
Historical bug and PR tracking data for the 2.4.0 release cycle:

  | High | Medium | Low | PRs | Weeks to Feature Freeze | Weeks to Release
-- | -- | -- | -- | -- | -- | --
7/22/2020 | 2 | 44 | 256 | 521 | 6.3 | 9.3
7/28/2020 | 2 | 48 | 232 | 537 | 5.4 | 8.4
8/4/2020 | 6 | 45 | 225 | 550 | 4.4 | 7.4
8/11/2020 | 7 | 41 | 225 | 540 | 3.4 | 6.4
8/19/2020 | 10 | 39 | 205 | 532 | 2.3 | 5.3
8/25/2020 | 7 | 43 | 202 | 526 | 1.4 | 4.4
9/1/2020 | 5 | 44 | 190 | 518 | 0.4 | 3.4
9/8/2020 | 9 | 51 | 199 | 460 | -0.6 | 2.4
9/15/2020 | 3 | 42 | 179 | 479 | -1.6 | 1.4
9/16/2020 | 3 | 34 | 175 | 479 | -1.7 | 1.3
9/22/2020 | 2 | 25 | 141 | 367 | -2.6 | 0.4
9/24/2020 | 1 | 18 | 142 | 357 | -2.9 | 0.1
9/27/2020 | 1 | 20 | 142 | 355 | -3.3 | -0.3

tejlmand, 2020-10-15:
@MaureenHelm Nice, do we have a similar list for older releases ?

MaureenHelm, 2020-10-15:
> Nice, do we have a similar list for older releases ?

No

closed 2020-10-20 by MaureenHelm