zephyrproject-rtos/zephyr #26096: cmake finds a DTC von Zephyr-SDK version, it tries to execute it, and it fails

Kreijstal, 2020-06-10:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
this line https://github.com/zephyrproject-rtos/zephyr/blob/603c24dec913749c0198b2a10afc52ac5c3443db/cmake/host-tools.cmake#L50
figured out that my dtc is supposed to be `/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc` instead of `/usr/bin/dtc` which shouldn't work anyway because I have DTC 1.4.5.
But this specific version of dtc that I get from the SDK automatically fails and gives no stdout but stderr:
`/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc: error while loading shared libraries: libunwind.so.8: cannot open shared object file: No such file or directory`
(I do have libunwind installed)
because of this line https://github.com/zephyrproject-rtos/zephyr/blob/603c24dec913749c0198b2a10afc52ac5c3443db/cmake/host-tools.cmake#L63
since it has no output cmake says
```
CMake Error at /content/zephyrproject/zephyr/cmake/host-tools.cmake:65 (string):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.
Call Stack (most recent call first):
  /content/zephyrproject/zephyr/cmake/app/boilerplate.cmake:490 (include)
  /content/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /content/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

CMake Error at /content/zephyrproject/zephyr/cmake/host-tools.cmake:70 (if):
  if given arguments:

    "VERSION_GREATER" "1.4.6"

  Unknown arguments specified
Call Stack (most recent call first):
  /content/zephyrproject/zephyr/cmake/app/boilerplate.cmake:490 (include)
  /content/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /content/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:5 (find_package)
```

**To Reproduce**
Steps to reproduce the behavior:
1. install sdk
2. west build -b qemu_x86 samples/hello_world
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

CMake should make a sanity check to see if ${dtc_version_output} is empty before making the regex check, this will prevent this bug.
The DTC of Zephyr-SDK shouldn't have libunwind issues, either.

stephanosio, 2020-06-10:
This is not a bug per se; it is more of an enhancement for providing more accurate error messages.

tejlmand, 2020-06-12:
as CMake processing is aborted, when DTC is an optional component, then I would actually regard this as a bug, albeit minor.

Fixed here: #26137

PR #26137 (2020-06-12): cmake: improved DTC detection handling
Fixes: #26096

In some cases DTC may be installed, but in a non-working state.
In such cases, CMake will abort processing with an cryptic error.

This commit now handles dtc version errors and print a warning., and
CMake processing continues.

dtc is an optional component and failures should not break CMake run.
commit 8e8c6cd9 (2020-06-12): cmake: improved DTC detection handling

Fixes: #26096

In some cases DTC may be installed, but in a non-working state.
In such cases, CMake will abort processing with an cryptic error.

This commit now handles dtc version errors and print a warning., and
CMake processing continues.

dtc is an optional component and failures should not break CMake run.
closed 2020-06-12 by carlescufi