zephyrproject-rtos/zephyr #26112: bug: cmake loops when passing overlay file with left slashes in file path

Aukstkalnis, 2020-06-11:
**Describe the bug**
When building simple samples/basic/blinky project with overlay file passed with left slashes '\' in file path CMake loops in build preparations. Changing all path separators from \ to / do not cause this loop problem.

**To Reproduce**
Steps to reproduce the behavior:
1. create overlay file in overlay folder in simple blinky directory
2. run west west build -b stm32f4_disco -- -DDTC_OVERLAY_FILE="D:\Github\zephyrproject\zephyr\samples/basic/blinky/overlay/stm32f4_disco.overlay"
4. See cmake loop

**Expected behaviour**
I expected the build successful or failed build.

**Impact**
The problem comes when automating builds from vscode IDE because I cant use ${path variables} to create workspace independent build commands.

**Screenshots or console output**
Commands:
west build -b stm32f4_disco -- -DDTC_OVERLAY_FILE="D:\Github\zephyrproject\zephyr\samples/basic/blinky/overlay/stm32f4_disco.overlay"

Result:
```
-- west build: generating a build system
Including boilerplate (Zephyr base): D:/Github/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: D:/Github/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.3.99 (D:/Github/zephyrproject/zephyr)
-- Found Python3: C:/Python38/python.exe (found suitable exact version "3.8.3") found components: Interpreter
-- Board: stm32f4_disco
-- Found west: C:/Python38/Scripts/west.exe (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (C:/gnu_arm_embedded)
-- Found BOARD.dts: D:/Github/zephyrproject/zephyr/boards/arm/stm32f4_disco/stm32f4_disco.dts
-- Found devicetree overlay: D:\Github\zephyrproject\zephyr\samples/basic/blinky/overlay/stm32f4_disco.overlay
-- Generated zephyr.dts: D:/Github/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: D:/Github/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/devicetree_unfixed.h
Parsing D:/Github/zephyrproject/zephyr/Kconfig
Loaded configuration 'D:/Github/zephyrproject/zephyr/boards/arm/stm32f4_disco/stm32f4_disco_defconfig'
Merged configuration 'D:/Github/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to 'D:/Github/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/.config'
Kconfig header saved to 'D:/Github/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/gnu_arm_embedded/bin/arm-none-eabi-gcc.exe
-- Cache files will be written to: C:\Users\aukstkalnis.de\AppData\Local/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: D:/Github/zephyrproject/zephyr/samples/basic/blinky/build
-- west build: building application
[0/1] Re-running CMake...
Including boilerplate (Zephyr base (cached)): D:/Github/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: D:/Github/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.3.99 (D:/Github/zephyrproject/zephyr)
-- Board: stm32f4_disco
-- Found west: C:/Python38/Scripts/west.exe (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (C:/gnu_arm_embedded)
-- Found BOARD.dts: D:/Github/zephyrproject/zephyr/boards/arm/stm32f4_disco/stm32f4_disco.dts
-- Found devicetree overlay: D:\Github\zephyrproject\zephyr\samples/basic/blinky/overlay/stm32f4_disco.overlay
-- Generated zephyr.dts: D:/Github/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: D:/Github/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/devicetree_unfixed.h
Parsing D:/Github/zephyrproject/zephyr/Kconfig
Loaded configuration 'D:/Github/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/.config'
No change to configuration in 'D:/Github/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/.config'
No change to Kconfig header in 'D:/Github/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/autoconf.h'
-- Cache files will be written to: C:\Users\aukstkalnis.de\AppData\Local/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: D:/Github/zephyrproject/zephyr/samples/basic/blinky/build
[0/1] Re-running CMake...
Including boilerplate (Zephyr base (cached)): D:/Github/zephyrproject/zephyr/cmake/app/boilerplate.cmake
[... 49 lines cut ...]
-- Generated devicetree_unfixed.h: D:/Github/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/devicetree_unfixed.h
Parsing D:/Github/zephyrproject/zephyr/Kconfig
Loaded configuration 'D:/Github/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/.config'
No change to configuration in 'D:/Github/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/.config'
No change to Kconfig header in 'D:/Github/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/autoconf.h'
Traceback (most recent call last):
  File "D:/Github/zephyrproject/zephyr/scripts/kconfig/kconfig.py", line 264, in <module>
    main()
  File "D:/Github/zephyrproject/zephyr/scripts/kconfig/kconfig.py", line 91, in main
    write_kconfig_filenames(kconf, args.kconfig_list_out)
  File "D:/Github/zephyrproject/zephyr/scripts/kconfig/kconfig.py", line 219, in write_kconfig_filenames
    for path in sorted({os.path.realpath(os.path.join(kconf.srctree, path))
  File "D:/Github/zephyrproject/zephyr/scripts/kconfig/kconfig.py", line 219, in <setcomp>
    for path in sorted({os.path.realpath(os.path.join(kconf.srctree, path))
  File "C:\Python38\lib\ntpath.py", line 626, in realpath
    path = normpath(path)
  File "C:\Python38\lib\ntpath.py", line 498, in normpath
    return prefix + sep.join(comps)
KeyboardInterrupt
```

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain arm gcc
 - Commit 7f9b85e8c39f3c7ee7a2dbdffce4a45b969b48c3

Aukstkalnis, 2020-06-11:
@tejlmand you asked me to fill github issue :)

tejlmand, 2020-06-12:
Did a little investigation on this.

In CMake, the following is done:
https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/dts.cmake#L139

and this creates a dependency file, which in this case contains:
```
empty_file.o: C:/projects/github/ncs/zephyr/misc/empty_file.c \
...
 c:\projects\github\ncs\zephyr/samples/basic/blinky/ov/stm32f4_disco.overlay
```
which then causes troubles to ninja, due to the mixing style of `/` and `\`.
Ninja will check up on the file, and believe it is missing, and thus re-run CMake.
However, the C compiler will create and identical file, and thus when CMake finishes and ninja then has to run, it still believe the file to be missing, and re-run CMake (again-and-again-and-again...)

A quick fix would be to simply change the slash, however, a more correct way would be to ensure the code makes use of CMake dependency handling instead.

PR #27839 (2020-09-03): cmake: dts overlay file with mixed separator style support
Fixes: #26112

All file paths provided on the list of overlay files are converted to
CMake style paths.

This ensures that even paths with mixed style `\` and `/` on Windows are
treated correctly. This fixes the issue where the deps file created
would cause ninja to re-invoke CMake in an endless loop on windows.
commit 6dc8f0e7 (2020-09-03): cmake: dts overlay file with mixed separator style support

Fixes: #26112

All file paths provided on the list of overlay files are converted to
CMake style paths.

This ensures that even paths with mixed style `\` and `/` on Windows are
treated correctly. This fixes the issue where the deps file created
would cause ninja to re-invoke CMake in an endless loop on windows.
closed 2020-09-03 by galak