zephyrproject-rtos/zephyr #26119: Compilation error when enabling MPU on STM32 L0 boards

ABOSTM, 2020-06-11:
**Describe the bug**
When enabling MPU on STM32 L0 boards
(see branch STM32_MPU_ENABLED_L0 of my fork https://github.com/ABOSTM/zephyr.git)
There are compilation error.

**To Reproduce**
Steps to reproduce the behavior:
1. take branch STM32_MPU_ENABLED_L0  
    of repo https://github.com/ABOSTM/zephyr.git
2. west build -p auto -b nucleo_l073rz tests/arch/arm/arm_ramfunc
3. See compilation error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Compilation successful 

**Screenshots or console output**
```
FAILED: zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/stdlib/strtoul.c.obj
C:\gnuarmemb\bin\arm-none-eabi-gcc.exe -DBUILD_VERSION=zephyr-v2.3.0-166-g56da857bb683 -DHSE_VALUE=8000000 -DKERNEL -DSTM32L073xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../include -Izephyr/include/generated -I../soc/arm/st_stm32/stm32l0 -I../drivers -I../subsys/testsuite/include -I../subsys/testsuite/ztest/include -IC:/zephyr/zephyr_project/modules/hal/cmsis/CMSIS/Core/Include -IC:/zephyr/zephyr_project/modules/hal/stm32/stm32cube/stm32l0xx/soc -IC:/zephyr/zephyr_project/modules/hal/stm32/stm32cube/stm32l0xx/drivers/include -IC:/zephyr/zephyr_project/modules/hal/stm32/stm32cube/stm32l0xx/drivers/include/Legacy -isystem ../lib/libc/minimal/include -isystem c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.2.1/include -isystem c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.2.1/include-fixed -Os -imacrosC:/zephyr/zephyr_project/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m0plus -mthumb -mabi=aapcs -imacrosC:/zephyr/zephyr_project/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=C:/zephyr/zephyr_project/zephyr/tests/arch/arm/arm_ramfunc=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/zephyr/zephyr_project/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/zephyr/zephyr_project=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/stdlib/strtoul.c.obj -MF zephyr\lib\libc\minimal\CMakeFiles\lib__libc__minimal.dir\source\stdlib\strtoul.c.obj.d -o zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/stdlib/strtoul.c.obj   -c C:/zephyr/zephyr_project/zephyr/lib/libc/minimal/source/stdlib/strtoul.c
In file included from ../include/arch/cpu.h:12,
                 from ../include/kernel_includes.h:39,
                 from ../soc/arm/st_stm32/stm32l0/soc.h:31,
                 from ../include/arch/arm/aarch32/cortex_m/cmsis.h:17,
                 from ../include/arch/arm/aarch32/syscall.h:29,
                 from ../include/arch/syscall.h:19,
                 from ../include/syscall.h:12,
                 from zephyr/include/generated/syscalls/errno_private.h:10,
                 from ../include/sys/errno_private.h:34,
                 from ../lib/libc/minimal/include/errno.h:20,
                 from C:/zephyr/zephyr_project/zephyr/lib/libc/minimal/source/stdlib/strtoul.c:35:
../include/sys/arch_interface.h:417:25: error: conflicting types for 'arch_syscall_invoke2'
  417 | static inline uintptr_t arch_syscall_invoke2(uintptr_t arg1, uintptr_t arg2,
      |                         ^~~~~~~~~~~~~~~~~~~~
In file included from ../include/sys/atomic.h:562,
                 from ../include/kernel_includes.h:21,
                 from ../soc/arm/st_stm32/stm32l0/soc.h:31,
                 from ../include/arch/arm/aarch32/cortex_m/cmsis.h:17,
                 from ../include/arch/arm/aarch32/syscall.h:29,
                 from ../include/arch/syscall.h:19,
                 from ../include/syscall.h:12,
                 from zephyr/include/generated/syscalls/errno_private.h:10,
                 from ../include/sys/errno_private.h:34,
                 from ../lib/libc/minimal/include/errno.h:20,
                 from C:/zephyr/zephyr_project/zephyr/lib/libc/minimal/source/stdlib/strtoul.c:35:
zephyr/include/generated/syscalls/atomic.h:55:25: note: previous implicit declaration of 'arch_syscall_invoke2' was here
   55 |   return (atomic_val_t) arch_syscall_invoke2(*(uintptr_t *)&target, *(uintptr_t *)&value, K_SYSCALL_ATOMIC_ADD);
      |                         ^~~~~~~~~~~~~~~~~~~~

...
```

**1st Analysis**
Because L0 is based on cortex M0+ with  Armv6-M architecture, there is no Atomic builtin operations.
So it activates switch CONFIG_ATOMIC_OPERATIONS_C=y
Then file zephyr/include/generated/syscalls/atomic.h is generated with C implementation of atomic operation.
Within those functions, we make use of `arch_syscall_invoke2()` , `z_syscall_trap()` , ...
But those functions  `arch_syscall_invoke2()` , `z_syscall_trap()` ... are not yet defined when used.
In other words, the inclusion model is so that thoses functions are defined after being used.

ABOSTM, 2020-06-11:
^^ @ioannisg

ioannisg, 2020-06-11:
@ABOSTM not sure about this. I am able to compile arm_ramfunc test for nucleo_g071rb, which also has ARM MPU enabled.

ioannisg, 2020-06-11:
@ABOSTM Edit: I am able to build arm_ramfunc test for nucleo_l073rz, with CONFIG_ARM_MPU=y (obviously, with CPU_HAS_ARM_MPU on stm32l0/Kconfig.soc

ABOSTM, 2020-06-11:
@ioannisg , very interesting.
I dig a bit more, and found that the difference comes from soc.h
which `#include <kernel_includes.h>` in case of L0 and not on G0.
If I add this include in G0 I face the same compilation error.
And if I remove the include from L0 the error vanish.

Now I don't know whether removing the include is the solution. It has been introduced by PR #8780
```
/* ARM CMSIS definitions must be included before kernel_includes.h.
 * Therefore, it is essential to include kernel_includes.h after including
 * core SOC-specific headers.
 */
#include <kernel_includes.h>
```
Note: This include is present on many series: F0, F1, F2, F3, L0, L1, MP1

@erwango can you help dertermining the need of this include ?

ioannisg, 2020-06-11:
> @ioannisg , very interesting.
> I dig a bit more, and found that the difference comes from soc.h
> which `#include <kernel_includes.h>` in case of L0 and not on G0.
> If I add this include in G0 I face [...]

Remove this include (<kernel_includes.h>) and include <devicetree.h> instead. So make soc.h look like the soc.h in G0

ioannisg, 2020-06-11:
Short story: when building with MPU, CMSIS headers are included via arch.h and there is an inclusion cycle, so don't include `kernel_includes.h`
You still need to get the NVIC priority bits' number, so you need to include devicetree.h in your soc.h

ABOSTM, 2020-06-11:
OK thanks. 
So I should probably do the same for all STM32 series.

closed 2020-06-11 by ABOSTM