zephyrproject-rtos/zephyr #26136: CMake Error in Windows Enviroment

WilliamGFish, 2020-06-12:
**Describe the bug**
Compilation error:
Have merged the latest changes and now unable to compile zephyr
The error suggest that I have the wrong SDK but am running in Windows environment

**To Reproduce**
Configure as previously and run west

**Expected behavior**
West should find GCC and commence build

**Impact**
Showstopper

**Screenshots or console output**
- west build: generating a build system
-- Application: C:/Users/BillyBob/zephyrGoodX/zephyrNew/sensor_server
-- Zephyr version: 2.3.99 (C:/Users/BillyBob/zephyrGoodX/zephyrNew)
-- Board: particle_argon
CMake Error at C:/Users/BillyBob/zephyrGoodX/zephyrNew/cmake/toolchain/zephyr/host-tools.cmake:124 (message):
  The SDK version you are using is not supported, please update your SDK.

  You need SDK version 0.11.1 or newer.

  You have version(s):

  The SDK can be downloaded from:

  https://github.com/zephyrproject-rtos/sdk-ng/releases/download/v0.11.3/zephyr-sdk-0.11.3-setup.run

Call Stack (most recent call first):
  C:/Users/BillyBob/zephyrGoodX/zephyrNew/cmake/host-tools.cmake:3 (include)
  C:/Users/BillyBob/zephyrGoodX/zephyrNew/cmake/app/boilerplate.cmake:490 (include)
  CMakeLists.txt:4 (include)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' '-BC:\Users\BillyBob\zephyrGoodX\zephyrNew\sensor_server\build' '-SC:\Users\BillyBob\zephyrGoodX\zephyrNew\sensor_server' -GNinja -DBOARD=particle_argon

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

stephanosio, 2020-06-12:
Did you configure your environment correctly? In particular, the `ZEPHYR_TOOLCHAIN_VARIANT` and the toolchain-specific environment variables.

WilliamGFish, 2020-06-12:
Hi Stephanos,
I have been using this set-up for several months without issue. Only after merging the latest “REQUIRED” changes (#25808?) do I receive this error.
I’m guessing that changes have been made that effect ‘old’ set-ups

set ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
set GNUARMEMB_TOOLCHAIN_PATH=C:\gnu_arm_embedded

stephanosio, 2020-06-12:
I just tested the latest master (288ae28c13c85d44772543894784682eb8cc8d63) on Windows and I am not seeing this problem.

Please double check that `ZEPHYR_TOOLCHAIN_VARIANT` is set to `gnuarmemb` from the context you are running cmake.

WilliamGFish, 2020-06-12:
I've checked by adding a couple of line to: \cmake\toolchain\zephyr\host-tools.cmake
message(ZEPHYR_TOOLCHAIN_VARIANT "Found:  (${ZEPHYR_TOOLCHAIN_VARIANT})")
message(ZEPHYR_SDK_INSTALL_DIR "Found:  (${ZEPHYR_SDK_INSTALL_DIR})")

Result:
ZEPHYR_TOOLCHAIN_VARIANTFound:  (gnuarmemb)
ZEPHYR_SDK_INSTALL_DIRFound:  ()

Hence I'm confused.

WilliamGFish, 2020-06-12:
I have completely removed zephyr and started again. All appears well.

confused, but closing

closed 2020-06-12 by WilliamGFish

stephanosio, 2020-06-12:
It might have been a cache related issue. Deleting `.cache` directory should usually fix these kind of problems.

enurseitov, 2021-07-19:
Had the same confusing issue on Linux, deleting `build` directory helped.