zephyrproject-rtos/zephyr #26142: cmake warning: "Policy CMP0077 is not set"

koffes, 2020-06-12:
When compiling an application using libmetal/open-amp I get the following warnings:

```modules/lib/open-amp/open-amp/cmake/options.cmake:58 (option):
  Policy CMP0077 is not set: option() honors normal variables.  Run "cmake
  --help-policy CMP0077" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.`
```

https://github.com/zephyrproject-rtos/libmetal/blob/master/libmetal/cmake/options.cmake#L55

Same warning for:
https://github.com/zephyrproject-rtos/open-amp/blob/master/open-amp/cmake/options.cmake#L51
https://github.com/zephyrproject-rtos/open-amp/blob/master/open-amp/cmake/options.cmake#L52

To reproduce:
You need cmake >= 3.13
Can be recreated with e.g. ```samples/bluetooth/hci_rpmsg```

tejlmand, 2020-08-17:
Waiting for:
- https://github.com/OpenAMP/open-amp/pull/213 to be upmerged into: https://github.com/OpenAMP/open-amp
- https://github.com/OpenAMP/libmetal/pull/119 to be upmerged into: https://github.com/OpenAMP/libmetal

koffes, 2020-08-17:
Can confirm that this is still an issue in latest master by compiling samples/bluetooth/hci_rpmsg.

koffes, 2020-10-05:
The discussed PRs have been merged, and the manifest for libmetal has been updated. Now, just missing an update for open-amp.

galak, 2020-10-19:
libmetal & openamp are working on a release now.  Once that gets done, I'll look at updating zephyr to those new released versions and pull in the fixes.

closed 2020-10-29 by carlescufi