zephyrproject-rtos/zephyr #26169: Enable -O0 for only one *.c file

Nukersson, 2020-06-13:
Is it possible to enable -O0 compilation option for one *.c file only in Zephyr's toolchain?

P.S.: In my case I would like use -O0 for one particular **eth_stm32_hal.c** file.

tejlmand, 2020-06-15:
Yes, using native CMake, this is possible:

```
set_source_file_properties(<path-to>/eth_stm32_hal.c PROPERTIES COMPILE_OPTIONS  "-O0")
```
https://cmake.org/cmake/help/v3.13/command/set_source_files_properties.html#command:set_source_files_properties

or
```
set_property(SOURCE <path-to>/eth_stm32_hal.c [APPEND] PROPERTY COMPILE_OPTIONS  "-O0")
```

But be aware of, that is the specific case of `eth_stm32_hal.c`, then there is a small detail:
>  Source file properties are visible only to targets added in the same directory (CMakeLists.txt).

which means, adding this line into https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/ethernet/CMakeLists.txt will not give the desired effect, as the file is compiled into zephyr lib, defined at top-level CMakeLists.txt file. (and not in the CMakeLists.txt file where `eth_stm32_hal.c` is added)

So the line would need to be added in:
https://github.com/zephyrproject-rtos/zephyr/blob/master/CMakeLists.txt

which is not so elegant.
(CMake 3.18-rc1 allows the DIRECTORY keyword for better control: https://cmake.org/cmake/help/latest/command/set_source_files_properties.html#command:set_source_files_properties
, but Zephyr only requires 3.13)

Note it is often possible to use `${CMAKE_CURRENT_LIST_DIR}` as `<path-to>` in above example.

Nukersson, 2020-06-15:
@tejlmand Thank you very much for very good explanation!

Nukersson, 2020-06-15:
@tejlmand based on your good explanation, would it be possible to add additional Zephyr doc's information, on how to use -O0 for separate *.c files?

tejlmand, 2020-06-15:
@Nukersson Thanks for your feedback.
Regarding additional docs, then I will consider your suggestion.
There is a fine line between describing useful CMake functionality in scope of Zephyr but avoid to duplicate CMake's own documentation into Zephyr.

In general, compile options are managed through Kconfig, but as your question shows, there are certain cases where there are reasons for using CMake features directly.
So the specific use of `set_source_files_properties` might be worth documenting.

closed 2020-06-15 by tejlmand

mikeh122, 2023-03-08:
Just an FYI:

Regarding the comment above that

> CMake 3.18-rc1 allows the DIRECTORY keyword for better control

With tejlmand's help, I was able to leverage that DIRECTORY keyword when using CMake 3.20 to adjust the default compiler options for specific Zephyr files.

In our project CMakeLists.txt file, I addded the example lines below to change the compiler optimzer option from our project default -O2 to -Os for certain Zephyr files.

Example 1:
```
file(GLOB SRCS_SUBSYS_CONSOLE ${ZEPHYR_BASE}/subsys/console/*.c)
set_source_files_properties(${SRCS_SUBSYS_CONSOLE} DIRECTORY ${ZEPHYR_BASE} PROPERTIES COMPILE_OPTIONS "-Os")

```
However, for other directories, I found I had to adjust the DIRECTORY argument such as below: 

Example 2:
```
file(GLOB SRCS_DRIVERS_GPIO ${ZEPHYR_BASE}/drivers/gpio/gpio_nrfx.c)
set_source_files_properties(${SRCS_DRIVERS_GPIO} DIRECTORY ${ZEPHYR_BASE}/drivers/gpio PROPERTIES COMPILE_OPTIONS "-Os")

```